fwupd/fwupd #14: test suite seems to fail

superm1, 2015-06-18:
With no changes to code, I'm finding the test suite is failing on Ubuntu 14.04.

```
=====================================
   fwupd 0.1.3: src/test-suite.log
=====================================

# TOTAL: 1
# PASS:  0
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0

.. contents:: :depth: 2

FAIL: fu-self-test
==================

/fwupd/cab: OK
/fwupd/pending: OK
/fwupd/provider: OK
/fwupd/keyring: **
Fu:ERROR:fu-self-test.c:326:fu_keyring_func: assertion failed (error == NULL): Could not check signature '17ACBA8DFA970E17' as no public key (FwupdError, 11)
```

hughsie, 2015-06-18:
I can't reproduce here; is this just "make check"?

superm1, 2015-06-19:
Yeah, just ```make check```.  The version of gpgme on the system is 1.4.3 with a few various distro provided patches laid on top for security fixes.  It's the latest version in Ubuntu 14.04 (https://launchpad.net/ubuntu/+source/gpgme1.0/1.4.3-0.1ubuntu5.1).

Here's the full output:
```
$ make check
Making check in libfwupd
make[1]: Entering directory `/home/supermario/src/efi-fw-packages/fwupd/libfwupd'
make[1]: Nothing to be done for `check'.
make[1]: Leaving directory `/home/supermario/src/efi-fw-packages/fwupd/libfwupd'
Making check in po
make[1]: Entering directory `/home/supermario/src/efi-fw-packages/fwupd/po'
INTLTOOL_EXTRACT="/usr/bin/intltool-extract" XGETTEXT="/usr/bin/xgettext" srcdir=. /usr/bin/intltool-update --gettext-package fwupd --pot
rm -f missing notexist
srcdir=. /usr/bin/intltool-update -m
if [ -r missing -o -r notexist ]; then \
          exit 1; \
        fi
make[1]: Leaving directory `/home/supermario/src/efi-fw-packages/fwupd/po'
Making check in data
make[1]: Entering directory `/home/supermario/src/efi-fw-packages/fwupd/data'
Making check in tests
make[2]: Entering directory `/home/supermario/src/efi-fw-packages/fwupd/data/tests'
make[2]: Nothing to be done for `check'.
make[2]: Leaving directory `/home/supermario/src/efi-fw-packages/fwupd/data/tests'
Making check in pki
make[2]: Entering directory `/home/supermario/src/efi-fw-packages/fwupd/data/pki'
make[2]: Nothing to be done for `check'.
make[2]: Leaving directory `/home/supermario/src/efi-fw-packages/fwupd/data/pki'
make[2]: Entering directory `/home/supermario/src/efi-fw-packages/fwupd/data'
make[2]: Nothing to be done for `check-am'.
make[2]: Leaving directory `/home/supermario/src/efi-fw-packages/fwupd/data'
make[1]: Leaving directory `/home/supermario/src/efi-fw-packages/fwupd/data'
Making check in man
make[1]: Entering directory `/home/supermario/src/efi-fw-packages/fwupd/man'
make[1]: Nothing to be done for `check'.
make[1]: Leaving directory `/home/supermario/src/efi-fw-packages/fwupd/man'
Making check in policy
make[1]: Entering directory `/home/supermario/src/efi-fw-packages/fwupd/policy'
make[1]: Nothing to be done for `check'.
make[1]: Leaving directory `/home/supermario/src/efi-fw-packages/fwupd/policy'
Making check in src
make[1]: Entering directory `/home/supermario/src/efi-fw-packages/fwupd/src'
make  check-am
make[2]: Entering directory `/home/supermario/src/efi-fw-packages/fwupd/src'
make  fu-self-test
make[3]: Entering directory `/home/supermario/src/efi-fw-packages/fwupd/src'
  CC       fu_self_test-fu-cab.o
  CC       fu_self_test-fu-device.o
  CC       fu_self_test-fu-keyring.o
  CC       fu_self_test-fu-pending.o
  CC       fu_self_test-fu-provider.o
  CC       fu_self_test-fu-provider-fake.o
  CC       fu_self_test-fu-self-test.o
  CCLD     fu-self-test
make[3]: Leaving directory `/home/supermario/src/efi-fw-packages/fwupd/src'
make  check-TESTS
make[3]: Entering directory `/home/supermario/src/efi-fw-packages/fwupd/src'
make[4]: Entering directory `/home/supermario/src/efi-fw-packages/fwupd/src'
../test-driver: line 107: 29294 Aborted                 (core dumped) "$@" > $log_file 2>&1
FAIL: /home/supermario/src/efi-fw-packages/fwupd/src/.libs/lt-fu-self-test
make[5]: Entering directory `/home/supermario/src/efi-fw-packages/fwupd/src'
make  all-am
make[6]: Entering directory `/home/supermario/src/efi-fw-packages/fwupd/src'
make[6]: Nothing to be done for `all-am'.
make[6]: Leaving directory `/home/supermario/src/efi-fw-packages/fwupd/src'
make[5]: Leaving directory `/home/supermario/src/efi-fw-packages/fwupd/src'
============================================================================
Testsuite summary for fwupd 0.1.3
============================================================================
# TOTAL: 1
# PASS:  0
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0
============================================================================
See src/test-suite.log
Please report to https://github.com/hughsie/fwupd
============================================================================
make[4]: *** [test-suite.log] Error 1
make[4]: Leaving directory `/home/supermario/src/efi-fw-packages/fwupd/src'
make[3]: *** [check-TESTS] Error 2
make[3]: Leaving directory `/home/supermario/src/efi-fw-packages/fwupd/src'
make[2]: *** [check-am] Error 2
make[2]: Leaving directory `/home/supermario/src/efi-fw-packages/fwupd/src'
make[1]: *** [check] Error 2
make[1]: Leaving directory `/home/supermario/src/efi-fw-packages/fwupd/src'
make: *** [check-recursive] Error 1
```

phomes, 2015-07-26:
I get the same error for master on fedora 22 (also gpgme 1.4.3). 

Here is a strace if that helps:

```
write(1, "/fwupd/keyring: ", 16)        = 16
futex(0x7fe3711b24a8, FUTEX_WAKE_PRIVATE, 2147483647) = 0
getcwd("/home/phomes/fwupd/src", 4096)  = 23
lstat("/home/phomes/fwupd/data", {st_mode=S_IFDIR|0775, st_size=582, ...}) = 0
lstat("/home/phomes/fwupd/data/tests", {st_mode=S_IFDIR|0775, st_size=182, ...}) = 0
lstat("/home/phomes/fwupd/data/tests/pki", {st_mode=S_IFDIR|0775, st_size=46, ...}) = 0
rt_sigaction(SIGPIPE, NULL, {SIG_IGN, [PIPE], SA_RESTORER|SA_RESTART, 0x7fe3708d1a50}, 8) = 0
open("/proc/meminfo", O_RDONLY|O_CLOEXEC) = 6
fstat(6, {st_mode=S_IFREG|0444, st_size=0, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fe372b22000
read(6, "MemTotal:        8023564 kB\nMemF"..., 1024) = 1024
close(6)                                = 0
munmap(0x7fe372b22000, 4096)            = 0
access("/usr/bin/gpgconf", F_OK)        = 0
pipe([6, 7])                            = 0
fcntl(6, F_SETFD, FD_CLOEXEC)           = 0
clone(child_stack=0, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0x7fe372b04ad0) = 26075
wait4(26075, [{WIFEXITED(s) && WEXITSTATUS(s) == 0}], 0, NULL) = 26075
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=26075, si_uid=0, si_status=0, si_utime=0, si_stime=0} ---
close(7)                                = 0
read(6, "sysconfdir:/etc/gnupg\nbindir:/us"..., 1023) = 280
read(6, "", 1023)                       = 0
close(6)                                = 0
pipe([6, 7])                            = 0
fcntl(6, F_SETFD, FD_CLOEXEC)           = 0
clone(child_stack=0, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0x7fe372b04ad0) = 26077
wait4(26077, [{WIFEXITED(s) && WEXITSTATUS(s) == 0}], 0, NULL) = 26077
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=26077, si_uid=0, si_status=0, si_utime=0, si_stime=0} ---
close(7)                                = 0
read(6, "gpg:GPG for OpenPGP:/usr/bin/gpg"..., 1023) = 252
read(6, "", 1023)                       = 0
close(6)                                = 0
pipe([6, 7])                            = 0
fcntl(6, F_SETFD, FD_CLOEXEC)           = 0
clone(child_stack=0, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0x7fe372b04ad0) = 26079
wait4(26079, [{WIFEXITED(s) && WEXITSTATUS(s) == 0}], 0, NULL) = 26079
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=26079, si_uid=0, si_status=0, si_utime=0, si_stime=0} ---
close(7)                                = 0
read(6, "gpg (GnuPG) 2.1.5\nlibgcrypt 1.6."..., 79) = 79
close(6)                                = 0
pipe([6, 7])                            = 0
fcntl(6, F_SETFD, FD_CLOEXEC)           = 0
clone(child_stack=0, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0x7fe372b04ad0) = 26081
wait4(26081, [{WIFEXITED(s) && WEXITSTATUS(s) == 0}], 0, NULL) = 26081
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=26081, si_uid=0, si_status=0, si_utime=0, si_stime=0} ---
close(7)                                = 0
read(6, "", 79)                         = 0
close(6)                                = 0
pipe([6, 7])                            = 0
fcntl(6, F_SETFD, FD_CLOEXEC)           = 0
clone(child_stack=0, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0x7fe372b04ad0) = 26083
wait4(26083, [{WIFEXITED(s) && WEXITSTATUS(s) == 0}], 0, NULL) = 26083
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=26083, si_uid=0, si_status=0, si_utime=0, si_stime=0} ---
close(7)                                = 0
read(6, "gpgconf (GnuPG) 2.1.5\nCopyright "..., 79) = 79
close(6)                                = 0
openat(AT_FDCWD, "/home/phomes/fwupd/data/tests/pki", O_RDONLY|O_NONBLOCK|O_DIRECTORY|O_CLOEXEC) = 6
getdents(6, /* 3 entries */, 32768)     = 96
stat("/home/phomes/fwupd/data/tests/pki/GPG-KEY-Hughski-Limited", {st_mode=S_IFREG|0664, st_size=1702, ...}) = 0
open("/home/phomes/fwupd/data/tests/pki/GPG-KEY-Hughski-Limited", O_RDONLY) = 7
fstat(7, {st_mode=S_IFREG|0664, st_size=1702, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fe372b22000
lseek(7, 0, SEEK_SET)                   = 0
read(7, "-----BEGIN PGP PUBLIC KEY BLOCK-"..., 4096) = 1702
close(7)                                = 0
[... 72 lines cut ...]
select(12, [], [11], NULL, {0, 0})      = 1 (out [11], left {0, 0})
write(11, "\2201\r(\0\0\0\0~\24\2201 \0\177\10\360\0p\10\377\0~\20\t\0\2201\17(\374\1"..., 4096) = 4096
select(12, [6], [9 11], NULL, {1, 0})   = 2 (out [9 11], left {0, 999997})
select(10, [], [9], NULL, {0, 0})       = 1 (out [9], left {0, 0})
close(9)                                = 0
select(12, [], [11], NULL, {0, 0})      = 1 (out [11], left {0, 0})
write(11, "\2341\206,\2341\211,\2341\273,\2341\211,\2341\273,\2341m,\2341v,\2341\244,"..., 3904) = 3904
select(12, [6], [11], NULL, {1, 0})     = 1 (out [11], left {0, 999998})
select(12, [], [11], NULL, {0, 0})      = 1 (out [11], left {0, 0})
close(11)                               = 0
select(7, [6], [], NULL, {1, 0})        = 1 (in [6], left {0, 999091})
select(7, [6], [], NULL, {0, 0})        = 1 (in [6], left {0, 0})
read(6, "[GNUPG:] PROGRESS -&8 ? 0 0\n[GNU"..., 1024) = 224
select(7, [6], [], NULL, {1, 0})        = 1 (in [6], left {0, 999998})
select(7, [6], [], NULL, {0, 0})        = 1 (in [6], left {0, 0})
read(6, "", 1024)                       = 0
close(6)                                = 0
write(2, "**\nFu:ERROR:fu-self-test.c:446:f"..., 161**
Fu:ERROR:fu-self-test.c:446:fu_keyring_func: assertion failed (error == NULL): Could not check signature '17ACBA8DFA970E17' as no public key (FwupdError, 11)
) = 161
rt_sigprocmask(SIG_UNBLOCK, [ABRT], NULL, 8) = 0
tgkill(26060, 26060, SIGABRT)           = 0
--- SIGABRT {si_signo=SIGABRT, si_code=SI_TKILL, si_pid=26060, si_uid=0} ---
+++ killed by SIGABRT (core dumped) +++
```

phomes, 2015-07-27:
BTW, this is with --disable-colorhug. Not sure if it matters.

closed 2015-07-28 by hughsie