fwupd/fwupd #29: ELF header incorrect for fwupd, fwupdmgr, fwsignd

superm1, 2015-08-14:
In trying to debug why Debian packaging was issuing a lintian error (https://lintian.debian.org/tags/sharedobject-in-library-directory-missing-soname.html) I found that the ELF header for these binaries is incorrect.

This is what it looks like currently:
```
$ readelf -h src/.libs/fwupd | grep Type
  Type:                              DYN (Shared object file)
```

I determined that this was happening because they were linked with ```-pie``` to make position independent executables.  Is this intended behavior?  It seems to me that there isn't a reason for them to be PIC.  It's a really simple patch to remove it.  Just wanted to confirm with you before making the change.  

```
$ readelf -h src/.libs/fwupd | grep Type
  Type:                              EXEC (Executable file)
```

ximion, 2015-08-14:
This might be intentionally compiled with PIE, to make Linux' ASLR work.
The Lintian warning just means that you shouldn't put this into /usr/lib/<triplet>/, sorry that I wasn't more explicit about that...

superm1, 2015-08-14:
Ah OK.  This is probably the right approach then:

```
diff --git a/data/org.freedesktop.fwupd.service.in b/data/org.freedesktop.fwupd.service.in
index df22a4c..a7b713a 100644
--- a/data/org.freedesktop.fwupd.service.in
+++ b/data/org.freedesktop.fwupd.service.in
@@ -1,5 +1,5 @@
 [D-BUS Service]
 Name=org.freedesktop.fwupd
-Exec=@servicedir@/fwupd
+Exec=@servicedir@/fwupd/fwupd
 User=root
 #SystemdService=fwupd.service
diff --git a/src/Makefile.am b/src/Makefile.am
index 11b8c93..6cf47cf 100644
--- a/src/Makefile.am
+++ b/src/Makefile.am
@@ -80,7 +80,7 @@ fu-resources.h: fwupd.gresource.xml
 		--c-name fu				\
 		$(srcdir)/fwupd.gresource.xml
 
-libexec_PROGRAMS =					\
+pkglibexec_PROGRAMS =					\
 	fwupd
 
 fwupd_SOURCES =						\
@@ -146,7 +146,7 @@ fwupd_CFLAGS =						\
 	$(WARNINGFLAGS_C)
 
 if HAVE_FWSIGND
-libexec_PROGRAMS += fwsignd
+pkglibexec_PROGRAMS += fwsignd
 
 fwsignd_SOURCES =					\
 	fu-cleanup.h					\
```

closed 2015-08-15 by superm1

ximion, 2015-08-15:
...or you could have passed --libexecdir= manually in the debian/rules file ;-)
But this solution also works.

superm1, 2015-08-15:
Passing libexecdir in debian/rules would have still meant needing to
override the dbus and systemd files.  This is a cleaner way I think.