fwupd/fwupd #31: Fallback from online to offline mode isn't working properly (again)

superm1, 2015-09-22:
For some reason I can't identify it seems that automatic fallback from online to offline mode on UEFI updates is failing.  
I know for sure this was working before, and nothing obvious in the code has changed.  

From adding debug code, at least to me it's looking like the quark that is found in fu-util.c isn't matching FwupdError even though that's the quark sent from fu-provider.c.  Previously I was testing on an older glib2.0, so this could be a glib bug?  
The test is now on 2.45.7.

Adding debug code to the error in fu-provider.c gets this output:
```
provider quark: 319
string for quark: FwupdError
provider code: 10
provider message No online update possible
```
Adding debug code to the error in fu-util.c 
```
$ ./fwupdmgr install ~/firmware.cab
 * Loading firmware
 * Decompressing firmware
> finished offline test
> priv->flags & FU_PROVIDER_UPDATE_FLAG_OFFLINE 0
> g_error_matches 0
> FWUPD_ERROR: 182
> string for quark(FWUPD_ERROR): FwupdError
> quark: 159
> string for quark(error_local->quark): g-io-error-quark
> FWUPD_ERROR_NOT_SUPPORTED: 10
> error_local->code: 36
> error_local->message: No online update possible
> returning false
No online update possible
```

superm1, 2015-09-22:
Okay, so I've got a fix for this that translates the dbus errors.  Based on the fix, I'm really not sure why this would have worked in the first place, but take a look at it.

https://github.com/superm1/fwupd/commit/c58db299aa15abe9580ff80150b79b81bf374077

If that looks good go ahead and commit, or i'll commit it tomorrow.  If you've got some minor nits go ahead and fix 'em and commit too.

hughsie, 2015-09-23:
It was a bit easier than that; basically, fwupd_error_quark() sets up the D-Bus remote <=> local mapping inside Glib. It must have worked before as we were using FWUPD_ERROR() for the error domain for the failure before we checked the failure code. It's much better to register these explicitly at startup, which is what my patch does.

closed 2015-09-23 by hughsie