fwupd/fwupd #34: metainfo version doesn't match inf

jiazhang0, 2015-10-17:
I got such an failure when running a online mode update on Dell SLA:

root@WR-IntelligentDevice:/tmp# fwupdmgr get-devices
Device: UEFI-e0f614ed-fb82-467a-a34e-71172cc07e4d-dev0
  Provider:        UEFI
  Guid:            e0f614ed-fb82-467a-a34e-71172cc07e4d
  Version:         3.1
  UrlHomepage:     http://support.dell.com/
  Name:            IOT Gateway 5500 System Update
  Summary:         Firmware for the Dell IOT Gateway 5500
  Description:     <p>Updating the system firmware improves performance.</p>
  Internal:        True
  AllowOnline:     False
  AllowOffline:    True
root@WR-IntelligentDevice:~# fwupdmgr refresh
root@WR-IntelligentDevice:~# fwupdmgr get-updates
UEFI-e0f614ed-fb82-467a-a34e-71172cc07e4d-dev0 has firmware updates:
Version:             3.3
Checksum:            73f12a2a3a8c4974bc8f7494e294583fba5bb3e7
Location:            https://secure-lvfs.rhcloud.com/downloads/6f03ffe3028bb2a233323595556e8fafce2678fb-firmware.cab
root@WR-IntelligentDevice:~# fwupdmgr update
Downloading 3.3 for UEFI-e0f614ed-fb82-467a-a34e-71172cc07e4d-dev0...
Updating 3.3 on UEFI-e0f614ed-fb82-467a-a34e-71172cc07e4d-dev0...
 * Decompressing firmware
 * Idle
metainfo version is 3.3 while inf has 00.03.03

This issue doesn't happen with the commit 56b61abea965c94246c957637cc5814285bdf1d2.

hughsie, 2015-10-17:
Well, 00.03.03 is < 3.3 using our rules. I really think we have to put some kind of limit on the version numbers we accept at some point, else we're going to be tangled up in complexity. I'll see if I can do something without making this too complicated.

hughsie, 2015-10-17:
@superm1 I think we're going to have to be a bot more strict about versioning, it's a bit of a mess at the moment. At the moment the as_utils_version_parse() function accepts:

* A dotted decimal, e.g. 1.2.3 or 01.02.03.04
* A base10 integer, e.g. "12345"
* A base16 string, e.g. "0x04030201"

I don't think you can actually do anything clever with removing preceding dots as this changes the first digit priority. You also can't assume that we just shunt 1.2 into the CC.DD of AA.BB.CC.DD as we've also got the MS-style AA.BB.CCDD version triples and non-UEFI mixed string versions like 1.2.alpha and 1.2.beta.

Could we just create firmware where the same number of dotted decimals are used in both the .inf file and the metainfo.xml file? This means we either have "0.3.3" in both files, or "3.3" in both files. I don't actually thing adding the internal 0 padding makes sense, unless you're using hex versions like 00.0E.03 as you've still got to change the "width" when you get to 00.00.100

For UEFI versions, I think either "3.3" makes sense in both, or 0.0.3.3, or 000.000.003.003 but I don't think we want to try to make huge assumptions when matching up the version numbers.

Also, I think it makes sense to use a triplet like 0.1.2 when we're using the MS/Intel encoding of AA.BB.CCDD, and using a quad like 0.0.1.2 when we're using the Dell AA.BB.CC.DD encoding. At the moment Dell seems to be using the triplet with an optional zero and completely ignoring the AA byte.

Furthermore, I think using 3.3 in both files is also a bad idea, as the lesser number of sections will win when you want to use the AA/BB byte in the future -- e.g.

    rpmdev-vercmp 3.3 1.0.0
    3.3 > 1.0.0

Of course, the alternative is to just not parse the .inf file at all, and just use whatever string is in the metainfo file. This means vendors could get it wrong, but solves the problem shown here. In this case you can put whatever you want in the .inf for display only, but you'd still probably want 0.0.3.3 in the metainfo file for the `rpmdev-vercmp` reason I showed above.

superm1, 2015-10-17:
Just to be clear, in this particular instance:

the INF has the following:
```
DriverVer: 00.03.03
HKR,,FirmwareVersion,%REG_DWORD%,0x303
```

Where the XML has:
```
    <release version="0x303" timestamp="1444293624">
```

Since in the end you are comparing to an ESRT table value on the system (in this example 0x301), I do think it makes most sense to internally store and compare the value of version from metainfo.xml or from FirmwareVersion in the INF at all times.  This doesn't have to affect what's rendered, only what's stored and compared in appstream data.

Doing it this way you can still parse DriverVer for "display" purposes only and you don't have to keep track of any vendor's versioning strategies.

In terms of the concern going 00.03.03 (rendering as 3.3) to 1.0.0, this would also be solved if always comparing the ESRT version.

To avoid the situation of a vendor getting it wrong you can do a test on uploads to make sure that the version in the XML matches FirmwareVersion from the INF.

jiazhang0, 2015-10-19:
This issue is fixed with the latest commits from appstream-glib and fwupd. This issue can be closed.

closed 2015-10-19 by jiazhang0