fwupd/fwupd #35: failed to update metadata

jiazhang0, 2015-10-21:
This issue doesn't always happen on DELL SLA board.

```
# fwupdmgr refresh
failed to update metadata: Failed to parse XML: Error on line 51 char 2: Document must begin with an element (e.g. <book>) 
```

The workaround is to kill fwupd: `pkill -9 fwupd`

hughsie, 2015-10-21:
So, what happens here I think is that you're getting garbage (e.g. a captive portal HTML page) rather than an XML file, which we happily send to fwupd. This then clears out any existing entries, loads the XML, fails, and returns with an error. Of course, by this point we've cleared the in-memory store of firmware.

To address both parts of this bug, can you try doing:

    wget https://secure-lvfs.rhcloud.com/downloads/firmware.xml.gz

And then trying to open the XML file in vim. This should show a file starting with `<components><component>` -- you might have to retry a few times if the network is flaky. I can fix the other side of this bug (not clearing the in-memory cache until we've validated the XML) on Friday. Thanks!

jiazhang0, 2015-10-27:
@hughsie Sorry for the late reply, because this defect occurs occasionally.
As you said, the XML file looks good and the md5sum of downloaded firmware.xml.gz(d2e1fc0e5ede26bf95a2833cc653cebc) is same with the one shen no such a failure happens.

closed 2015-10-27 by hughsie