fwupd/fwupd #38: Failed to parse /var/cache/app-info/xmls/fwupd.xml file

jiazhang0, 2015-11-11:
All related components (efivar, fwupd, fwupdate and appstream-glib) are updated to the latest. The following failure is observed:

```
root@WR-IntelligentDevice:~# ls /var/cache/app-info/xmls
root@WR-IntelligentDevice:~# fwupdmgr refresh
root@WR-IntelligentDevice:~# md5sum /var/cache/app-info/xmls/fwupd.xml 
4bd6637c7bda81a6753b58653a9a9ab2  /var/cache/app-info/xmls/fwupd.xml
root@WR-IntelligentDevice:~# fwupdmgr get-devices
Device: UEFI-e0f614ed-fb82-467a-a34e-71172cc07e4d-dev0
  Provider:        UEFI
  Guid:            e0f614ed-fb82-467a-a34e-71172cc07e4d
  Version:         0.0.774
  Internal:        True
  AllowOnline:     False
  AllowOffline:    True
root@WR-IntelligentDevice:~# fwupdmgr get-updates
UEFI-e0f614ed-fb82-467a-a34e-71172cc07e4d-dev0 has firmware updates:
Version:             775
Checksum:            023b40254142652049aff1f0948cf0d7e60e80ae
Location:            https://secure-lvfs.rhcloud.com/downloads/1770ea641d2b97c7b59f4173931d423ae225da59-firmware.cab
root@WR-IntelligentDevice:~# fwupdmgr update
Downloading 775 for UEFI-e0f614ed-fb82-467a-a34e-71172cc07e4d-dev0...
Updating 775 on UEFI-e0f614ed-fb82-467a-a34e-71172cc07e4d-dev0...
 * Decompressing firmware
Retrying as an offline update...
 * Idle
 * Decompressing firmware
 * Scheduling upgrade
Message did not receive a reply (timeout by message bus)
root@WR-IntelligentDevice:~# fwupdmgr get-devices
Failed to connect to fwupd: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached
root@WR-IntelligentDevice:~# /usr/lib64/fwupd/fwupd/fwupd -v
02:35:33        Verbose debugging enabled (on console 1)
FuMain: failed to load AppStream data: Failed to parse /var/cache/app-info/xmls/fwupd.xml file: cannot process file of type text/plain
02:35:33        FuMain: failed to load AppStream data: Failed to parse /var/cache/app-info/xmls/fwupd.xml file: cannot process file of type text/plain
root@WR-IntelligentDevice:~# md5sum /var/cache/app-info/xmls/fwupd.xml 
4bd6637c7bda81a6753b58653a9a9ab2  /var/cache/app-info/xmls/fwupd.xml
```

After removing /var/cache/app-info/xmls/fwupd.xml, fwupdmgr becomes working.

superm1, 2015-11-11:
Do you still have a copy of fwupd.xml when this happened?  I think it might be helpful to see what garbage was in it at this time.

superm1, 2015-11-11:
I think this is a two part problem.  Part 1 is the metadata got corrupted at some point after download.  Part 2 is that it can't recover from corrupt metadata on the system.

I think part 2 can be solved by this:

```
From 63f99a88c6df0934d709d0db3352294952635c37 Mon Sep 17 00:00:00 2001
From: Mario Limonciello <mario_limonciello@dell.com>
Date: Tue, 10 Nov 2015 21:49:14 -0600
Subject: [PATCH] If local appstream data is corrupted, don't crash the app.

Just leave the store empty.  On the next refresh it will be
rewritten.
---
 src/fu-main.c | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/src/fu-main.c b/src/fu-main.c
index 9832f47..10306ea 100644
--- a/src/fu-main.c
+++ b/src/fu-main.c
@@ -1762,7 +1762,7 @@ main (int argc, char *argv[])
                            NULL, &error)){
                g_warning ("FuMain: failed to load AppStream data: %s",
                           error->message);
-               return FALSE;
+               g_clear_error(&error);
        }

        /* read config file */
--
1.9.1
```

jiazhang0, 2015-11-11:
@superm1 I post the content of fwupd.xml:

```
<?xml version="1.0" encoding="UTF-8"?>
<components version="0.9">
  <component type="firmware">
    <id>com.dell.uefi6180aaaa.firmware</id>
    <name>Latitude E5X60 System Update</name>
    <summary>Firmware for the Dell Latitude E5X60</summary>
    <developer_name>Dell Inc.</developer_name>
    <description><p>Updating the system firmware improves performance.</p></description>
    <url type="homepage">http://support.dell.com/</url>
    <releases>
      <release version="65796" timestamp="1446309800">
        <location>https://secure-lvfs.rhcloud.com/downloads/c1b448c3b1111e0a7d47a4517180d435cf7d790d-firmware.cab</location>
        <checksum filename="c1b448c3b1111e0a7d47a4517180d435cf7d790d-firmware.cab" target="container" type="sha1">de7885cdd1b344ca4c278ede3195c1718040de8c</checksum>
        <checksum filename="firmware.bin" target="content" type="sha1">5fecf123098465f1bae5b7072a4cfbf38eb00b28</checksum>
        <size type="installed">13479936</size>
        <size type="download">13193944</size>
      </release>
      <release version="65793" timestamp="1445016935">
        <location>https://secure-lvfs.rhcloud.com/downloads/8442086bf350f3867bd3f07a50aad0b699950bc9-firmware.cab</location>
        <checksum filename="8442086bf350f3867bd3f07a50aad0b699950bc9-firmware.cab" target="container" type="sha1">fe4edf539aa30617bb1f71e03654cfe4828de23d</checksum>
        <checksum filename="firmware.bin" target="content" type="sha1">714c75d9c49f21504d61cbaa85232d8f0afe86ff</checksum>
        <size type="installed">13472400</size>
        <size type="download">13189944</size>
      </release>
    </releases>
    <provides>
      <firmware type="flashed">6180aaaa-5529-4bbf-b4fd-65b4f788de5b</firmware>
    </provides>
  </component>
  <component type="firmware">
    <id>com.dell.uefi7039436b.firmware</id>
    <name>Alienware Echo MLK 13 System Update</name>
    <summary>Firmware for the Dell Echo MLK 13</summary>
    <developer_name>Dell Inc.</developer_name>
    <description><p>Updating the system firmware improves performance.</p></description>
    <url type="homepage">http://support.dell.com/</url>
    <releases>
      <release version="8931408" timestamp="1441967847">
        <location>https://secure-lvfs.rhcloud.com/downloads/65dc8a1334e48dc11c5c1d55c296afb4d21df67b-firmware.cab</location>
        <checksum filename="65dc8a1334e48dc11c5c1d55c296afb4d21df67b-firmware.cab" target="container" type="sha1">b09fd6b4abbe285ead6a6f30858680048b205655</checksum>
        <checksum filename="firmware.bin" target="content" type="sha1">7cb58ae43101ce1ac435beef1bb24cd7279fb4af</checksum>
        <size type="installed">16777216</size>
        <size type="download">4697151</size>
      </release>
      <release version="5779507" timestamp="1445839433">
        <location>https://secure-lvfs.rhcloud.com/downloads/afc3624de514f0a5d09c5b2b4f382681913cba5c-firmware.cab</location>
        <checksum filename="afc3624de514f0a5d09c5b2b4f382681913cba5c-firmware.cab" target="container" type="sha1">186fba97bba62e8244cf8d2f1be6659f1eb66ec2</checksum>
        <checksum filename="firmware.bin" target="content" type="sha1">7cb58ae43101ce1ac435beef1bb24cd7279fb4af</checksum>
        <size type="installed">16777216</size>
        <size type="download">4697151</size>
      </release>
    </releases>
    <provides>
      <firmware type="flashed">7039436b-6acf-433b-86a1-368ec2ef7e1f</firmware>
    </provides>
  </component>
  <component type="firmware">
    <id>com.dell.uefibc6bf60c.firmware</id>
    <name>Inspiron 13-7359 System Update</name>
    <summary>Firmware for the Dell Inspiron 13-7359</summary>
    <developer_name>Dell Inc.</developer_name>
    <description><p>Updating the system firmware improves performance.</p></description>
    <url type="homepage">http://support.dell.com/</url>
    <releases>
      <release version="66050" timestamp="1444186932">
        <location>https://secure-lvfs.rhcloud.com/downloads/35c5716495b2b6280d97cfe389605664aaa0a368-firmware.cab</location>
        <checksum filename="35c5716495b2b6280d97cfe389605664aaa0a368-firmware.cab" target="container" type="sha1">71ebddaa815dd9acf84f42f5773f32444855378d</checksum>
[... 20 lines cut ...]
        <checksum filename="firmware.bin" target="content" type="sha1">c6de16dcf81b4d5f124ce9002289013c0364a5a1</checksum>
        <size type="installed">4598985</size>
        <size type="download">4333870</size>
      </release>
      <release version="774" timestamp="1446115434">
        <location>https://secure-lvfs.rhcloud.com/downloads/0d7c4b8e46a6c9d91ba764d67f9c8d1e6265079f-firmware.cab</location>
        <checksum filename="0d7c4b8e46a6c9d91ba764d67f9c8d1e6265079f-firmware.cab" target="container" type="sha1">ecc9711ebcf3ce9dacfa38c277270a434a2753d8</checksum>
        <checksum filename="firmware.bin" target="content" type="sha1">fd62a5cfb74c424d44a408c08e66731c85de788a</checksum>
        <size type="installed">4597057</size>
        <size type="download">4331535</size>
      </release>
      <release version="773" timestamp="1445528162">
        <location>https://secure-lvfs.rhcloud.com/downloads/1d2534c4d9960ce0619e7fe0426db20666116628-firmware.cab</location>
        <checksum filename="1d2534c4d9960ce0619e7fe0426db20666116628-firmware.cab" target="container" type="sha1">1705e78417826c0748cabaebe2c1cf2fef73e856</checksum>
        <checksum filename="firmware.bin" target="content" type="sha1">62bee4747a4f141d6ab62df7876d2d6d79489200</checksum>
        <size type="installed">4595833</size>
        <size type="download">4331454</size>
      </release>
    </releases>
    <provides>
      <firmware type="flashed">e0f614ed-fb82-467a-a34e-71172cc07e4d</firmware>
    </provides>
  </component>
</components>
```

superm1, 2015-11-11:
Well that looks entirely accurate and has the exact same md5sum you had as previously.  Can you reliably reproduce this?

If so, before starting can you kill all running fwupd processes and run ```fwupd -v``` manually in another console and capture that output?  Hopefully it's more indicative of where the failure is.

superm1, 2015-11-11:
When you say you're running the latest versions, that means you are on fwupdate 1001d6a and fwupd 6fade1f with no patches on either, correct?

Also, there have recently been some changes at fwupdate that cause the NVRAM entry to change filenames (fwupdate.efi -> fwupx64.efi) and/or to use shim.  Due to these changes, I wonder if perhaps this could be causing a crash on libfwup0 specifically around the current state of BootXXXX and fwupdate-* entries in efivars on your system from the changes.

jiazhang0, 2015-11-12:
I'm using the following commits:

fwupd: 6fade1ffea19246987667a6acf35d9a6917693c9
fwupdate: 1001d6ac4b81ee5ed070e6026a5f7e0dc80b2797
appstream-glib: 1f100b43a4b719893b20792ffb1f3a0d07c4af38
efivar: 88c3ca81c9278baccdc2a23cd1de5cecb19398be

and can reproduce this issue 100%.

I follow your instruction to kill fwupd and see what happens next:
```
...
root@WR-IntelligentDevice:~# fwupdmgr update     
Downloading 775 for UEFI-e0f614ed-fb82-467a-a34e-71172cc07e4d-dev0...
Updating 775 on UEFI-e0f614ed-fb82-467a-a34e-71172cc07e4d-dev0...
 * Decompressing firmware
 * Idle
Retrying as an offline update...
 * Decompressing firmware
 * Scheduling upgrade
Message did not receive a reply (timeout by message bus)
root@WR-IntelligentDevice:~# ps aux|grep fw
root      2379  0.0  0.0   4400   504 pts/0    S+   03:29   0:00 grep fw
root@WR-IntelligentDevice:~# /usr/lib64/fwupd/fwupd/fwupd -v
03:30:20        Verbose debugging enabled (on console 1)
FuMain: failed to load AppStream data: Failed to parse /var/cache/app-info/xmls/fwupd.xml file: cannot process file of type text/plain
03:30:20        FuMain: failed to load AppStream data: Failed to parse /var/cache/app-info/xmls/fwupd.xml file: cannot process file of type text/plain
root@WR-IntelligentDevice:~# rm /var/cache/app-info/xmls/fwupd.xml 
root@WR-IntelligentDevice:~# /usr/lib64/fwupd/fwupd/fwupd -v
03:30:32        Verbose debugging enabled (on console 1)
03:30:32        Loading fallback values from /etc/fwupd.conf
03:30:32        adding udev device: /sys/devices/pci0000:00/0000:00:02.0
Failed to parse ROM from /sys/devices/pci0000:00/0000:00:02.0/rom: Error reading from file: Input/output error
03:30:32        Failed to parse ROM from /sys/devices/pci0000:00/0000:00:02.0/rom: Error reading from file: Input/output error
03:30:32        ignoring ro-usb:00:04:02 [Bluetooth and Wireless LAN Composite Device] as no version
03:30:32        ignoring ro-usb:00:02 [Dell USB Keyboard] as no version
03:30:32        ignoring ro-usb:00:01 [Cruzer Blade] as no version
03:30:32        emit added: UEFI-e0f614ed-fb82-467a-a34e-71172cc07e4d-dev0
####################                                                                   @0027ms AsStore:load 27ms
                                                   ##################################  @0111ms FuMain:coldplug 44ms
                                                   #######################             @0096ms FuMain:coldplug{Udev} 30ms
                                                   ######                              @0076ms FuProviderUdev:coldplug{usb} 9ms
                                                           ###############             @0096ms FuProviderUdev:coldplug{pci} 20ms
                                                             ########                  @0090ms FuProviderUdev:client-add{0x8086:0x0f31} 11ms
                                                              ######                   @0089ms FuRom:reading-data 8ms
                                                                           ########    @0108ms FuMain:coldplug{USB} 11ms
                                                                             ####      @0106ms FuProviderUsb:added{413c:2105} 6ms
                                                                             ###       @0105ms FuProviderUsb:get-string-desc 5ms
03:30:32        FuMain: acquired name: org.freedesktop.fwupd
03:30:32        devices now in store:
CTRL-C
root@WR-IntelligentDevice:~# ps aux|grep fw
root      2389  0.0  0.0   4400   504 pts/0    S+   03:30   0:00 grep fw
root@WR-IntelligentDevice:~# fwupdmgr get-devices
Device: UEFI-e0f614ed-fb82-467a-a34e-71172cc07e4d-dev0
  Provider:        UEFI
  Guid:            e0f614ed-fb82-467a-a34e-71172cc07e4d
  Version:         0.0.774
  Internal:        True
  AllowOnline:     False
  AllowOffline:    True
root@WR-IntelligentDevice:~# ps aux|grep fw
root      2394  6.0  0.2 399948  4940 ?        Sl   03:30   0:00 /usr/lib64/fwupd/fwupd/fwupd
root      2401  0.0  0.0   4400   504 pts/0    S+   03:30   0:00 grep fw
root@WR-IntelligentDevice:~# pkill -9 fwupd
root@WR-IntelligentDevice:~# ps aux|grep fw
root      2427  0.0  0.0   4400   504 pts/0    S+   03:33   0:00 grep fw
root@WR-IntelligentDevice:~# /usr/lib64/fwupd/fwupd/fwupd -v
03:33:05        Verbose debugging enabled (on console 1)
03:33:05        Loading fallback values from /etc/fwupd.conf
03:33:05        adding udev device: /sys/devices/pci0000:00/0000:00:02.0
Failed to parse ROM from /sys/devices/pci0000:00/0000:00:02.0/rom: Error reading from file: Input/output error
03:33:05        Failed to parse ROM from /sys/devices/pci0000:00/0000:00:02.0/rom: Error reading from file: Input/output error
03:33:05        ignoring ro-usb:00:04:02 [Bluetooth and Wireless LAN Composite Device] as no version
03:33:05        ignoring ro-usb:00:02 [Dell USB Keyboard] as no version
03:33:05        ignoring ro-usb:00:01 [Cruzer Blade] as no version
03:33:05        emit added: UEFI-e0f614ed-fb82-467a-a34e-71172cc07e4d-dev0
#######################                                                                @0028ms AsStore:load 28ms
                                                   ################################### @0103ms FuMain:coldplug 42ms
                                                   #######################             @0090ms FuMain:coldplug{Udev} 28ms
[... cut ...]
                                                                           #########   @0101ms FuMain:coldplug{USB} 11ms
                                                                             #####     @0100ms FuProviderUsb:added{413c:2105} 6ms
                                                                             ####      @0099ms FuProviderUsb:get-string-desc 5ms
03:33:05        FuMain: acquired name: org.freedesktop.fwupd
03:33:05        devices now in store:
```

For fwupx64.efi, we still move to it but currently we still use fwupdate.efi. Also, I don't see crash failure.

superm1, 2015-11-12:
What I was meaning was to have on one console ```fwupd -v ``` running and then on the other run ```fwupdmgr update```.  That should hopefully identify the first problem that's happening.  I've tried to reproduce this on my end without much success.

What version of glib2.0 do you have?  The error:
```FuMain: failed to load AppStream data: Failed to parse /var/cache/app-info/xmls/fwupd.xml file: cannot process file of type text/plain```

Comes directly from what glib is detecting the file as.  It feeds that into appstream-glib which feeds it back to fwupd (https://github.com/hughsie/appstream-glib/blob/master/libappstream-glib/as-node.c#L837)

jiazhang0, 2015-11-12:
@superm1 

Here is the log:
```
root@WR-IntelligentDevice:~# /usr/lib64/fwupd/fwupd/fwupd -v
07:02:21        Verbose debugging enabled (on console 1)
07:02:21        Loading fallback values from /etc/fwupd.conf
07:02:21        adding udev device: /sys/devices/pci0000:00/0000:00:02.0
Failed to parse ROM from /sys/devices/pci0000:00/0000:00:02.0/rom: Error reading from file: Input/output error
07:02:21        Failed to parse ROM from /sys/devices/pci0000:00/0000:00:02.0/rom: Error reading from file: Input/output error
07:02:21        ignoring ro-usb:00:04:02 [Bluetooth and Wireless LAN Composite Device] as no version
07:02:21        ignoring ro-usb:00:02 [Dell USB Keyboard] as no version
07:02:21        ignoring ro-usb:00:01 [Cruzer Blade] as no version
07:02:21        emit added: UEFI-e0f614ed-fb82-467a-a34e-71172cc07e4d-dev0
#####################                                                                  @0027ms AsStore:load 27ms
                                                  ###################################  @0110ms FuMain:coldplug 45ms
                                                  ######################               @0094ms FuMain:coldplug{Udev} 29ms
                                                  #######                              @0074ms FuProviderUdev:coldplug{usb} 9ms
                                                          ##############               @0094ms FuProviderUdev:coldplug{pci} 19ms
                                                            #########                  @0089ms FuProviderUdev:client-add{0x8086:0x0f31} 12ms
                                                            #######                    @0087ms FuRom:reading-data 9ms
                                                                         ##########    @0107ms FuMain:coldplug{USB} 13ms
                                                                            #####      @0105ms FuProviderUsb:added{413c:2105} 7ms
                                                                            ####       @0104ms FuProviderUsb:get-string-desc 6ms
07:02:21        FuMain: acquired name: org.freedesktop.fwupd
07:02:21        devices now in store:
07:02:57        using GZip decompressor for data
07:02:57        Adding public key /etc/pki/fwupd-metadata/GPG-KEY-Linux-Vendor-Firmware-Service

(fwupd:2315): As-WARNING **: failed to rescan: Failed to parse /var/cache/app-info/xmls/fwupd.xml file: cannot process file of type text/plain
07:02:58        devices now in store:
07:02:58        1       com.dell.uefi6180aaaa.firmware  Latitude E5X60 System Update
07:02:58        2       com.dell.uefi7039436b.firmware  Alienware Echo MLK 13 System Update
07:02:58        3       com.dell.uefibc6bf60c.firmware  Inspiron 13-7359 System Update
07:02:58        4       com.dell.uefie0f614ed.firmware  Edge Gateway 5000/5100 System Update
```

glib version I used is 2.45.8.

jiazhang0, 2015-11-12:
@superm1 Get the root cause.
```
root@WR-IntelligentDevice:/tmp/test/DriverPackage# fwupdate -a `cat /sys/firmware/efi/esrt/entries/entry0/fw_class ` firmware.bin 
fwupdate: Could not set up firmware update: No such file or directory
```
Here is the strace log:
```
openat(AT_FDCWD, "/sys/firmware/efi/", O_RDONLY|O_NONBLOCK|O_DIRECTORY|O_CLOEXEC) = 7
openat(7, "fw_platform_size", O_RDONLY) = -1 ENOENT (No such file or directory)
close(7)                                = 0
lseek(5, 0, SEEK_SET)                   = 0
close(5)                                = 0
munmap(0x7f533d8f7000, 4096)            = 0
write(6, "\23\351\204pY\366,\7y\304@\247\271\247C\303oX\2719\271\355\356\232F<}6;\314W\346"..., 59) = 59
close(6)                                = 0
munmap(0x7f533d8f6000, 4096)            = 0
close(6)                                = -1 EBADF (Bad file descriptor)
write(2, "fwupdate: ", 10fwupdate: )              = 10
write(2, "Could not set up firmware update", 32Could not set up firmware update) = 32
write(2, ": ", 2: )                       = 2
write(2, "No such file or directory\n", 26No such file or directory
) = 26
exit_group(2)                           = ?
+++ exited with 2 +++
```

and IDP kernel lacks of this patch to add /sys/firmware/efi/fw_platform_size file node:
```
commit 2859dff97e54db4795b8b7d9606cb8efcec722ff
Author: Steve McIntyre <steve@einval.com>
Date:   Fri Jan 9 15:29:53 2015 +0000

    efi: Expose underlying UEFI firmware platform size to userland
    
    In some cases (e.g. Intel Bay Trail machines), the kernel will happily
    run in 64-bit even if the underlying UEFI firmware platform is
    32-bit. That's great, but it's difficult for userland utilities like
    grub-install to do the right thing in such a situation.
    
    The kernel already knows about the size of the firmware via
    efi_enabled(EFI_64BIT). Add an extra sysfs interface
    /sys/firmware/efi/fw_platform_size to expose that information to
    userland for low-level utilities to use.
    
    Signed-off-by: Steve McIntyre <steve@einval.com>
    Cc: Matthew Garrett <mjg59@srcf.ucam.org>
    Signed-off-by: Matt Fleming <matt.fleming@intel.com>
```

I'll backport this patch and see what happens.

So the issue is changed: if fwupdate returns a failure, then fwupd becomes not working any more, unless removing /var/cache/app-info/xmls/fwupd.xml. I think we should fix it.

superm1, 2015-11-12:
Ah good find.

That's because this change happened in fwupdate recently then:

https://github.com/rhinstaller/fwupdate/commit/67bf8448ccf96853a47f0bdd4ed3df42cfd70dfa

I agree need to figure out why this makes fwupd unhappy however. This
should make it easier to reproduce.

superm1, 2015-11-12:
That commit I just made will fix the fwupd crash.  Can you see if that's enough to clean up the problem you encountered?

I downgraded to glib2.0 2.45.8 (I was using 2.46.1 previously) but didn't replicate the problem of the wrong content type.

Can you check what gvfs-info detects the mimetype as?  This is what it shows for me:

```
$ gvfs-info /var/cache/app-info/xmls/fwupd.xml | grep content-type
  standard::content-type: application/xml
  standard::fast-content-type: application/xml
```

hughsie, 2015-11-12:
I agree this looks like a content-type detection problem. I'm pretty sure standard::fast-content-type will be wrong for you; this will also cause other strange errors on your system too as this is expected to work in quite a few different programs.

jiazhang0, 2015-11-13:
@superm1
You patch perfectly resolves my issue. Thanks! Here is the test result:
```
root@WR-IntelligentDevice# fwupdmgr update     
Downloading 775 for UEFI-e0f614ed-fb82-467a-a34e-71172cc07e4d-dev0...
Updating 775 on UEFI-e0f614ed-fb82-467a-a34e-71172cc07e4d-dev0...
 * Decompressing firmware
 * Idle
Retrying as an offline update...
 * Decompressing firmware
 * Scheduling upgrade
 * Idle
UEFI firmware update failed: -1
```
After porting the kernel patch mentioned above, fwupdate now works well.

In addition, I got the different result with gvfs-info.
```
# gvfs-info /var/cache/app-info/xmls/fwupd.xml |grep content
  standard::content-type: text/plain
  standard::fast-content-type: application/octet-stream
```
Do I miss something here?

hughsie, 2015-11-13:
gvfs-info /var/cache/app-info/xmls/fwupd.xml |grep content
    standard::content-type: text/plain

This is wrong; your mime detection is broken. The g_clear_error(&error); isn't a proper fix as it's just papering over the issue; the reason the content check is there is to avoid feeding random binary files into a XML parser.

Richard.

jiazhang0, 2015-11-17:
Thanks for your helps. Close this issue.

closed 2015-11-17 by jiazhang0