fwupd/fwupd #40: docbook2man not available, yet required for make manpages

josephgbr, 2016-02-10:
Currently, 'configure' will tell if docbook2man is not available (e.g. not installed), but 'make' still tries to build man page.

Please make sure that if docbook2man is not found, don't try making manpages, or bail out right after failing in 'configure' if you consider manpage is mandatory, to avoid unnecessary compilation.

Below, you find output logs. Some strings are in Brazilian Portuguese due to the use of JHBuild, but the main text is readable. If not, please let me know.

Output: http://pastie.org/10716819

superm1, 2016-03-22:
I just tried this myself with current master and can't reproduce this problem.

When I run autogen.sh (which calls configure) it will error out if docbook2man isn't available.

```
$ sh autogen.sh 
Copying file po/Makefile.in.in
autoreconf: Entering directory `.'
autoreconf: running: intltoolize --automake --copy --force
autoreconf: running: aclocal  --force -I m4 ${ACLOCAL_FLAGS}
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in `.'.
libtoolize: copying file `./ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
libtoolize: copying file `m4/libtool.m4'
libtoolize: copying file `m4/ltoptions.m4'
libtoolize: copying file `m4/ltsugar.m4'
libtoolize: copying file `m4/ltversion.m4'
libtoolize: copying file `m4/lt~obsolete.m4'
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:98: warning: The 'AM_PROG_MKDIR_P' macro is deprecated, and its use is discouraged.
configure.ac:98: You should use the Autoconf-provided 'AC_PROG_MKDIR_P' macro instead,
configure.ac:98: and use '$(MKDIR_P)' instead of '$(mkdir_p)'in your Makefile.am files.
docs/libdfu/Makefile.am:90: warning: CLEANFILES multiply defined in condition TRUE ...
gtk-doc.make:57: ... 'CLEANFILES' previously defined here
docs/libdfu/Makefile.am:79:   'gtk-doc.make' included from here
autoreconf: Leaving directory `.'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether UID '1001' is supported by ustar format... yes
checking whether GID '1001' is supported by ustar format... yes
checking how to create a ustar tar archive... gnutar
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking how to print strings... printf
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert x86_64-unknown-linux-gnu file names to x86_64-unknown-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-unknown-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
[... 78 lines cut ...]
checking for CFPreferencesCopyAppValue... no
checking for CFLocaleCopyCurrent... no
checking for GNU gettext in libc... yes
checking whether to use NLS... yes
checking where the gettext function comes from... libc
checking whether C compiler accepts -fPIE... yes
checking whether the linker accepts -fPIE -pie... yes
checking whether the linker accepts -Wl,-z,relro,-z,now... yes
checking for cos in -lm... yes
checking for GLIB... yes
checking for GUDEV... yes
checking for POLKIT... yes
checking for GCAB... yes
checking for APPSTREAM_GLIB... yes
checking for GUSB... yes
checking for SQLITE... yes
checking for ARCHIVE... yes
checking for SOUP... yes
checking for docbook2man... no
configure: error: docbook2man program not found
```

This check has been in the code since well before your issue was filed (7447fd16ad5d4f9f0941aba4228864828a702a81)

Can you check if this is still a problem for you?  Are you not running autogen.sh but just configure?

josephgbr, 2016-03-22:
Now I understand what happened. I used GNOME's JHBuild, which fetches the source code and run the whole autogen-configure-make proccess. However, taking a closer look at the output log I provided, I notice it is getting a tarball from version 0.5.1, which surely is an out-dated version. This check was only implemented in 0.5.2.
Anyway, the problem is not here.
Thanks of the help.

closed 2016-03-22 by josephgbr