fwupd/fwupd #42: Failures with Dell XPS 13 9350

danielkza, 2016-02-19:
Tried `fwupd` from the Fedora repos first (0.5.4) then master, same issues.

```
$ sudo  fwupdmgr verify-update
Processing /sys/devices/pci0000:00/0000:00:02.0/rom...
Error reading from file: Input/output error

$ sudo fwupdmgr get-results  UEFI-33773727-8ee7-4d81-9fa0-57e8d889e1fa-dev0
Cannot get UEFI status for 33773727-8ee7-4d81-9fa0-57e8d889e1fa

Usage:
  fwupdmgr [OPTION...]

  clear-results                     Clears the results from the last update
  dump-rom                          Dump the ROM checksum
  get-details                       Gets details about a firmware file
  get-devices                       Get all devices that support firmware updates
  get-results                       Gets the results from the last update
  get-updates                       Gets the list of updates for connected hardware
  install                           Install a firmware file on this hardware
  install-prepared                  Install prepared updates now
  refresh                           Refresh metadata from remote server
  update                            Updates all firmware to latest versions availableate t
  verify                            Gets the cryptographic hash of the dumped firmware
  verify-update                     Update the stored metadata with current ROM contents

Help Options:
  -h, --help            Show help options

Application Options:
  -v, --verbose         Show extra debugging information
  --offline             Perform the installation offline where possible
  --allow-reinstall     Allow re-installing existing firmware versions
  --allow-older         Allow downgrading firmware versions
```

dmesg:
```
[ 1891.148474] i915 0000:00:02.0: Invalid PCI ROM header signature: expecting 0xaa55, got 0xffff
```

Trying to update the firmware seems to work: the upgrade was scheduled and found by firmware on reboot, but installation never progressed. The upgrade screen showed up but progress never moved from 0%, then the machine rebooted with no error message.

superm1, 2016-02-19:
Just to be clear, you are trying UEFI firmware updates right?  Not PCI ROM updates (those shouldn't work for now).  The verify and verify-update shouldn't be used for UEFI updates.

The get-results error you have found has been fixed in 0.6.2 release (and master).  If you try again with get-results you should have better information.

superm1, 2016-02-19:
Also, can you please share #fwupdmgr get-devices after the update (from 0.6.2 or master)?  That would also indicate if you were success filled by the version number contained.

danielkza, 2016-02-20:
Previously I only saw the device I attempted to update (which does seem to be the system firmware) in `get-devices`, but after a reboot I see quite a few more:

```
Device: usb:00:03
  DisplayName:     BCM2045A0
  Provider:        DFU
  Guid:            7a1ba7b9-6bcd-54a4-8a36-d60cc5ee935c
  Version:         1.18
  Internal:        False
  AllowOnline:     True
  AllowOffline:    True
Device: UEFI-33773727-8ee7-4d81-9fa0-57e8d889e1fa-dev0
  Provider:        UEFI
  Guid:            33773727-8ee7-4d81-9fa0-57e8d889e1fa
  Version:         0.1.1.9
  Internal:        True
  AllowOnline:     False
  AllowOffline:    True
Device: usb:00:02:03
  DisplayName:     AX88179
  Provider:        USB
  Guid:            da6082b0-a033-5b25-b280-bcba3720eddd
  Version:         1.0
  Internal:        False
  AllowOnline:     False
  AllowOffline:    False
Device: usb:00:05
  DisplayName:     Integrated_Webcam_HD
  Provider:        USB
  Guid:            4ec19583-c132-5c27-a9cf-061419e838dc
  Version:         86.38
  Internal:        False
  AllowOnline:     False
  AllowOffline:    False
Device: usb:00:04
  DisplayName:     Touchscreen
  Provider:        USB
  Guid:            31cc46f6-9bdc-5b1e-b48b-1c791781dcdc
  Version:         17.17
  Internal:        False
  AllowOnline:     False
  AllowOffline:    False
```

And `get-results` now works correctly. The weird part is that I did have AC power connected, at least apparently, and I saw no message about it during the attempted update. Maybe it was a a glitch on Dell's part, but it will be hard to tell if I don't see it again.

```
Guid           33773727-8ee7-4d81-9fa0-57e8d889e1fa
PendingError   AC power required
PendingState   failed
Provider       UEFI
UpdateVersion  0
Version        0.1.1.9
VersionLowest  0.1.1.9
Flags          5
Created        1455945292
Modified       0
```

I'll do another update attempt and report how it went.

danielkza, 2016-02-20:
Update went fine this time, as smoothly as I could have hoped for. It seems a simple reboot fixed whatever weirdness was going on with Dell's firmware. So it wasn't `fwupd`'s fault at all, maybe except for that small issue with `verify-update`.

Very nice work with this software, and I'll try to keep up with master whenever there are new updates available to try and spot any issues.

closed 2016-02-20 by danielkza

hughsie, 2016-02-20:
So, I'm glad this worked but I think we do have some bugs here. I think the missing failure notification was probably caused by an older version of fwupd without the "get-results" fix from Mario. I'm guessing we should have warned (or held back) the pending firmware update when on battery power.