fwupd/fwupd #43: Test Suite fails on big endian architectures

superm1, 2016-04-01:
Since turning on the test suite in Launchpad builds, it was apparent that there are some assumptions made in the code about endian-ness (signed vs unsigned comparisons and such).  A few warnings in those architectures builds stand out:

```
gcc -DHAVE_CONFIG_H -I. -I..  -pthread -I/usr/include/libappstream-glib -I/usr/include/libgcab-1.0 -I/usr/include/uuid -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng12 -I/usr/include/glib-2.0 -I/usr/lib/powerpc-linux-gnu/glib-2.0/include -pthread -I/usr/include/colord-1/colorhug -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -I/usr/include/colord-1 -I/usr/include/glib-2.0 -I/usr/lib/powerpc-linux-gnu/glib-2.0/include -pthread -I/usr/include/gusb-1 -I/usr/include/glib-2.0 -I/usr/lib/powerpc-linux-gnu/glib-2.0/include -I/usr/include/libusb-1.0 -pthread -I/usr/include/libgcab-1.0 -I/usr/include/glib-2.0 -I/usr/lib/powerpc-linux-gnu/glib-2.0/include -pthread -I/usr/include/gio-unix-2.0/ -I/usr/include/glib-2.0 -I/usr/lib/powerpc-linux-gnu/glib-2.0/include -I/usr/include/gudev-1.0 -I/usr/include/glib-2.0 -I/usr/lib/powerpc-linux-gnu/glib-2.0/include -fPIE -pthread -I/usr/include/polkit-1 -I/usr/include/glib-2.0 -I/usr/lib/powerpc-linux-gnu/glib-2.0/include   -pthread -I/usr/include/libsoup-2.4 -I/usr/include/libxml2 -I/usr/include/glib-2.0 -I/usr/lib/powerpc-linux-gnu/glib-2.0/include  -I.. -I.. -I../libfwupd -DG_LOG_DOMAIN=\"Fu\" -DLIBEXECDIR=\"/usr/lib/powerpc-linux-gnu\" -DLIBDIR=\"/usr/lib/powerpc-linux-gnu\" -DDATADIR=\"/usr/share\" -DSYSCONFDIR=\""/etc"\" -DVERSION="\"0.7.0\"" -DDAEMON_USER="\"\"" -DTESTDATADIR=\""../data/tests"\" -DG_UDEV_API_IS_SUBJECT_TO_CHANGE -DLOCALEDIR=\""/usr/share/locale"\" -Wdate-time -D_FORTIFY_SOURCE=2 -DFU_OFFLINE_DESTDIR=\"\" -DLOCALSTATEDIR=\""/var"\" -Wall -Wcast-align -Wno-uninitialized -Wmissing-declarations -Wpointer-arith -Wcast-align -Wwrite-strings -Winit-self -Wreturn-type -Wformat-nonliteral -Wformat-security -Wmissing-include-dirs -Wmissing-format-attribute -Wclobbered -Wempty-body -Wignored-qualifiers -Wsign-compare -Wtype-limits -Wuninitialized -Waggregate-return -Wdeclaration-after-statement -Wshadow -Wno-strict-aliasing -Winline -Wmissing-parameter-type -Woverride-init -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -c -o fwupdmgr-fu-rom.o `test -f 'fu-rom.c' || echo './'`fu-rom.c
In file included from /usr/include/glib-2.0/glib.h:62:0,
                 from /usr/include/glib-2.0/gobject/gbinding.h:28,
                 from /usr/include/glib-2.0/glib-object.h:23,
                 from ../libfwupd/fwupd-client.h:25,
                 from ../libfwupd/fwupd.h:32,
                 from fu-rom.c:24:
fu-rom.c: In function ‘fu_rom_find_and_blank_serial_numbers’:
fu-rom.c:386:13: warning: format ‘%lx’ expects argument of type ‘long unsigned int’, but argument 5 has type ‘int’ [-Wformat=]
    g_debug ("cleared %i chars @ 0x%04lx", len, tmp - &hdr->rom_data[hdr->data_len]);
             ^
/usr/include/glib-2.0/glib/gmessages.h:170:32: note: in definition of macro ‘g_debug’
                                __VA_ARGS__)
                                ^
fu-rom.c: In function ‘fu_rom_load_file’:
fu-rom.c:712:12: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
  while (sz > hdr_sz + jump) {
            ^
fu-rom.c:785:13: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
  if (hdr_sz > sz) {
```
Full build log for this one (powerpc): https://launchpadlibrarian.net/251158299/buildlog_ubuntu-xenial-powerpc.fwupd_0.7.0-0ubuntu1_BUILDING.txt.gz

```
make  check-TESTS
make[3]: Entering directory '/<<PKGBUILDDIR>>/libfwupd'
make[4]: Entering directory '/<<PKGBUILDDIR>>/libfwupd'
==53583== 
==53583== Process terminating with default action of signal 4 (SIGILL)
==53583==  Illegal opcode at address 0x400B916
==53583==    at 0x400B916: _dl_new_object (dl-object.c:98)
==53583==    by 0x400B8EB: _dl_new_object (dl-object.c:87)
/bin/bash: line 5: 53583 Illegal instruction     libtool --mode=execute valgrind --quiet --leak-check=full --show-possibly-lost=no /bin/bash ../test-driver --test-name "$f" --log-file $b.log --trs-file $b.trs --color-tests "$am__color_tests" --enable-hard-errors "$am__enable_hard_errors" --expect-failure "$am__expect_failure" -- "$tst"
Makefile:1249: recipe for target 'fwupd-self-test.log' failed
```
Full build log for this one (s390x): https://launchpadlibrarian.net/251157893/buildlog_ubuntu-xenial-s390x.fwupd_0.7.0-0ubuntu1_BUILDING.txt.gz

superm1, 2016-04-01:
Here's a more detailed failure from powerpc:

```
FAIL: dfu-self-test
===================

/libdfu/enums: OK
/libdfu/target(DfuSe}: (/«PKGBUILDDIR»/libdfu/.libs/lt-dfu-self-test:15605): libdfu-DEBUG: DfuSe nice alt-name: Flash3
(/«PKGBUILDDIR»/libdfu/.libs/lt-dfu-self-test:15605): libdfu-DEBUG: 
(/«PKGBUILDDIR»/libdfu/.libs/lt-dfu-self-test:15605): libdfu-DEBUG: DfuSe nice alt-name: Internal Flash
(/«PKGBUILDDIR»/libdfu/.libs/lt-dfu-self-test:15605): libdfu-DEBUG: Zone:0, Sec#:0, Addr:0x08000000, Size:0x0400, Caps:0x1
Zone:0, Sec#:0, Addr:0x08000400, Size:0x0400, Caps:0x1
(/«PKGBUILDDIR»/libdfu/.libs/lt-dfu-self-test:15605): libdfu-DEBUG: DfuSe nice alt-name: Flash1
(/«PKGBUILDDIR»/libdfu/.libs/lt-dfu-self-test:15605): libdfu-DEBUG: Zone:0, Sec#:0, Addr:0x08000000, Size:0x0400, Caps:0x1
Zone:0, Sec#:0, Addr:0x08000400, Size:0x0400, Caps:0x1
Zone:0, Sec#:1, Addr:0x08000000, Size:0x0400, Caps:0x7
Zone:0, Sec#:1, Addr:0x08000400, Size:0x0400, Caps:0x7
Zone:0, Sec#:1, Addr:0x08000800, Size:0x0400, Caps:0x7
Zone:0, Sec#:1, Addr:0x08000c00, Size:0x0400, Caps:0x7
(/«PKGBUILDDIR»/libdfu/.libs/lt-dfu-self-test:15605): libdfu-DEBUG: DfuSe nice alt-name: Flash2
(/«PKGBUILDDIR»/libdfu/.libs/lt-dfu-self-test:15605): libdfu-DEBUG: Zone:0, Sec#:0, Addr:0x0000f000, Size:0x0064, Caps:0x1
Zone:0, Sec#:0, Addr:0x0000f064, Size:0x0064, Caps:0x1
Zone:0, Sec#:0, Addr:0x0000f0c8, Size:0x0064, Caps:0x1
Zone:0, Sec#:0, Addr:0x0000f12c, Size:0x0064, Caps:0x1
Zone:1, Sec#:0, Addr:0x0000e000, Size:0x2000, Caps:0x7
Zone:1, Sec#:0, Addr:0x00010000, Size:0x2000, Caps:0x7
Zone:1, Sec#:0, Addr:0x00012000, Size:0x2000, Caps:0x7
Zone:2, Sec#:0, Addr:0x00080000, Size:0x6000, Caps:0x7
Zone:2, Sec#:0, Addr:0x00086000, Size:0x6000, Caps:0x7
(/«PKGBUILDDIR»/libdfu/.libs/lt-dfu-self-test:15605): libdfu-DEBUG: DfuSe nice alt-name: Internal Flash
(/«PKGBUILDDIR»/libdfu/.libs/lt-dfu-self-test:15605): libdfu-DEBUG: DfuSe nice alt-name: Internal Flash
(/«PKGBUILDDIR»/libdfu/.libs/lt-dfu-self-test:15605): libdfu-DEBUG: DfuSe nice alt-name: Flash|XTEA
(/«PKGBUILDDIR»/libdfu/.libs/lt-dfu-self-test:15605): libdfu-DEBUG: 
OK
/libdfu/firmware{raw}: OK
/libdfu/firmware{dfu}: **
libdfu:ERROR:dfu-self-test.c:213:dfu_firmware_dfu_func: assertion failed (error == NULL): CRC failed, expected d4dc86d4, got d486dcd4 (DfuError, 0)
FAIL /«PKGBUILDDIR»/libdfu/.libs/lt-dfu-self-test (exit status: 134)
```

superm1, 2016-04-01:
Here's a more detailed failure from s390x:

```
FAIL: dfu-self-test
===================

/libdfu/enums: OK
/libdfu/target(DfuSe}: (/<<PKGBUILDDIR>>/libdfu/.libs/lt-dfu-self-test:52368): libdfu-DEBUG: DfuSe nice alt-name: Flash3
(/<<PKGBUILDDIR>>/libdfu/.libs/lt-dfu-self-test:52368): libdfu-DEBUG: 
(/<<PKGBUILDDIR>>/libdfu/.libs/lt-dfu-self-test:52368): libdfu-DEBUG: DfuSe nice alt-name: Internal Flash
(/<<PKGBUILDDIR>>/libdfu/.libs/lt-dfu-self-test:52368): libdfu-DEBUG: Zone:0, Sec#:0, Addr:0x08000000, Size:0x0400, Caps:0x1
Zone:0, Sec#:0, Addr:0x08000400, Size:0x0400, Caps:0x1
(/<<PKGBUILDDIR>>/libdfu/.libs/lt-dfu-self-test:52368): libdfu-DEBUG: DfuSe nice alt-name: Flash1
(/<<PKGBUILDDIR>>/libdfu/.libs/lt-dfu-self-test:52368): libdfu-DEBUG: Zone:0, Sec#:0, Addr:0x08000000, Size:0x0400, Caps:0x1
Zone:0, Sec#:0, Addr:0x08000400, Size:0x0400, Caps:0x1
Zone:0, Sec#:1, Addr:0x08000000, Size:0x0400, Caps:0x7
Zone:0, Sec#:1, Addr:0x08000400, Size:0x0400, Caps:0x7
Zone:0, Sec#:1, Addr:0x08000800, Size:0x0400, Caps:0x7
Zone:0, Sec#:1, Addr:0x08000c00, Size:0x0400, Caps:0x7
(/<<PKGBUILDDIR>>/libdfu/.libs/lt-dfu-self-test:52368): libdfu-DEBUG: DfuSe nice alt-name: Flash2
(/<<PKGBUILDDIR>>/libdfu/.libs/lt-dfu-self-test:52368): libdfu-DEBUG: Zone:0, Sec#:0, Addr:0x0000f000, Size:0x0064, Caps:0x1
Zone:0, Sec#:0, Addr:0x0000f064, Size:0x0064, Caps:0x1
Zone:0, Sec#:0, Addr:0x0000f0c8, Size:0x0064, Caps:0x1
Zone:0, Sec#:0, Addr:0x0000f12c, Size:0x0064, Caps:0x1
Zone:1, Sec#:0, Addr:0x0000e000, Size:0x2000, Caps:0x7
Zone:1, Sec#:0, Addr:0x00010000, Size:0x2000, Caps:0x7
Zone:1, Sec#:0, Addr:0x00012000, Size:0x2000, Caps:0x7
Zone:2, Sec#:0, Addr:0x00080000, Size:0x6000, Caps:0x7
Zone:2, Sec#:0, Addr:0x00086000, Size:0x6000, Caps:0x7
(/<<PKGBUILDDIR>>/libdfu/.libs/lt-dfu-self-test:52368): libdfu-DEBUG: DfuSe nice alt-name: Internal Flash
(/<<PKGBUILDDIR>>/libdfu/.libs/lt-dfu-self-test:52368): libdfu-DEBUG: DfuSe nice alt-name: Internal Flash
(/<<PKGBUILDDIR>>/libdfu/.libs/lt-dfu-self-test:52368): libdfu-DEBUG: DfuSe nice alt-name: Flash|XTEA
(/<<PKGBUILDDIR>>/libdfu/.libs/lt-dfu-self-test:52368): libdfu-DEBUG: 
OK
/libdfu/firmware{raw}: OK
/libdfu/firmware{dfu}: **
libdfu:ERROR:dfu-self-test.c:213:dfu_firmware_dfu_func: assertion failed (error == NULL): CRC failed, expected d4dc86d4, got d486dcd4 (DfuError, 0)
FAIL /<<PKGBUILDDIR>>/libdfu/.libs/lt-dfu-self-test (exit status: 134)
```

superm1, 2016-04-02:
The next failure is:
```
/libdfu/firmware{dfu}: **
libdfu:ERROR:dfu-self-test.c:219:dfu_firmware_dfu_func: assertion failed (dfu_firmware_get_size (firmware) == 256): (272 == 256)
```

hughsie, 2016-04-03:
I've fixed up your commit (it was an endian write issue, not read) -- I think I'd prefer to review these sorts of commits before you push them to master if that's okay. We don't have any release pressure now, so we can take things a bit slower.

I'm struggling to work out why the dfu_firmware_get_size() is returning the size of the (firmware+footer) on BE and just (firmware) on LE. Ideas welcome! Also, the dfuse header stuff looks wildly endlan-unsafe, so that'll likely explode too.

hughsie, 2016-04-03:
I've tested master with PPC64BE, but if you can also check on all the targets Debian supports that'd be great; thanks.

closed 2016-04-03 by hughsie

superm1, 2016-04-03:
Thanks, will make sure to push stuff like this to it's own branch before pushing to master in the future.
This fix does resolve all of the architectures that Ubuntu supports.

Can't check in the extra architectures Debian has yet, we're waiting on the maintainer to update libgusb to 0.2.9 there first to be able to build against fwupd 0.7.0.