fwupd/fwupd #46: Dell XPS 13 9350 firmware upgrade fails with "Unknown error -1"

danielkza, 2016-05-09:
I can't upgrade the firmware in my Dell XPS 13 9350 using fwupd 0.5.4-1.fc23. I did previous upgrades successfully (even after I originally had issues in #42), but it doesn't work anymore.

```
$ sudo fwupdmgr update
Downloading 66307 for UEFI-33773727-8ee7-4d81-9fa0-57e8d889e1fa-dev0...
Updating 66307 on UEFI-33773727-8ee7-4d81-9fa0-57e8d889e1fa-dev0...
 * Decompressing firmware
Retrying as an offline update...
 * Idle
 * Decompressing firmware
 * Scheduling upgrade
UEFI firmware update failed: Unknown error -1
```

And this shows up in the system journal:
```
org.freedesktop.fwupd[1640]: fwupd: put_info failed.
org.freedesktop.fwupd[1640]: : Operation not permitted
```

I'm currently running Fedora 23 with kernel 4.6-rc4 from the rawhide-kernel-nodebug repository. I can test it with other kernels (such as newer RCs or stables) if necessary.

superm1, 2016-05-10:
Could you share some more information about the rest of the stack on your system?

- kernel version
- efivar version
- fwupdate version

Your error is coming from here:
https://github.com/rhinstaller/fwupdate/blob/master/linux/libfwup.c#L1137 or here https://github.com/rhinstaller/fwupdate/blob/master/linux/libfwup.c#L1235

My gut feeling is that this is the immutable stuff that happened in the linux kernel.  If you don't have a new enough efivar, fwupdate won't know how to handle it properly.

danielkza, 2016-05-10:
kernel 4.6.0-0.rc4.git0.1.fc25.x86_64
efivar-libs.x86_64 0.21-2.fc23
fwupd.x86_64 0.5.4-1.fc23

What version of efivar-libs is needed? There are no updates for it available in Fedora 23 nor Rawhide.

superm1, 2016-05-10:
0.23, or at least this commit:
https://github.com/rhinstaller/efivar/commit/df78c364a1d362655f5495e781b647553b5ae144

danielkza, 2016-05-10:
I see, I'll ask about it in the fedora-devel list, since it's probably a good idea to get 0.23 in before F24.

danielkza, 2016-05-10:
My bad, it is actually available in Rawhide. I'll update it and see if it fixes the issue.

superm1, 2016-05-10:
OK so you are probably running into a problem that is fixed already in master for fwupdate and efivar because I've verified the full stack in Ubuntu 16.04 works properly, but it's tracking extra commits that aren't in released versions yet.

Does your system have an NVMe disk?  If so, it's likely a mess of efifvar commits for NVMe support.  You should rebuild efivar from master and if it fixes it encourage pjones to release a new version.  These are the most relevant ones:
https://github.com/rhinstaller/efivar/commit/32fa1ac4a6fbd7328e8647c16d2358eace5e3e70
https://github.com/rhinstaller/efivar/commit/d76c86b3d2474c6623f85023d850ef94e9e42115

If your don't have an NVMe disk, need to know what version of 'fwupdate' you have (don't mistake fwupd which you provided details already).  There's been a lot of activity in fwupdate so if it's a snapshot it might need to be updated to a newer one.  

FWIW in Debian and Ubuntu I've verified it working with fwupdate 0.5 and efivar 0.23 + those two NVMe patches on a system with NVMe.

danielkza, 2016-05-16:
I do have an NVMe drive. I'll test with newer versions of fwupd and efivar and report back.

superm1, 2016-05-16:
You are probably better off just backporting the two NVMe fixes I mentioned above otherwise the whole stack has to come backward due to some API changes in efivar (efivar, efibootmgr, fwupdate all need to be updated!).

danielkza, 2016-05-16:
I'll just try to rebuild the 0.23 RPM with the patches then.

superm1, 2016-05-16:
If you still have a problem you should delete your Linux firmware boot
entry from efibootmgr. I forget if it will automatically recreate it or
assume old one works when it finds one.

danielkza, 2016-05-18:
@superm1 It works! I compiled efivar 0.23 with the two patches you mentioned and deleted the old boot entry, and the update ran fine (a new entry was created automatically). Thanks for pointing me in the right direction.

closed 2016-05-18 by danielkza

superm1, 2016-05-18:
@danielkza Great, glad to hear it worked.  Can you raise this in the appropriate development lists to get these patches put in in case efivar 0.24 isn't released soon?

danielkza, 2016-05-18:
Sure. in the meantime I may get the patched packages in a COPR repo for anyone that has the same trouble.

stephanmol, 2016-06-10:
@danielkza I'm facing this exact same issue. I tried searching for your (or any updated) efivar package in COPR but I find none. Could you please upload it so I can update my 9350 FW too? Or better even, I'm curious to know how I go about getting those patches in myself.. if it's not too asked too much :-)

danielkza, 2016-06-10:
@stephanmol I cherry-picked the commits mentioned by @superm1 and
recompiled the 0.23 package. I'll get you the SRPM when I get to my home PC.

danielkza, 2016-06-13:
@stephanmol Sorry for the delay. I created a COPR with my patched packages. It's still building, but should be done shortly. https://copr.fedorainfracloud.org/coprs/danielkza/efivar-f23

superm1, 2016-06-13:
Daniel,

Do you know the right way to get those packages out to the affected stable
fedora releases? I think it would be best to distro patch for now like
Ubuntu and Debian both did.

danielkza, 2016-06-13:
@superm1 You're absolutely correct, it's much better to get this fixed in the distro packages. The COPR is just a stop-gap measure if anyone needs it like I did. I'll raise the issue on the Fedora mailing list or the RH bugtracker.

danielkza, 2016-06-13:
Reported at https://bugzilla.redhat.com/show_bug.cgi?id=1346030.

stephanmol, 2016-06-14:
It worked @danielkza. Thanks a lot!