fwupd/fwupd #48: Unpluging USB devices causes GetUpdates to get called

superm1, 2016-05-20:
With gnome-software's session service running, I've noticed that when I remove a USB device there is a popup saying that updates are available.

If I check fwupd -v output I can see that GetUpdates() gets run shortly after the USB device is removed.  If I kill the gnome-software session service, the GetUpdates() call doesn't happen.

This seems to be caused by some event that fwupd emits that gnome-software is reacting to.

This is with fwupd git master and the gnome-software in ubuntu 16.04 (3.20.1 git hash a976144).

hughsie, 2016-05-20:
Yes, this is due to an over-eager invalidation of the updates list in the fwupd plugin. I actually fixed this a few days ago in gnome-software: https://git.gnome.org/browse/gnome-software/commit/src/plugins/gs-plugin-fwupd.c?id=a39c27561f47e190fa8eb4164570c55f84928692 although you also need >= fwupd 0.7.1 to get the new flag.

superm1, 2016-05-20:
OK, that would explain it.  Since you've got version checks on the fwupd version compiled against, could you also backport that fix to 3.20.x?

I'll close up this issue.

closed 2016-05-20 by superm1

hughsie, 2016-05-23:
gnome-3-20 isn't using FwupdClient so it's not possible to cherry pick; sorry.