fwupd/fwupd #50: Reports of USB Audio failing when fwupd running

superm1, 2016-05-29:
There have been a variety of reports of USB audio failing to work properly when booted with it plugged in.  fwupd has a spin lock on the CPU in these situations, and replugging the device seems to resolve the issue.

Devices that the problem has been reported so far:
```
Bus 001 Device 006: ID 1235:801c Focusrite-Novation
Bus 001 Device 003: ID 1235:8006 Focusrite-Novation Focusrite Scarlett 2i2
```

Additional details about one device ```lsusb -v```
```
Bus 001 Device 004: ID 1235:801c Focusrite-Novation 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1235 Focusrite-Novation
  idProduct          0x801c 
  bcdDevice            2.80
  iManufacturer           1 (error)
  iProduct                2 (error)
  iSerial                 0 
  bNumConfigurations      2
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          287
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         3
      bFunctionClass          1 Audio
      bFunctionSubClass       0 
      bFunctionProtocol      32 
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol     32 
      iInterface              2 (error)
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               2.00
        bCategory               8
        wTotalLength          119
        bmControl            0x00
      AudioControl Interface Descriptor:
        bLength                 8
        bDescriptorType        36
        bDescriptorSubtype     10 (CLOCK_SOURCE)
        bClockID               41
        bmAttributes         0x03 Internal programmable Clock 
        bmControls           0x07
[... 507 lines cut ...]
      Device Firmware Upgrade Interface Descriptor:
        bLength                             9
        bDescriptorType                    33
        bmAttributes                        7
          Will Not Detach
          Manifestation Tolerant
          Upload Supported
          Download Supported
        wDetachTimeout                    250 milliseconds
        wTransferSize                      64 bytes
        bcdDFUVersion                   1.10
Device Status:     0x4c90
  (Bus Powered)
```

Additional details are in the bug report link https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1574079.  Unfortunately they're pretty scattered with everyone giving varying amounts of information.  Something that did seem potentially common (and interesting?)  was that there is a DFU end point on some of these devices.  furthermore one person noted in their debug log that there was a DFU device emitted.
```
...
21:26:16	15	com.hughski.ColorHugALS.firmware	ColorHugALS
21:26:30	waiting a small time for other providers
21:26:30	using b69159f2-3778-5728-80d0-01be18b2ea30 for USB\VID_1397&PID_0501
21:26:30	emit added from DFU: usb:00:01:03
21:26:31	no product string descriptor
21:26:33	Called GetUpdates()
```

hughsie, 2016-05-29:
I've added a few requests for info on the lp bug. TL;DR: I need a backtrace for the fatal warning or to find out what's spinning, or a device itself.

hughsie, 2016-05-30:
So, known devices that fail:

* Sony PHA-3
* RME Fireface UCX
* Focusrite 2i2
* Focusrite solo
* Cambridge audio dacmagic plus (x2)
* Audiobox 22 VSL
* PreSonus AudioBox 1818 VSL
* BEHRINGER UMC204

mackeyja92, 2016-05-31:
@hughsie, confirmed an issue on Focusrite 2i4 as well.

bosiakov, 2016-05-31:
I can confirm this problem. System: Ubuntu 16.04. Card: Focusrite 2i2.

Running sudo killall fwupd and unplugging and plugging device back helps.

But `fwupdmgr get-devices` destroys all.

hughsie, 2016-05-31:
@mackeyja92 If you kill fwupd, then run `sudo dfu-tool watch` then replug the device, do you see the same symptoms?

TarasZakus, 2016-05-31:
Confirmed. Running `sudo dfu-tool watch` causes the same symptoms. Here is the output:

> $ sudo dfu-tool watch -v
(dfu-tool:4264): libdfu-DEBUG: device was not DFU capable
(dfu-tool:4264): libdfu-DEBUG: device was not DFU capable
(dfu-tool:4264): libdfu-DEBUG: device was not DFU capable
(dfu-tool:4264): libdfu-DEBUG: device was not DFU capable
(dfu-tool:4264): libdfu-DEBUG: device was not DFU capable
(dfu-tool:4264): libdfu-DEBUG: device was not DFU capable
(dfu-tool:4264): libdfu-DEBUG: device was not DFU capable
(dfu-tool:4264): libdfu-DEBUG: device was not DFU capable
(dfu-tool:4264): libdfu-DEBUG: basic DFU, no DfuSe support
(dfu-tool:4264): libdfu-DEBUG: refreshed status=OK and state=appIDLE
Added:          1397:0501 [appIDLE:OK]
(dfu-tool:4264): libdfu-DEBUG: device 1397:0501 [usb:00:01:03] was added
(dfu-tool:4264): libdfu-DEBUG: invalidating backing GUsbDevice
(dfu-tool:4264): libdfu-DEBUG: usb:00:01:03 did not come back as a DFU capable device
(dfu-tool:4264): libdfu-DEBUG: usb:00:01:03 was removed
Removed:        1397:0501 [removed]
^C(dfu-tool:4264): libdfu-DEBUG: Handling SIGINT
Cancelled

hughsie, 2016-05-31:
@TarasZakus thanks for the confirmation. That means we know it's not an issue with having multiple GUsbContext's or with the way fwupd is querying the device. The next thing we need to do is work out why using the DFU runtime interface is broken :)

Could someone upload the output of `sudo strace dfu-tool watch -v` please, but only for when it fails. Thanks!

TarasZakus, 2016-05-31:
Ok, so here what happens, when I connecting the device when dfu-tool "watching":

> )    = 1 ([{fd=4, revents=POLLIN}])
read(4, "\1\0\0\0\0\0\0\0", 16)         = 8
open("/sys/bus/usb/devices/2-1.3/bConfigurationValue", O_RDONLY) = 11
read(11, "1\n", 5)                      = 2
close(11)                               = 0
open("/sys/bus/usb/devices/2-1.3/bConfigurationValue", O_RDONLY) = 11
read(11, "1\n", 5)                      = 2
close(11)                               = 0
futex(0x7f92ba03e428, FUTEX_WAKE_PRIVATE, 2147483647) = 0
write(1, "(dfu-tool:3051): libdfu-DEBUG: b"..., 59(dfu-tool:3051): libdfu-DEBUG: basic DFU, no DfuSe support
) = 59
open("/dev/bus/usb/002/008", O_RDWR)    = 11
ioctl(11, USBDEVFS_GET_CAPABILITIES, 0x17ff110) = 0
write(9, "\1", 1)                       = 1
futex(0x17d5990, FUTEX_WAKE_PRIVATE, 1) = 1
ioctl(11, USBDEVFS_CLAIMINTERFACE, 0x7fff46553d0c) = 0
timerfd_settime(10, TFD_TIMER_ABSTIME, {it_interval={0, 0}, it_value={233, 401618000}}, NULL) = 0
ioctl(11, USBDEVFS_SUBMITURB, 0x17fc710) = 0
futex(0x17d5964, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 1, {1464726515, 9042770}, ffffffff) = 0
futex(0x17d5938, FUTEX_WAKE_PRIVATE, 1) = 0
timerfd_settime(10, TFD_TIMER_ABSTIME, {it_interval={0, 0}, it_value={233, 401927000}}, NULL) = 0
ioctl(11, USBDEVFS_SUBMITURB, 0x17fc710) = 0
futex(0x17d5964, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 3, {1464726515, 9330992}, ffffffff) = 0
futex(0x17d5938, FUTEX_WAKE_PRIVATE, 1) = 0
futex(0x7f92ba03e428, FUTEX_WAKE_PRIVATE, 2147483647) = 0
write(3, "\1\0\0\0\0\0\0\0", 8)         = 8
futex(0x17cfc00, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x7f92ba03e428, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x7f92ba03e428, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x7f92ba03e428, FUTEX_WAKE_PRIVATE, 2147483647) = 0
open("/usr/share/locale/en_US/LC_MESSAGES/glib20.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
open("/usr/share/locale/en/LC_MESSAGES/glib20.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
open("/usr/share/locale-langpack/en_US/LC_MESSAGES/glib20.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
open("/usr/share/locale-langpack/en/LC_MESSAGES/glib20.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
timerfd_settime(10, TFD_TIMER_ABSTIME, {it_interval={0, 0}, it_value={232, 903018000}}, NULL) = 0
ioctl(11, USBDEVFS_SUBMITURB, 0x17fef20) = 0
poll([{fd=4, events=POLLIN}], 1, -1)    = 1 ([{fd=4, revents=POLLIN}])
read(4, "\1\0\0\0\0\0\0\0", 16)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
futex(0x17cfeb0, FUTEX_WAKE_PRIVATE, 2147483647) = 0
write(1, "(dfu-tool:3051): libdfu-DEBUG: r"..., 69(dfu-tool:3051): libdfu-DEBUG: refreshed status=OK and state=appIDLE
) = 69
write(9, "\1", 1)                       = 1
futex(0x17d5990, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x17d5964, FUTEX_CMP_REQUEUE_PRIVATE, 1, 2147483647, 0x17d5938, 6) = 1
futex(0x17d5938, FUTEX_WAKE_PRIVATE, 1) = 1
close(11)                               = 0
futex(0x17d5964, FUTEX_CMP_REQUEUE_PRIVATE, 1, 2147483647, 0x17d5938, 8) = 1
futex(0x17d5938, FUTEX_WAKE_PRIVATE, 1) = 1
write(1, "Added:", 6Added:)                   = 6
write(1, " ", 1 )                        = 1
write(1, " ", 1 )                        = 1
write(1, " ", 1 )                        = 1
write(1, " ", 1 )                        = 1
write(1, " ", 1 )                        = 1
write(1, " ", 1 )                        = 1
write(1, " ", 1 )                        = 1
write(1, " ", 1 )                        = 1
write(1, " ", 1 )                        = 1
write(1, " ", 1 )                        = 1
write(1, "1397:0501 [appIDLE:OK]\n", 231397:0501 [appIDLE:OK]
) = 23
write(1, "(dfu-tool:3051): libdfu-DEBUG: d"..., 73(dfu-tool:3051): libdfu-DEBUG: device 1397:0501 [usb:00:01:03] was added
) = 73
poll([{fd=4, events=POLLIN}], 1, -1)    = 1 ([{fd=4, revents=POLLIN}])
read(4, "\1\0\0\0\0\0\0\0", 16)         = 8
poll([{fd=4, events=POLLIN}], 1, -1

hughsie, 2016-06-01:
So, that looks like the device (endpoint?) -- one thing to try would be to stop opening the device and see if it's that. If anyone is feeling clever, can you edit fwupd/libdfu/dfu-tool.c, find dfu_tool_get_device_string() and change:

    }
    if (!dfu_device_open (device,

to:

    }
    return g_strdup ("test");
    if (!dfu_device_open (device,

And then compile, then run the new tool with `.sudo /dfu-tool watch --verbose` and try to reproduce. If it still happens this points to something lower down the stack, if it fixes it we can look at another part of DFU. Note: I still think this is a device quirk; but I do acknowledge we have to handle this better.

TarasZakus, 2016-06-01:
@hughsie Yes, you are right! :+1: The problem is somewhere inside `dfu_device_open`-function. If we place return statement before it - the sound-card is detected correctly, after it - the same problem arises.
Unfortunately, I have no more time today to "go deeper". It will be helpful if you give me some tips where to search this next part of DFU.

hughsie, 2016-06-02:
@TarasZakus thanks for doing that; could you try the `wip/hughsie/dfu-no-interface` branch of fwupd please -- the `dfu-tool watch` issue should be gone, and if you can get the daemon working it should fix that too. Yell if you need any help. Thanks.

TarasZakus, 2016-06-02:
Unfortunately, the issue still exists. I've tested on `dfu-no-interface`-branch as well as on fwupd version **0.7.0-0ubuntu4.2~lp1574079.2** installed from `ppa:superm1/lp1574079`.

TarasZakus, 2016-06-05:
I just did some investigation. It turns out that problem lays a little deeper, than I was expecting. Here is some kinda call-stack:

_github.com:hughsie/fwupd.git_
starting from `dfu_device_open` function [here](https://github.com/hughsie/fwupd/blob/wip/hughsie/dfu-no-interface/libdfu/dfu-device.c#L1232)
-> calls `dfu_device_refresh` [here](https://github.com/hughsie/fwupd/blob/wip/hughsie/dfu-no-interface/libdfu/dfu-device.c#L1308)
-> calls `g_usb_device_control_transfer` [here](https://github.com/hughsie/fwupd/blob/wip/hughsie/dfu-no-interface/libdfu/dfu-device.c#L947)

uses _github.com:hughsie/libgusb_
starting from `g_usb_device_control_transfer` [here](https://github.com/hughsie/libgusb/blob/gusb_0_2_9/gusb/gusb-device.c#L935)
-> calls `g_usb_device_control_transfer_async` [here](https://github.com/hughsie/libgusb/blob/gusb_0_2_9/gusb/gusb-device.c#L957)
-> calls `libusb_submit_transfer` [here](https://github.com/hughsie/libgusb/blob/gusb_0_2_9/gusb/gusb-device.c#L1301)

uses _github.com:libusb/libusb_
starting from `libusb_submit_transfer` [here](https://github.com/libusb/libusb/blob/v1.0.20/libusb/io.c#L1459)

I assume that the bug might be somewhere between calling `g_usb_device_control_transfer` and `libusb_submit_transfer`.
Maybe some inappropriate argument passed or so.

mackeyja92, 2016-06-05:
Once this is resolved, how long does it generally take to get propagated down into the package managers for the various distros?

superm1, 2016-06-05:
Distribution policies will vary on this matter. For example Ubuntu's
policies mandate a 7 day aging period of the fix in the proposed repository.

https://wiki.ubuntu.com/StableReleaseUpdates

closed 2016-06-08 by hughsie

tomrbland, 2016-06-17:
Stupid Question time: What's the recommended way of me applying this fix so that automatic updates still happen in future etc? Am I just best waiting until it trickles through the system? apt-cashe policy fwupd: 0.7.0-0ubuntu4.1

superm1, 2016-06-17:
Since you are on Ubuntu, apply the fix from the proposed repository. Future
fixes will come through the updates repository automatically.

tomrbland, 2016-06-17:
Ok, that took a bit of learning (I don't know much about how software gets in to repos). For anyone else needing to know - this is the information you're looking for: [How to enable proposed](https://wiki.ubuntu.com/Testing/EnableProposed)

ipuustin, 2017-08-02:
I have the problem (or something very like it with lots of `usb_set_interface failed (-71)` messages) with Focusrite Solo still in Fedora 26, running fwupd 0.9.5. I'm disabling `fwupd` for now to see if the problem goes away...