fwupd/fwupd #54: Dell XPS 15 9550 EFI scheduled firmware update never occurs

jhoblitt, 2016-06-29:
I have a Dell 9550 running F24.  Much to my surprise, an EFI update poped up via the gnome software manager.  Clicking reboot&install doesn't seem to do anything.

```
$ fwupdmgr get-updates
XPS 15 9550 has firmware updates:
ID:                      com.dell.uefi124c207d.firmware
GUID:                    124c207d-5db8-4d95-bd31-34fd971b34f9
Update Version:          0.1.2.0
Update Checksum:         4902dcbeed2b1d3cf1e9fd14a4b1aedd5ce00a4f
Update Checksum Type:    sha1
Update Location:         https://secure-lvfs.rhcloud.com/downloads/ec95a09f68afa49e02420e59bb8889af80a709a8-firmware_1.2.0.wu.cab
$ sudo fwupdmgr update
Downloading 0.1.2.0 for XPS 15 9550...
Updating 0.1.2.0 on XPS 15 9550...
 * Decompressing firmware
Retrying as an offline update...
 * Idle
 * Decompressing firmware
 * Scheduling upgrade
```

And rebooting doesn't seem to do anything either.  I manually installed the fwupdate/fwupdate-efi packages with no change in behavior.  Do I need to manually install some other package(s)? Is EFI supposed to pick up an alternative boot-loader for the update or is it something that happens from the initrd?

superm1, 2016-06-29:
Normally the flow is supposed to be:

1. Boot entry is created for Linux Firmware Updater
2. BootNext is set by fwupdate.
3. Payload is staged on the ESP in /boot/efi/efi/$distro/fw/$file
4. Metadata is staged in EFI NVRAM
5. Reboot the box
6. BootNext is run - shimx64.efi loads
7. shimx64.efi loads fwupx64.efi
8. fwupx64.efi will process metadata from NVRAM and call UpdateCapsule on the capsule file
9. BIOS takes over

We have to figure out where your problem is, which will require a bunch more information.

Can you share the following details:

Versions of:
- fwupdate
- fwupd
- efivar
- efibootmgr

Output of
```# efibootmgr -v```

Other questions:
- Are you using an NVMe disk in the machine?
- Do you have secure boot turned on?
- If so - can you confirm if there is a signed version of fwupx64.efi on your EFI system partition?
- Is the capsule staged in the ESP in that directory I listed above?
- Is the metadata created in /sys/firmware/efi/efivars?  You can check by looking for fwupdate-* in there.

Hopefully with all those details we can identify what's happening.

@hughsie 
I know we talked about catching common problems in the provider code.  I haven't gotten around to that myself, but in the interim I think it would be useful to create some sort of helper script for people to run to at least capture all these details that are needed for debugging better.  What do you think?

jhoblitt, 2016-06-29:
I am fairly clueless about EFI... please excuse my bumbling.

```
$ rpm -qa fwupdate fwupd efivar efivar-libs efibootmgr
efibootmgr-0.12-3.fc24.x86_64
fwupd-0.7.1-1.fc24.x86_64
efivar-libs-0.23-1.fc24.x86_64
fwupdate-0.5-4.fc24.x86_64
$ # note that efivar is not installed  but efivar-libs is
$ efibootmgr -v
BootNext: 0004
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0003,0002
Boot0000  Windows Boot Manager	HD(1,GPT,bf4d21f3-d8bf-4d28-b39a-90d8fb5728e6,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* Fedora	HD(1,GPT,ba79953c-d5b8-4598-9f25-13c79ed61ebf,0x800,0x80000)/File(\EFI\fedora\shim.efi)
Boot0002* UEFI: PM951 NVMe SAMSUNG 512GB, Partition 1	HD(1,GPT,ba79953c-d5b8-4598-9f25-13c79ed61ebf,0x800,0x80000)/File(EFI\boot\bootx64.efi)..BO
Boot0003* fallback.efi	PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-25-38-41-61-B4-97-7B)/HD(1,GPT,ba79953c-d5b8-4598-9f25-13c79ed61ebf,0x800,0x80000)/File(\EFI\BOOT\fallback.efi)
Boot0004* Linux-Firmware-Updater \fwupx64.efi	HD(1,MBR,0x0,0x0,0x0)/File(\EFI\fedora\shim.efi)\.f.w.u.p.x.6.4...e.f.i...
```
>     Are you using an NVMe disk in the machine?

Yes.  The efi partition is `nvme0n1p1`.

```
$ cat /proc/partitions 
major minor  #blocks  name

 259        0  500107608 nvme0n1
 259        1     262144 nvme0n1p1
 259        2     524288 nvme0n1p2
 259        3  300326912 nvme0n1p3
 259        4   33562624 nvme0n1p4
 253        0     118784 dm-0
 253        1  243269632 dm-1
 253        2  243269632 dm-2
 253        3   33554432 dm-3
 253        4   33558528 dm-4
 253        5   33556480 dm-5
 253        6  243269632 dm-6
 253        7  209715200 dm-7
 253        8     303104 dm-8
 253        9   22601728 dm-9
 253       10   22601728 dm-10
```
>     Do you have secure boot turned on?

It is off.  I'm not sure how to check that from the cli.  I am responding from the machine in question -- I will verify shortly.

>     If so - can you confirm if there is a signed version of fwupx64.efi on your EFI system partition?

SB is off, so it shouldn't matter, but I'm not sure how to check the signature.  Google suggests that there  is a `sbverify` util but it doesn't seem to be packaged for fedora.
```
$ sudo dnf whatprovides \*sbverify\*
Last metadata expiration check: 1:20:58 ago on Wed Jun 29 13:37:29 2016.
Error: No Matches found
$ sudo file /boot/efi/EFI/fedora/fwupx64.efi
/boot/efi/EFI/fedora/fwupx64.efi: PE32+ executable (EFI application) x86-64 (stripped to external PDB), for MS Windows
```
>     Is the capsule staged in the ESP in that directory I listed above?

I believe so:

```
$ sudo tree /boot/efi/
/boot/efi/
├── EFI
│   ├── BOOT
│   │   ├── BOOTX64.EFI
│   │   └── fallback.efi
│   └── fedora
│       ├── BOOT.CSV
│       ├── fonts
│       │   └── unicode.pf2
│       ├── fw
│       │   └── fwupdate-u4rCPf.cap
│       ├── fwupx64.efi
│       ├── gcdx64.efi
│       ├── grub.cfg
│       ├── grubenv
│       ├── grubx64.efi
│       ├── MokManager.efi
│       ├── shim.efi
│       └── shim-fedora.efi
├── mach_kernel
└── System
    └── Library
        └── CoreServices
            └── SystemVersion.plist

8 directories, 15 files
```

>     Is the metadata created in /sys/firmware/efi/efivars? You can check by looking for fwupdate-* in there.

```
$ ls /sys/firmware/efi/efivars | grep -i fwup
fwupdate-124c207d-5db8-4d95-bd31-34fd971b34f9-0-0abba7dc-e516-4167-bbf5-4d9d1c739416
```

jhoblitt, 2016-06-29:
I am fairly clueless about EFI... please excuse my bumbling.

```
$ rpm -qa fwupdate fwupd efivar efivar-libs efibootmgr
efibootmgr-0.12-3.fc24.x86_64
fwupd-0.7.1-1.fc24.x86_64
efivar-libs-0.23-1.fc24.x86_64
fwupdate-0.5-4.fc24.x86_64
$ # note that efivar is not installed  but efivar-libs is
$ efibootmgr -v
BootNext: 0004
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0003,0002
Boot0000  Windows Boot Manager	HD(1,GPT,bf4d21f3-d8bf-4d28-b39a-90d8fb5728e6,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* Fedora	HD(1,GPT,ba79953c-d5b8-4598-9f25-13c79ed61ebf,0x800,0x80000)/File(\EFI\fedora\shim.efi)
Boot0002* UEFI: PM951 NVMe SAMSUNG 512GB, Partition 1	HD(1,GPT,ba79953c-d5b8-4598-9f25-13c79ed61ebf,0x800,0x80000)/File(EFI\boot\bootx64.efi)..BO
Boot0003* fallback.efi	PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-25-38-41-61-B4-97-7B)/HD(1,GPT,ba79953c-d5b8-4598-9f25-13c79ed61ebf,0x800,0x80000)/File(\EFI\BOOT\fallback.efi)
Boot0004* Linux-Firmware-Updater \fwupx64.efi	HD(1,MBR,0x0,0x0,0x0)/File(\EFI\fedora\shim.efi)\.f.w.u.p.x.6.4...e.f.i...
```
>     Are you using an NVMe disk in the machine?

Yes.  The efi partition is `nvme0n1p1`.

```
$ cat /proc/partitions 
major minor  #blocks  name

 259        0  500107608 nvme0n1
 259        1     262144 nvme0n1p1
 259        2     524288 nvme0n1p2
 259        3  300326912 nvme0n1p3
 259        4   33562624 nvme0n1p4
 253        0     118784 dm-0
 253        1  243269632 dm-1
 253        2  243269632 dm-2
 253        3   33554432 dm-3
 253        4   33558528 dm-4
 253        5   33556480 dm-5
 253        6  243269632 dm-6
 253        7  209715200 dm-7
 253        8     303104 dm-8
 253        9   22601728 dm-9
 253       10   22601728 dm-10
```
>     Do you have secure boot turned on?

It is off.  I'm not sure how to check that from the cli -- it doesn't seem to be in the DMI data.

https://www.dropbox.com/s/1x36jjfqcp32kc7/2016-06-29%2015.05.18.jpg?dl=0

>     If so - can you confirm if there is a signed version of fwupx64.efi on your EFI system partition?

SB is off, so it shouldn't matter, but I'm not sure how to check the signature.  Google suggests that there  is a `sbverify` util but it doesn't seem to be packaged for fedora.
```
$ sudo dnf whatprovides \*sbverify\*
Last metadata expiration check: 1:20:58 ago on Wed Jun 29 13:37:29 2016.
Error: No Matches found
$ sudo file /boot/efi/EFI/fedora/fwupx64.efi
/boot/efi/EFI/fedora/fwupx64.efi: PE32+ executable (EFI application) x86-64 (stripped to external PDB), for MS Windows
```
>     Is the capsule staged in the ESP in that directory I listed above?

I believe so:

```
$ sudo tree /boot/efi/
/boot/efi/
├── EFI
│   ├── BOOT
│   │   ├── BOOTX64.EFI
│   │   └── fallback.efi
│   └── fedora
│       ├── BOOT.CSV
│       ├── fonts
│       │   └── unicode.pf2
│       ├── fw
│       │   └── fwupdate-u4rCPf.cap
│       ├── fwupx64.efi
│       ├── gcdx64.efi
│       ├── grub.cfg
│       ├── grubenv
│       ├── grubx64.efi
│       ├── MokManager.efi
│       ├── shim.efi
│       └── shim-fedora.efi
├── mach_kernel
└── System
    └── Library
        └── CoreServices
            └── SystemVersion.plist

8 directories, 15 files
```

>     Is the metadata created in /sys/firmware/efi/efivars? You can check by looking for fwupdate-* in there.

```
$ ls /sys/firmware/efi/efivars | grep -i fwup
fwupdate-124c207d-5db8-4d95-bd31-34fd971b34f9-0-0abba7dc-e516-4167-bbf5-4d9d1c739416
```

jhoblitt, 2016-06-29:
SB is confirmed as off:

https://www.dropbox.com/s/1x36jjfqcp32kc7/2016-06-29%2015.05.18.jpg?dl=0

One other possibly bit of pertinent information is that this system will hang forever on reboot when the brcm* drivers for the wifi nic are loaded.  `reboot -f` or the atx power button 5s timeout is required.  I did try booting the system with the recovery initrd to schedule the update.  It rebooted clean but didn't seem to install the firmware.

superm1, 2016-06-30:
Thanks for sharing all those details.

OK so your issue is likely the same thing as the issue that was in
https://github.com/hughsie/fwupd/issues/46

Look at your Boot0004 you can see that it's got an invalid path.

Boot0004* Linux-Firmware-Updater \fwupx64.efi
HD(1,MBR,0x0,0x0,0x0)/File(\EFI\fedora\shim.efi)\.f.w.u.p.x.6.4...e.f.i...

You're missing these two patches:
https://github.com/rhinstaller/efivar/commit/32fa1ac4a6fbd7328e8647c16d2358eace5e3e70
https://github.com/rhinstaller/efivar/commit/d76c86b3d2474c6623f85023d850ef94e9e42115

Which resolve issues with NVMe drives.

This has been reported here:
https://bugzilla.redhat.com/show_bug.cgi?id=1346030 against fedora 23.
You should probably file a note there that fedora 24 is also affected.

In the interim,
@danielkza had created a copr repository for f23 with the patches
backported but hasn't done this for f24.
You can try to run his f23 rpms, will need to rebuild for f24 or wait for
that rbz to be fixed.

danielkza, 2016-06-30:
@superm1 @jhoblitt I can rebuild the COPR for F24, should work just the same. I have upgrade to F24 myself, I just forgot to do it.

jhoblitt, 2016-06-30:
@danielkza Fantastic. That certainly saves me some trouble.

jhoblitt, 2016-06-30:
@superm1 Sorry for the noise on a resolved issue...  I will chime in on rh bz just as soon as the forced password reset email shows up (waiting 10 mins now...).

danielkza, 2016-06-30:
It seems COPR is having some trouble right now, I'm getting timeouts and server errors. Bad luck! I'll update you whenever I manage to get the build through.

danielkza, 2016-06-30:
Fortunately that didn't take long, build is up for F24: https://copr.fedorainfracloud.org/coprs/danielkza/efivar-f23/

jhoblitt, 2016-06-30:
Testing the copr build right now.

jhoblitt, 2016-06-30:
Hmm.  No luck.  Do I need to mess with the efi entries?

    sudo dnf copr enable -y danielkza/efivar-f23
    sudo dnf update -y efivar
    sudo fwupdmgr update
    sudo reboot -f
```sh
$ rpm -qa efivar efivar-libs
efivar-0.23-2.fc24.x86_64
efivar-libs-0.23-2.fc24.x86_64
$ sudo efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0003,0002
Boot0000  Windows Boot ManagerHD(1,GPT,bf4d21f3-d   8bf-4d28-b39a-90d8fb5728e6,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* FedoraHD(1,GPT,ba79953c-d5b8-4598-9f  25-13c79ed61ebf,0x800,0x80000)/File(\EFI\fedora\shim.efi)
Boot0002* UEFI: PM951 NVMe SAMSUNG 512GB, Partition 1HD(1,GPT,ba79953c-d5b8-4598-9f25-13c79ed   61ebf,0x800,0x80000)/File(EFI\boot\bootx64.efi)..BO
Boot0003* fallback.efiPciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-25-38-41-61-B4-97-7B)/HD(1,G  PT,ba79953c-d5b8-4598-9f25-13c79ed61ebf,0x800,0x80000)/File(\EFI\BOOT\fallback.efi)
Boot0004* Linux-Firmware-Updater \fwupx64.efiHD(1,MBR,0x0,0x0,0x0)/File(\EFI\fedora \shim.efi)\.f.w.u.p.x.6.4...e.f.i...
```

danielkza, 2016-06-30:
Try removing the updater entry and starting a new update.

jhoblitt, 2016-06-30:
This looks promising:

```sh
[root@dakine ~]# efibootmgr -B -b 004
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0003,0002
Boot0000  Windows Boot Manager
Boot0001* Fedora
Boot0002* UEFI: PM951 NVMe SAMSUNG 512GB, Partition 1
Boot0003* fallback.efi
[root@dakine ~]# efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0003,0002
Boot0000  Windows Boot Manager	HD(1,GPT,bf4d21f3-d8bf-4d28-b39a-90d8fb5728e6,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* Fedora	HD(1,GPT,ba79953c-d5b8-4598-9f25-13c79ed61ebf,0x800,0x80000)/File(\EFI\fedora\shim.efi)
Boot0002* UEFI: PM951 NVMe SAMSUNG 512GB, Partition 1	HD(1,GPT,ba79953c-d5b8-4598-9f25-13c79ed61ebf,0x800,0x80000)/File(EFI\boot\bootx64.efi)..BO
Boot0003* fallback.efi	PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-25-38-41-61-B4-97-7B)/HD(1,GPT,ba79953c-d5b8-4598-9f25-13c79ed61ebf,0x800,0x80000)/File(\EFI\BOOT\fallback.efi)
[root@dakine ~]# fwupdmgr update
Downloading 0.1.2.0 for XPS 15 9550...
Updating 0.1.2.0 on XPS 15 9550...
 * Decompressing firmware
Retrying as an offline update...
 * Idle
 * Decompressing firmware
 * Scheduling upgrade
[root@dakine ~]# efibootmgr -v
BootNext: 0004
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0003,0002
Boot0000  Windows Boot Manager	HD(1,GPT,bf4d21f3-d8bf-4d28-b39a-90d8fb5728e6,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* Fedora	HD(1,GPT,ba79953c-d5b8-4598-9f25-13c79ed61ebf,0x800,0x80000)/File(\EFI\fedora\shim.efi)
Boot0002* UEFI: PM951 NVMe SAMSUNG 512GB, Partition 1	HD(1,GPT,ba79953c-d5b8-4598-9f25-13c79ed61ebf,0x800,0x80000)/File(EFI\boot\bootx64.efi)..BO
Boot0003* fallback.efi	PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-25-38-41-61-B4-97-7B)/HD(1,GPT,ba79953c-d5b8-4598-9f25-13c79ed61ebf,0x800,0x80000)/File(\EFI\BOOT\fallback.efi)
Boot0004* Linux-Firmware-Updater \fwupx64.efi	HD(1,GPT,ba79953c-d5b8-4598-9f25-13c79ed61ebf,0x800,0x80000)/File(\EFI\fedora\shim.efi)\.f.w.u.p.x.6.4...e.f.i...
```

jhoblitt, 2016-06-30:
and success!  This is really exciting.  No BMC and no freedos needed.

Plus, the firmware update fixed the hanging reboot issue.

Thanks guys!

danielkza, 2016-06-30:
Wonderful! Now we just need to get the Fedora efivar maintainers to apply the patches. Having that nice GUI on GNOME Software isn't of much use if the upgrade itself doesn't work.

closed 2016-06-30 by superm1