fwupd/fwupd #55: Wishlist: central community rating / review infrastructure

andim2, 2016-07-20:
Hi,

it is very nice to see large efforts to finally get a central firmware update mechanism in place!

Having now said that, I believe that it could be very useful to have some central community infrastructure somewhere closely nearby / around this central firmware update handling.

I am thinking of features such as:
- doing reviews / annotations of devices and individual firmware versions
- blacklisting particular firmware versions for upgrading, unless manually overridden by the user (this feature could be implemented by letting users rate stability of a firmware version, and e.g. default-to-blacklisted of upgrading process in case average rating of a firmware version with more than say 3 ratings went below 3 stars out of 10 stars)

Obviously one would not want to have the central fwupd daemon part develop any grave dependency on such much more external/optional review / rating infrastructure - however, creating such infrastructure around main fwupd handling (and then letting fwupd handling act on certain key criteria) might be possible and very useful - by offering such review infrastructure centrally around fwupd, one would be able to offer tremendous features / flexibility around the usual boring / risky firmware update churn.

Thanks for a very useful project!

hughsie, 2016-07-21:
So, we could certainly use the ODRS currently at http://apps-xdgapp.rhcloud.com/ but this would be rating the firmware generally, and not for the specific version. @allanday is this something we should enable in gnome-software? At the moment I'm explicitly turning off the review section for firmware.

superm1, 2016-07-21:
I think any review system in use that would cause automated blacklisting,
it's important that the reviews be tied to specific versions of firmware
paired with specific versions of a software stack.  In many cases firmware
and kernel drivers go too closely in hand with one-another to allow a
problem that was seemingly raised by firmware to be 100% blamed on the
firmware.

Here's a particular (recent) example:
https://patchwork.kernel.org/patch/9240703/
The problem discussed there seemingly started to happen when upgrading to
firmware 1.4.4 on that laptop.  After digging in further it seems that
actually some combination of the patches in the version of GRUB in Fedora
that causes assumptions the kernel made about the location and size of the
EBDA to be wrong.  It doesn't reproduce with the EFI stub loader and it
doesn't reproduce on another distro, so would it make sense to blacklist
the firmware for all distros if it only happened from that particular
software stack in Fedora?

To me there could still certainly be value for generally reviewing feedback
of firmware updates in a central location, even if it's people commenting
"Works great for me!".

hughsie, 2017-05-23:
I think the ODRS is that system, although I'm not sure how it should be integrated into the GUI.

hughsie, 2018-01-17:
We've done a bit of work on this. We didn't want people to rate the reviews, but we did want to know if they were applied successfully or with an error. See https://blogs.gnome.org/hughsie/2018/01/10/phoning-home-after-updating-firmware/ for an overview. There's still more work to do, but enough done to close this issue I think.

closed 2018-01-17 by hughsie

andim2, 2018-01-17:
On Wed, Jan 17, 2018 at 08:31:23PM +0000, Richard Hughes wrote:
>    We've done a bit of work on this. We didn't want people to rate the
>    reviews, but we did want to know if they were applied successfully or with
>    an error. See
>    [1]https://blogs.gnome.org/hughsie/2018/01/10/phoning-home-after-updating-firmware/
>    for an overview. There's still more work to do, but enough done to close
>    this issue I think.

To which I say: good!

(replies with interesting thoughts on that URL, too:
- G+ [non-]use counts of privacy-minded people
- [non-]details breadth of reply data
)

It might even be useful to allow for
feedback of e.g. custom single-sentence replies about firmware specifics
(any good? Experience? Notable benefits? ...), however
management/privacy of such a data source would be somewhat of an additional concern.

Haven't had the opportunity yet to test this brave new infrastructure, however
thumbs up!

Andreas Mohr

-- 
Das "S" in "IoT" steht für "Security"
[Signatur von betateilchen:
https://forum.fhem.de/index.php/topic,23008.msg163661.html#msg163661 ]

superm1, 2018-01-17:
The important part really is getting actionable data so if there are field problems specifically with the installation process they can be fixed.
Right now the data that comes with success/failure isn't going to be exposed publicly.  It's just coming into a view on various LVFS accounts for knowing when there are problems happening.  I think it's logical to expose a subset of it more publicly eventually (maybe just thumbs up/down against a device? or plugin? or an OEM? or percentages?).  That will probably be another survey in the future.