fwupd/fwupd #58: "Software" hangs trying to update 8bitdo firmware

hadess, 2016-08-23:
Backtrace:
```
Thread 6 (Thread 0x7f217fdf9700 (LWP 2040)):
#0  0x00007f21ba6568d1 in __pthread_rwlock_rdlock_slow () at /lib64/libpthread.so.0
#1  0x000055cd48b2f358 in gs_plugin_action_start (plugin=plugin@entry=0x55cd49bef290, exclusive=exclusive@entry=0) at gs-plugin.c:264
#2  0x000055cd48b4bcac in gs_plugin_loader_action_start (plugin_loader=plugin_loader@entry=0x55cd49b9b410, plugin=plugin@entry=0x55cd49bef290, exclusive=exclusive@entry=0) at gs-plugin-loader.c:212
#3  0x000055cd48b4c1a9 in gs_plugin_loader_run_refine_internal (list=0x55cd4baf7390, plugin_loader=0x55cd49b9b410) at gs-plugin-loader.c:263
#4  0x000055cd48b4c1a9 in gs_plugin_loader_run_refine_internal (plugin_loader=plugin_loader@entry=0x55cd49b9b410, function_name_parent=function_name_parent@entry=0x0, list=list@entry=0x55cd4baf7390, flags=flags@entry=2097160, cancellable=cancellable@entry=0x0, error=error@entry=0x7f217fdf8b40)
    at gs-plugin-loader.c:321
#5  0x000055cd48b4ca7d in gs_plugin_loader_run_refine (plugin_loader=plugin_loader@entry=0x55cd49b9b410, function_name_parent=function_name_parent@entry=0x0, list=list@entry=0x55cd4baf7390, flags=2097160, cancellable=cancellable@entry=0x0, error=error@entry=0x7f217fdf8b40) at gs-plugin-loader.c:533
#6  0x000055cd48b4f2be in gs_plugin_loader_app_refine_thread_cb (task=0x55cd4d1bf020, object=<optimized out>, task_data=0x55cd4d2c0520, cancellable=0x0) at gs-plugin-loader.c:2605
#7  0x00007f21bb36d9bd in g_task_thread_pool_thread () at /lib64/libgio-2.0.so.0
#8  0x00007f21badf76ee in g_thread_pool_thread_proxy () at /lib64/libglib-2.0.so.0
#9  0x00007f21badf6cf5 in g_thread_proxy () at /lib64/libglib-2.0.so.0
#10 0x00007f21ba6525ca in start_thread () at /lib64/libpthread.so.0
#11 0x00007f21ba38bf6d in clone () at /lib64/libc.so.6

Thread 5 (Thread 0x7f2190f4d700 (LWP 1990)):
#0  0x00007f21ba3860b9 in syscall () at /lib64/libc.so.6
#1  0x00007f21bae14b6f in g_cond_wait () at /lib64/libglib-2.0.so.0
#2  0x00007f21badcfd30 in g_main_context_wait () at /lib64/libglib-2.0.so.0
#3  0x00007f21badd0d33 in g_main_loop_run () at /lib64/libglib-2.0.so.0
#4  0x00007f21941eeb63 in fwupd_client_update_metadata (client=0x55cd49c011e0, metadata_fn=metadata_fn@en---Type <return> to continue, or q <return> to quit---
try=0x7f2184844230 "/home/hadess/.cache/gnome-software/3.21/firmware/firmware.xml.gz", signature_fn=<optimized out>, cancellable=cancellable@entry=0x55cd4bb33420, error=error@entry=0x7f2190f4cb30)
    at fwupd-client.c:968
#5  0x00007f21943fcb67 in gs_plugin_refresh (error=0x7f2190f4cb30, cancellable=0x55cd4bb33420, cache_age=<optimized out>, plugin=0x55cd49bef290) at gs-plugin-fwupd.c:660
#6  0x00007f21943fcb67 in gs_plugin_refresh (plugin=0x55cd49bef290, cache_age=<optimized out>, flags=GS_PLUGIN_REFRESH_FLAGS_METADATA, cancellable=0x55cd4bb33420, error=0x7f2190f4cb30) at gs-plugin-fwupd.c:683
#7  0x000055cd48b501ba in gs_plugin_loader_refresh_thread_cb (error=0x7f2190f4cb20, cancellable=0x55cd4bb33420, flags=<optimized out>, cache_age=4294967295, plugin_loader=0x55cd49b9b410)
    at gs-plugin-loader.c:4072
#8  0x000055cd48b501ba in gs_plugin_loader_refresh_thread_cb (task=0x55cd4d28ad00, object=<optimized out>, task_data=<optimized out>, cancellable=0x55cd4bb33420) at gs-plugin-loader.c:4119
#9  0x00007f21bb36d9bd in g_task_thread_pool_thread () at /lib64/libgio-2.0.so.0
#10 0x00007f21badf76ee in g_thread_pool_thread_proxy () at /lib64/libglib-2.0.so.0
#11 0x00007f21badf6cf5 in g_thread_proxy () at /lib64/libglib-2.0.so.0
#12 0x00007f21ba6525ca in start_thread () at /lib64/libpthread.so.0
#13 0x00007f21ba38bf6d in clone () at /lib64/libc.so.6

Thread 4 (Thread 0x7f21a791d700 (LWP 1676)):
#0  0x00007f21ba3803ed in poll () at /lib64/libc.so.6
#1  0x00007f21badd0a06 in g_main_context_iterate.isra () at /lib64/libglib-2.0.so.0
#2  0x00007f21badd0b1c in g_main_context_iteration () at /lib64/libglib-2.0.so.0
#3  0x00007f21a7924fad in dconf_gdbus_worker_thread () at /usr/lib64/gio/modules/libdconfsettings.so
#4  0x00007f21badf6cf5 in g_thread_proxy () at /lib64/libglib-2.0.so.0
#5  0x00007f21ba6525ca in start_thread () at /lib64/libpthread.so.0
#6  0x00007f21ba38bf6d in clone () at /lib64/libc.so.6

Thread 3 (Thread 0x7f21a832b700 (LWP 1675)):
#0  0x00007f21ba3803ed in poll () at /lib64/libc.so.6
#1  0x00007f21badd0a06 in g_main_context_iterate.isra () at /lib64/libglib-2.0.so.0
---Type <return> to continue, or q <return> to quit---
#2  0x00007f21badd0d92 in g_main_loop_run () at /lib64/libglib-2.0.so.0
#3  0x00007f21bb3c9e46 in gdbus_shared_thread_func () at /lib64/libgio-2.0.so.0
#4  0x00007f21badf6cf5 in g_thread_proxy () at /lib64/libglib-2.0.so.0
#5  0x00007f21ba6525ca in start_thread () at /lib64/libpthread.so.0
#6  0x00007f21ba38bf6d in clone () at /lib64/libc.so.6

Thread 2 (Thread 0x7f21a8b2c700 (LWP 1674)):
#0  0x00007f21ba3803ed in poll () at /lib64/libc.so.6
#1  0x00007f21badd0a06 in g_main_context_iterate.isra () at /lib64/libglib-2.0.so.0
#2  0x00007f21badd0b1c in g_main_context_iteration () at /lib64/libglib-2.0.so.0
#3  0x00007f21badd0b61 in glib_worker_main () at /lib64/libglib-2.0.so.0
#4  0x00007f21badf6cf5 in g_thread_proxy () at /lib64/libglib-2.0.so.0
#5  0x00007f21ba6525ca in start_thread () at /lib64/libpthread.so.0
#6  0x00007f21ba38bf6d in clone () at /lib64/libc.so.6

Thread 1 (Thread 0x7f21be862a80 (LWP 1648)):
#0  0x00007f21ba3803ed in poll () at /lib64/libc.so.6
#1  0x00007f21badd0a06 in g_main_context_iterate.isra () at /lib64/libglib-2.0.so.0
#2  0x00007f21badd0d92 in g_main_loop_run () at /lib64/libglib-2.0.so.0
#3  0x000055cd48b57047 in gs_plugin_loader_app_refine (plugin_loader=plugin_loader@entry=0x55cd49b9b410, app=app@entry=0x7f218c587970, flags=flags@entry=2097160, cancellable=cancellable@entry=0x0, error=error@entry=0x7ffe7dad9a18) at gs-plugin-loader-sync.c:471
#4  0x000055cd48b570c9 in gs_plugin_loader_get_app_by_id (plugin_loader=0x55cd49b9b410, id=<optimized out>, flags=flags@entry=2097160, cancellable=cancellable@entry=0x0, error=error@entry=0x7ffe7dad9a18)
    at gs-plugin-loader-sync.c:37
#5  0x000055cd48b577b5 in handle_get_result_metas (skeleton=<optimized out>, invocation=0x7f21a000b8f0, results=0x55cd4cf768f0, user_data=0x55cd4c3c4200) at gs-shell-search-provider.c:192
#6  0x00007f21b43d5c58 in ffi_call_unix64 () at /lib64/libffi.so.6
#7  0x00007f21b43d56ba in ffi_call () at /lib64/libffi.so.6
#8  0x00007f21bb0a4c1e in g_cclosure_marshal_generic () at /lib64/libgobject-2.0.so.0
---Type <return> to continue, or q <return> to quit---
#9  0x00007f21bb0a43e5 in g_closure_invoke () at /lib64/libgobject-2.0.so.0
#10 0x00007f21bb0b6432 in signal_emit_unlocked_R () at /lib64/libgobject-2.0.so.0
#11 0x00007f21bb0be241 in g_signal_emitv () at /lib64/libgobject-2.0.so.0
#12 0x000055cd48b5e6a9 in _gs_shell_search_provider2_skeleton_handle_method_call (connection=<optimized out>, sender=<optimized out>, object_path=<optimized out>, interface_name=0x7f21a0001190 "org.gnome.Shell.SearchProvider2", method_name=0x7f21a000a060 "GetResultMetas", parameters=<optimized out>, invocation=0x7f21a000b8f0, user_data=0x55cd49c6c270) at gs-shell-search-provider-generated.c:1669
#13 0x00007f21bb3d27c7 in skeleton_intercept_handle_method_call () at /lib64/libgio-2.0.so.0
#14 0x00007f21bb3ba7ec in call_in_idle_cb () at /lib64/libgio-2.0.so.0
#15 0x00007f21badd06ba in g_main_context_dispatch () at /lib64/libglib-2.0.so.0
#16 0x00007f21badd0a70 in g_main_context_iterate.isra () at /lib64/libglib-2.0.so.0
#17 0x00007f21badd0b1c in g_main_context_iteration () at /lib64/libglib-2.0.so.0
#18 0x00007f21bb39348d in g_application_run () at /lib64/libgio-2.0.so.0
#19 0x000055cd48b18404 in main (argc=2, argv=0x7ffe7dada4a8) at gs-main.c:55
```

```
gnome-software-3.21.91-0.236.20160819git.fc24.x86_64
fwupd-0.7.3-0.194.20160823git.fc24.x86_64
libebitdo-0.7.3-0.194.20160823git.fc24.x86_64
```

hughsie, 2016-08-23:
How did you get that?! Got some reproducing instructions?

hadess, 2016-08-23:
Just started gnome-software with the 8bitdo controller attached. I saw the screen saying it was downloading firmware information.
```
Bus 001 Device 008: ID 2002:9000 DAP Technologies 
```

Killing and restarting gnome-software seems to have avoided that particular race, it's not hanging, and it downloaded the firmware information, but it didn't offer to update the firmware. I can't run the ebitdo-tool as it's already opened by the backend. lsof info:
```
fwupd     1873         root   31u      CHR              189,7         0t0      43444 /dev/bus/usb/001/008
gmain     1873 1876    root   31u      CHR              189,7         0t0      43444 /dev/bus/usb/001/008
fwupd     1873 1877    root   31u      CHR              189,7         0t0      43444 /dev/bus/usb/001/008
GUsbEvent 1873 1878    root   31u      CHR              189,7         0t0      43444 /dev/bus/usb/001/008
GUsbEvent 1873 1879    root   31u      CHR              189,7         0t0      43444 /dev/bus/usb/001/008
GUsbEvent 1873 1880    root   31u      CHR              189,7         0t0      43444 /dev/bus/usb/001/008
GUsbEvent 1873 1881    root   31u      CHR              189,7         0t0      43444 /dev/bus/usb/001/008
GUsbEvent 1873 1882    root   31u      CHR              189,7         0t0      43444 /dev/bus/usb/001/008
gdbus     1873 1883    root   31u      CHR              189,7         0t0      43444 /dev/bus/usb/001/008
```

hadess, 2016-08-26:
Moved to https://bugzilla.gnome.org/show_bug.cgi?id=770435

closed 2016-08-26 by hadess