fwupd/fwupd #62: 0.7.3 fails to compile with appstream-glib 0.5.13

superm1, 2016-08-29:
configure.ac currently lists the dependency upon appstream-glib as:
> PKG_CHECK_MODULES(APPSTREAM_GLIB, appstream-glib >= 0.5.10)

On my system I have 0.5.13 installed.  Unfortunately this fails to compile:

> fu-main.c: In function ‘fu_main_daemon_method_call’:                                                                         
> fu-main.c:1711:10: warning: implicit declaration of function ‘as_utils_unique_id_build’ [-Wimplicit-function-declaration]    
>     id2 = as_utils_unique_id_build (AS_APP_SCOPE_SYSTEM,                                                                     
>           ^                                                                                                                  
> fu-main.c:1711:4: warning: nested extern declaration of ‘as_utils_unique_id_build’ [-Wnested-externs]                        
>     id2 = as_utils_unique_id_build (AS_APP_SCOPE_SYSTEM,
>     ^
> fu-main.c:1711:36: error: ‘AS_APP_SCOPE_SYSTEM’ undeclared (first use in this function)
>     id2 = as_utils_unique_id_build (AS_APP_SCOPE_SYSTEM,
>                                     ^
> fu-main.c:1711:36: note: each undeclared identifier is reported only once for each function it appears in

The missing method was added in appstream-glib from 0.6.1 version.

hughsie, 2016-08-29:
Can you test master please; if this works I'll either do a new release in a few days, an appstream-glib release, or both.

closed 2016-08-29 by hughsie

superm1, 2016-08-29:
Yes, this fixes compilation error, thanks.