fwupd/fwupd #63: fwupd 0.7.3 test failure on s390x

sharkcz, 2016-08-29:
the test-suite run in Fedora Rawhide on s390x fails with

...
/libdfu/firmware{elf}: **
libdfu:ERROR:dfu-self-test.c:357:dfu_firmware_elf_func: assertion failed (_g_bytes_compare_verbose (roundtrip, roundtrip_orig) == NULL): ("got 0x6c, expected 0x68 @ 0x0034" == NULL)
(/builddir/build/BUILD/fwupd-0.7.3/libdfu/.libs/lt-dfu-self-test:59144): libdfu-DEBUG: loading 32b ELF object
FAIL /builddir/build/BUILD/fwupd-0.7.3/libdfu/.libs/lt-dfu-self-test (exit status: 134)
...

please see http://s390.koji.fedoraproject.org/koji/taskinfo?taskID=2331142 for more details

hughsie, 2016-08-31:
Is there any chance you could install that build on an s390 machine and then attach the example2.elf file when running `dfu-tool convert elf example.elf example2.elf` -- thanks.

[example.elf.zip](https://github.com/hughsie/fwupd/files/446685/example.elf.zip)

sharkcz, 2016-08-31:
Done, see the attached file. If you need access to the s390 (or ppc if it's endian related bug) machine, let me know.
[example2.elf.zip](https://github.com/hughsie/fwupd/files/446739/example2.elf.zip)

hughsie, 2016-08-31:
Looking at the hexdump I guess the problem is pretty obvious:

    -00000030  03 00 02 00 68 65 6c 6c  6f 20 77 6f 72 6c 64 0a  |....hello world.|
    +00000030  03 00 02 00 6c 6c 65 68  6f 77 20 6f 0a 64 6c 72  |....llehow o.dlr|

closed 2016-08-31 by hughsie

sharkcz, 2016-08-31:
ack, the test-suite now passes

hughsie, 2016-08-31:
Great, thanks for verifying.