fwupd/fwupd #65: Listing of all firmware available and letting the user chose a specific one

fb87, 2016-09-15:
For e.g. downgrading from a buggy firmware update, it would be helpful if all available firmware versions could be made accessible, so that the user can chose a specific one to install.

I own a Dell XPS 15 and the new firmware is far worse than the shipped one, because of several bugs introduced. So I'd like to be able to downgrade the UEFI firmware.

superm1, 2016-09-15:
At least on the XPS 15, this would generally be prohibited because the ESRT prohibits downgrades from capsule updates (for security reasons).

You can check in the ESRT for the lowest supported version number.

It's possible to downgrade from the F12 post menu by putting the binary (EXE) on a fat32 partition such as a USB disk or the ESP.

If you haven't already you should report the issues you've found to the forums or to your support representatives. They can relay the information to the applicable teams.

superm1, 2016-09-15:
You can also put the payload from within the capsule (firmware.bin) on fat32 and pick from F12 menu too.

fb87, 2016-09-15:
I can activate downgrades in the UEFI. I've also read of users downgrading to an older UEFI on the same machine.

But this would benefit other users as well – I didn't create this bug report just because of me having problems. ;)

superm1, 2016-09-16:
Marking them in BIOS setup won't actually change the ESRT though. You still can't use a capsule to downgrade to an earlier version. That BIOS setting only allows F12 and EXE interactive downgrades.

To check this, go and look at the following:
```
# cat /sys/firmware/efi/esrt/entries/entry0/lowest_supported_fw_version
# cat /sys/firmware/efi/esrt/entries/entry0/fw_version
```

You'll see they both come back with the same value meaning you can't flash the "same" or an older version.  This won't change even if you change the setting in the BIOS that allows downgrades.

fb87, 2016-09-17:
Ok, you're right. But maybe this might change in future and is different on other hardware.

I still think it would be beneficial – but I'll understand, if you don't want to implement this due to not much usage yet.

superm1, 2016-09-19:
@fb87 If you have problems with any of the open source tools (fwupd/fwupdmgr/efibootmgr), it's best to report them on github to the appropriate projects.  If they're something specific to Dell we're watching for issues and working to make the experience solid on Dell systems.  If it's a BIOS specific issue you unfortunately will not find a centralized bug tracking system.  The best feedback I can give would be to report it through support.  They're the interface to the various teams that would produce a fix.  If they see patterns of particular issues they'll be able to work with the teams to prioritize getting them fixed.

You can also report it to http://en.community.dell.com/techcenter/os-applications/f/4613 and there are agents that will look for trends too.

Also, fwupd supports lots of flashing technologies.  I do think generally downgrade is a good feature to have for the providers that it would work well.  fwupd 0.7.3 added the feature to blacklist updates that are problematic for you.  When paired with a downgrade feature it could help prevent always being prompted for an update that is causing you problems.

superm1, 2017-06-06:
This should be closed with: https://github.com/hughsie/fwupd/commit/45c1545b0aaad017467989c6dd81980ad2a7ac07

closed 2017-06-06 by superm1