fwupd/fwupd #67: Unable to install UEFI-dummy-dev0 firmware

BenjaminSchubert, 2016-09-27:
On fedora 25, on my Dell XPS 13 9350, I'm unable to install the UEFI-dummy-dev0 firmware.

I've tried to use fwupdate and fwupdmgr to do this, but both fail without explanation. I also tried to reboot the system in between and this did nothing.

Here is my dmidecode output : 

```
# dmidecode 3.0
Getting SMBIOS data from sysfs.
SMBIOS 2.8 present.
94 structures occupying 5688 bytes.
Table at 0x000E0000.

Handle 0xDA00, DMI type 218, 251 bytes
OEM-specific Type
	Header and Data:
		DA FB 00 DA B2 00 04 FF 1F B6 40 05 00 05 00 03
		00 06 00 06 00 05 00 0B 00 0B 00 01 00 0C 00 0C
		00 02 00 0D 00 0D 00 03 00 0F 00 0F 00 00 00 11
		00 11 00 02 00 12 00 12 00 04 00 1E 00 1E 00 00
		00 22 00 22 00 01 00 23 00 23 00 00 00 28 00 28
		00 00 00 29 00 29 00 01 00 2A 00 2A 00 02 00 2B
		00 2B 00 FF FF 2C 00 2C 00 FF FF 40 00 40 00 01
		00 41 00 41 00 00 00 42 00 42 00 01 00 43 00 43
		00 00 00 50 00 50 00 01 00 55 00 55 00 00 00 5C
		00 5C 00 01 00 5D 00 5D 00 00 00 65 00 65 00 00
		00 66 00 66 00 01 00 7D 00 7D 00 FF FF 93 00 93
		00 01 00 94 00 94 00 00 00 9B 00 9B 00 01 00 9C
		00 9C 00 00 00 9F 00 9F 00 00 00 A0 00 A0 00 01
		00 A1 00 A1 00 00 00 A3 00 A3 00 01 00 D1 00 D1
		00 01 00 D2 00 D2 00 00 00 EA 00 EA 00 00 00 EB
		00 EB 00 01 00 FF FF FF FF 00 00

Handle 0xDA01, DMI type 218, 251 bytes
OEM-specific Type
	Header and Data:
		DA FB 01 DA B2 00 04 FF 1F B6 40 EC 00 EC 00 02
		00 ED 00 ED 00 00 00 F0 00 F0 00 01 00 F1 00 F1
		00 00 00 F2 00 F2 00 01 00 F3 00 F3 00 02 00 09
		01 09 01 00 00 0E 01 0E 01 01 00 0F 01 0F 01 00
		00 10 01 10 01 01 00 11 01 11 01 00 00 1B 01 1B
		01 00 00 1C 01 1C 01 01 00 2B 01 2B 01 01 00 2C
		01 2C 01 00 00 2D 01 2D 01 01 00 2E 01 2E 01 00
		00 35 01 35 01 FF 00 38 01 38 01 01 00 39 01 39
		01 02 00 40 01 40 01 00 00 41 01 41 01 01 00 44
		01 44 01 00 00 45 01 45 01 01 00 46 01 46 01 00
		00 47 01 47 01 01 00 4A 01 4A 01 00 00 4B 01 4B
		01 01 00 4C 01 4C 01 01 00 4D 01 4D 01 00 00 52
		01 52 01 01 00 53 01 53 01 00 00 75 01 75 01 02
		00 76 01 76 01 01 00 7F 01 7F 01 00 00 80 01 80
		01 01 00 81 01 81 01 00 00 82 01 82 01 01 00 89
		01 89 01 00 00 FF FF FF FF 00 00

Handle 0xDA02, DMI type 218, 251 bytes
OEM-specific Type
	Header and Data:
		DA FB 02 DA B2 00 04 FF 1F B6 40 8A 01 8A 01 01
		00 93 01 93 01 00 00 94 01 94 01 01 00 9B 01 9B
		01 00 00 9C 01 9C 01 01 00 DE 01 DE 01 00 00 DF
		01 DF 01 01 00 E1 01 E1 01 00 00 EA 01 EA 01 00
		00 EB 01 EB 01 01 00 02 02 02 02 00 00 03 02 03
		02 01 00 04 02 04 02 00 00 05 02 05 02 01 00 16
		02 16 02 06 00 2D 02 2D 02 01 00 2E 02 2E 02 00
		00 32 02 32 02 02 00 33 02 33 02 01 00 35 02 35
		02 01 00 36 02 36 02 00 00 4A 02 4A 02 01 00 4B
		02 4B 02 01 00 4C 02 4C 02 00 00 64 02 64 02 01
		00 65 02 65 02 00 00 66 02 66 02 01 00 67 02 67
		02 00 00 68 02 68 02 01 00 69 02 69 02 00 00 6C
		02 6C 02 01 00 6D 02 6D 02 00 00 6E 02 6E 02 00
		00 85 02 85 02 01 00 86 02 86 02 00 00 94 02 94
[... 985 lines cut ...]
		EC-EC Protocol Version
		Royal Park Version
		BP1.2.2.0_RP03

Handle 0xDE00, DMI type 222, 16 bytes
OEM-specific Type
	Header and Data:
		DE 10 00 DE 01 04 00 00 16 07 12 16 40 01 00 00

Handle 0x0033, DMI type 255, 8 bytes
OEM-specific Type
	Header and Data:
		FF 08 33 00 01 02 00 00
	Strings:
		_SIDLLEwPLyhIgmg
		0704
		_SIDAvf90w7wKZiZ
		DELL

Handle 0xF03E, DMI type 127, 4 bytes
End Of Table

```

Please tell me if you need something else !

superm1, 2016-09-27:
What exactly are you trying to do?  `# fwupdmgr update`?  Currently the UEFI capsule support is not turned on on your box and needs to be enabled. 

So it might not be immediately clear, but what you need to do is enable capsule updates.
3 ways to acomplish this:

1) In Gnome-Software find the UEFI dummy device and double click it.
2) `# fwupdmgr unlock UEFI-dummy-dev0` and reboot.
3) Go into BIOS setup and find the menu option to turn on UEFI capsule updates.

It would be nice to see confirmation that all the code that's landed in Fedora 25 related to this is working properly if you can do it using gnome-software.  Gnome Software will use fwupd's unlock mechanism which uses fwupdate's unlock mechanism which uses libsmbios code to communicate with the BIOS.

BenjaminSchubert, 2016-09-27:
Sorry I was not clear enough, 

I tried to install it through gnome-software, which fails silently and only offers to reinstall it afterwards... which still does not work

The `fwupdmgr unlock` command fails with `timeout was reached`

I also tried rebooting before both to be sure.

I'm available to troubleshoot if I can be of any help, just tell me what to do !

P.s. I have a password on my UEFI options, could that be it ?

superm1, 2016-09-27:
Ah OK.  Yeah if we can get to the bottom of this problem without you manually doing it in BIOS setup that would be ideal.

So can you try to run this:
`# fwupdate -e`
That will try to use fwupdate to try to enable it (with fwupd not in the picture).

Post back what it sends back.

I don't believe that a password set should affect this particular option, but if it's failing with `fwupdate -e `can you try to temporarily remove the password to see if it helps?  Also, which type of BIOS password is set - _Administrator_ or _User_?

BenjaminSchubert, 2016-09-27:
The result of `fwupdate -e` is :

```
Firmware updates can not be enabled on this machine from this tool.
```

Disabling the Admin Password in the bios does the trick and fwupdate -e worked afterwards

superm1, 2016-09-27:
Great thanks.  I've created a patch for fwupdate that tests for admin password and will only offer this  functionality when applicable.  I'll close this issue in fwupd and you can track it in that pull request above for fwupdate if you want.

closed 2016-09-27 by superm1