fwupd/fwupd #68: Dell XPS 15 9550 isn't detected as updatable

oskarb, 2016-10-10:
Hi,

I have a Dell XPS 15 9550, for which there are BIOS files available in LVFS. However, fwupdmgr doesn't show any available updates, and there is nothing to indicate that the BIOS is upgradeable. On Fedora 24:

```
$ fwupdmgr get-updates
No devices can be updated: Nothing to do
```

```
$ fwupdmgr get-devices
GM107M [GeForce GTX 960M]
  Guid:                 520d05a3-ece8-519e-88de-0755806314f4
  DeviceID:             ro__sys_devices_pci0000_00_0000_00_01_0_0000_01_00_0
  Provider:             Udev
  Flags:                internal|locked
  DeviceVendor:         NVIDIA Corporation
  Created:              2016-10-10

Skylake Integrated Graphics
  Guid:                 8ec3e242-65bd-58ee-9c5e-b28f1e32e7ae
  DeviceID:             ro__sys_devices_pci0000_00_0000_00_02_0
  Provider:             Udev
  Flags:                internal|locked
  FirmwareHash:         0ef770b17699c97f5c08d5c1fcd46517f39e7c07
  DeviceChecksumKind:   sha1
  DeviceVendor:         Intel Corporation
  Created:              2016-10-10

BCM2045A0
  Guid:                 f8498646-6724-5f55-9806-57b3f8636ad8
  Guid:                 35fd074d-7c78-599e-b37a-88931c63ba01
  DeviceID:             usb:00:04
  Provider:             DFU
  Flags:                allow-online|allow-offline
  Version:              1.18
  Created:              2016-10-10

Touchscreen
  Guid:                 01693e7d-7382-5e67-a7fc-5aa70fc55163
  Guid:                 51e923fe-c806-5601-bb4e-31f300e75069
  DeviceID:             usb:00:09
  Provider:             USB
  Flags:                none
  Version:              17.18
  Created:              2016-10-10

Integrated Webcam HD
  Guid:                 4c03e6af-b94c-5c18-8689-e77ceadbe524
  Guid:                 fb0df457-b15e-5d4a-b73a-8491dca96a07
  DeviceID:             usb:00:0c
  Provider:             USB
  Flags:                none
  Version:              86.5
  Created:              2016-10-10
```

Dmidecode indicates I'm on BIOS 1.2.00, but a 1.2.10 is available in LVFS (and a 1.2.14 on Dell's own web):
```
Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
        Vendor: Dell Inc.
        Version: 01.02.00
        Release Date: 04/07/2016
```

fwupd packages:
```
$ dnf list '*fwupd*'
Last metadata expiration check: 0:21:42 ago on Mon Oct 10 16:54:13 2016.
Installed Packages
fwupd.x86_64             0.7.2-4.fc24
fwupdate-efi.x86_64      8-2.fc24
fwupdate-libs.x86_64     8-2.fc24
```

superm1, 2016-10-10:
It looks to me that you don't currently have UEFI capsule updates turned on in the BIOS.  fwupd and fwupdate can turn this on remotely using libsmbios, but for some reason it's not listed as offered.

1. Do you have a BIOS password turned on?  If so, this unfortunately can not be done from the OS with a BIOS password.  You'll need to go into the BIOS setup and turn it on manually.
2. If you don't have a BIOS password, can you please share the output of this command (as root):
`# fwupdate -s`?  That will indicate whether fwupdate should be able to do the task of turning on capsule updates.

oskarb, 2016-10-12:
There is no BIOS password, and and fwupdate -s said updates were not supported. However, I enabled the capsule updates manually in BIOS and now fwupdate -s claims that firmware _can_ be updated.

Still doesn't quite work though. fwupdmgr get-devices now includes this:
```
XPS 15 9550
  Guid:                 124c207d-5db8-4d95-bd31-34fd971b34f9
  DeviceID:             UEFI-124c207d-5db8-4d95-bd31-34fd971b34f9-dev0
  Description:          <p>Updating the system firmware improves performance.</p>
  Provider:             UEFI
  Flags:                internal|allow-offline|require-ac|supported
  Version:              0.1.2.0
  VersionLowest:        0.1.2.0
  Created:              2016-10-12
  AppstreamId:          com.dell.uefi124c207d.firmware
  Name:                 XPS 15 9550/Precision 5510 System Update
  Summary:              Firmware for the Dell XPS 15 9550/Precision 5510
  UpdateVersion:        0.1.2.10
  UpdateHash:           7bbc0b7b2da905c1cf7e4fab73826f6bdb4d27b4
  UpdateChecksumKind:   sha1
  License:              proprietary
  UpdateUri:            https://secure-lvfs.rhcloud.com/downloads/47096f0c58188f4f7ffe4a3a6c4d9b70f92f37c1-firmware.cab
  UrlHomepage:          http://support.dell.com/
  Vendor:               Dell Inc.
  Trusted:              none
```
I take it that "UpdateVersion" is a new version available from LVFS, which fits wiith my own checking. fwupdmgr still refuses to perform an updating:

```
 fwupdmgr -v get-updates
No devices can be updated: Nothing to do
```

Trying to use the "install" command eventually revealed a hint:
```
$ fwupdmgr -v install Downloads/47096f0c58188f4f7ffe4a3a6c4d9b70f92f37c1-firmware.cab 
 * Decompressing firmware
Retrying as an offline update...
Cannot install update when not on AC power
```

The AC was connected when I initially opened this issue, but not when I started today. Perhaps get-updates should also mention this hint. The update is scheduled now...

oskarb, 2016-10-12:
The update worked. Apart from possibly adding some more hints (get-updates might show a warning if AC is missing and any device have "require-ac" perhaps)? Or perhaps get-updates should show the available update even if it cannot currently be applied due to some possibly transient issue?

superm1, 2016-10-12:
The first part of why it wasn't showing the option to turn on, for some reason or another the SMBIOS tokens 0x461 and 0x462 that allow turning on and off from within the OS aren't showing up on the XPS 9550 or Precision 5510.  That's a BIOS issue on the Dell side that I can reproduce.  The setting you made in the BIOS manually will stick however.

The second issue (where you don't get information about updates while not on AC), that is currently expected behavior for fwupd.  The daemon `fwupd `feeds this same information to both `gnome-software` and `fwupdmgr `frontends.  Gnome Software will also only show updates in the U/I when power is detected.  
@hughsie  can add some more comments, but the main thought process behind this is that gnome-software will be running in the background in your session.  If you don't have power, why do you need to know about the update you can't run?  When you do get power, it will notify that you can actually run the update.

oskarb, 2016-10-12:
I agree, with regards to unsolicited upgrade prompts, it makes sense to avoid those if they can't be applied anyway. It was an issue in this case only because I was actively working on it.

I was thinking something along these lines:
```
$ fwupdmgr get-updates
No devices can be updated: Nothing to do
Warning: AC power is currently disconnected. Some updates may not be displayed.
```

So well, the question is if the behaviour might differ slightly between a solicited and an unsolicited check for available updates. Certainly not a big issue though.

closed 2016-10-19 by superm1