fwupd/fwupd #69: Retries as offline without first failing (bad wording)

oskarb, 2016-10-12:
This issue is about the messages in the UI. No functionality issue.

```
$ fwupdmgr -v install Downloads/47096f0c58188f4f7ffe4a3a6c4d9b70f92f37c1-firmware.cab 
 * Decompressing firmware
Retrying as an offline update...
```

Since it's "retrying", I would have expected it to first fail with something else. The message would be more logical if it mentions some failed attempt before retrying.

superm1, 2016-10-12:
So what's going on here is that if you don't specify on the command line to install in _offline_ or _online_ mode `fwupdmgr `tries first _online_ mode.  If that fails for the reason online isn't supported, it "falls back" to _offline_ mode.

Looking around the documentation isn't very clear about what online mode and offline mode actually mean to an outsider.  This is what they mean.

**Online mode:** Flash the update from within the OS.
**Offline mode:** Schedule the update to happen on the next boot.

So what you actually saw happen here was the _UEFI provider_ doesn't support online mode (UEFI updates need to be scheduled for the next bootup).

So I think I understand the confusion.  At a minimum the documentation and --help need to explain offline and online better.  As for the particular command's confusion at the risk of bikeshedding, there are a few ideas i've got.  

* The most obvious solution to me is to test for _verbose mode_ and if so show
`Retrying as an offline update...` message only when in _verbose mode_.  What you'll instead see is:
```
$ fwupdmgr -v install Downloads/47096f0c58188f4f7ffe4a3a6c4d9b70f92f37c1-firmware.cab
 * Decompressing firmware
 * Decompressing firmware

```
That may still cause confusion since why did the firmware get decompressed twice?

* Be more verbose about the actual reason it fell back to offline update.  Make the error: 
```
$ fwupdmgr -v install Downloads/47096f0c58188f4f7ffe4a3a6c4d9b70f92f37c1-firmware.cab
 * Decompressing firmware
Online update is not supported, retrying as an offline update..._
 * Decompressing firmware
```
But then again, as a user you probably don't know what the difference between online update and offline update is in the first place.

*  Tell the user what is happening even more verbosely when the _verbose_ flag is found.  Leave it simple otherwise.   Something like this:
```
$ fwupdmgr install Downloads/47096f0c58188f4f7ffe4a3a6c4d9b70f92f37c1-firmware.cab
 * Decompressing firmware
 * Decompressing firmware
.
.
```
```
$ fwupdmgr install --offline Downloads/47096f0c58188f4f7ffe4a3a6c4d9b70f92f37c1-firmware.cab
 * Decompressing firmware
.
.
```
```
$ fwupdmgr -v install Downloads/47096f0c58188f4f7ffe4a3a6c4d9b70f92f37c1-firmware.cab
Attempting to flash firmware with OS (online update).
 * Decompressing firmware
Flashing in OS not supported, attempting to schedule update for next reboot (offline update)
 * Decompressing firmware
.
.
```

```
$ fwupdmgr --offline -v install Downloads/47096f0c58188f4f7ffe4a3a6c4d9b70f92f37c1-firmware.cab
Attempting to schedule update for next reboot (offline update)
 * Decompressing firmware
.
.
```

@hughsie think need your take on this one on how you want to message it.

oskarb, 2016-10-12:
I had verbose mode when I tried, so I'm not sure how it would have looked in non-verbose mode, but certainly, if "Decompressing firmware" is considered important enough to print, then "Writing firmware" or "Performing online update" or something should have at least the same level of importance?

hughsie, 2017-08-25:
I'm trying to make this whole thing unnecessary with https://github.com/hughsie/fwupd/tree/wip/hughsie/clarify-online-offline -- feedback welcome.

superm1, 2017-08-29:
That tree was merged, closing this issue.

closed 2017-08-29 by superm1