fwupd/fwupd #70: fwupd constantly using 100% of one CPU on Ubuntu 16.10

marco-c, 2016-10-19:


superm1, 2016-10-19:
@marco-c:

Will need some details to debug this further.

1. Can you comment more on the reproducibility?  Is this happening every boot?  When you plug in a USB device?  After some apt updates?
2. Share the output of `# lsusb`
3. Share output of `# dmesg`
4. Make a copy of all files in /var/lib/app-info/yaml/.  They may be needed to investigate further if the stacktrace shows the need, but they get updated daily with apt updates so if this problem is transient it would be important to see them in what would be a corrupted state.
5. A current stack trace to figure out where fwupd is spinning while using 100% CPU.  See more details at https://github.com/hughsie/fwupd/wiki/Ubuntu-backtracing for how to capture this stacktrace.

marco-c, 2016-10-19:
> Can you comment more on the reproducibility? Is this happening every boot? When you plug in a USB device? After some apt updates?

As far as I can tell, it is intermittent. It's unrelated to USB devices (I reproduced it once without any USB device plugged-in, another time with a USB mouse).
The regression started after I updated to 16.10, so it's really hard to tell if a specific package is at fault.

> Share the output of # lsusb 

```
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 138a:0017 Validity Sensors, Inc. Fingerprint Reader
Bus 003 Device 003: ID 058f:9540 Alcor Micro Corp. AU9540 Smartcard Reader
Bus 003 Device 002: ID 046d:c05b Logitech, Inc. M-U0004 810-001317 [B110 Optical USB Mouse]
Bus 003 Device 007: ID 0765:5010 X-Rite, Inc. 
Bus 003 Device 006: ID 5986:026a Acer, Inc 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

> Share output of # dmesg 

I've currently uninstalled fwupd as a workaround. Do you want me to send you the dmesg output after I reinstall it and while I run into the problem, or does any output work?

> Make a copy of all files in /var/lib/app-info/yaml/

[yaml.zip](https://github.com/hughsie/fwupd/files/539195/yaml.zip)

> A current stack trace to figure out where fwupd is spinning while using 100% CPU

Will do after I run into the problem again.

superm1, 2016-10-19:
> Will do after I run into the problem again.

Thanks.  The stack trace will be the most important piece to narrow down further.

> I've currently uninstalled fwupd as a workaround. Do you want me to send you the dmesg output after I reinstall it and while I run into the problem, or does any output work?

After you run into the problem.  The main thing looking for in the output is any messages from other parts of the system having problems (such as kernel drivers).  This would help to narrow it down.

> Make a copy of all files in /var/lib/app-info/yaml/

If you do reproduce this again, please repeat this step and we can use it as a basis for comparison in case it is indeed yaml related.

> Share the output of # lsusb

You do have a few "built-in" devices that are there even without your USB mouse plugged in, so it's entirely possible that one of them say has a DFU endpoint with some problems.

Same thing with this, let's compare if a USB device drops off the bus or goes into DFU mode (different VID/PID) if/when you reproduce this.

superm1, 2016-10-20:
> yaml.zip
 Next time you attach this, can you have it follow symlinks?  The files are empty in what you attached.

marco-c, 2016-10-20:
> Next time you attach this, can you have it follow symlinks? The files are empty in what you attached.

Oops, I hadn't noticed :)

marco-c, 2016-10-22:
[fwupd.zip](https://github.com/hughsie/fwupd/files/545803/fwupd.zip)

Looks like it might be happening after some apt updates.

marco-c, 2016-10-27:
It's happening every time I boot now.

Burusutazu, 2016-10-30:
I'm having the same issue. It happens at startup and is caused by gnome-software/ubuntu-software. If I kill the processes I can make it reoccur by opening ubuntu-software. Apologies if this is not the correct format, never used GitHub before.

lsusb output:
```
Bus 001 Device 002: ID 8087:8001 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 005: ID 04f3:2033 Elan Microelectronics Corp. 
Bus 002 Device 004: ID 8087:0a2a Intel Corp. 
Bus 002 Device 003: ID 0c45:64cb Microdia 
Bus 002 Device 006: ID 0483:91d1 STMicroelectronics Sensor Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

dmesg output:
```
[    0.000000] microcode: CPU0 microcode updated early to revision 0x22, date = 2015-09-11
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.0-45-generic (buildd@lgw01-34) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.2) ) #66-Ubuntu SMP Wed Oct 19 14:12:37 UTC 2016 (Ubuntu 4.4.0-45.66-generic 4.4.21)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.4.0-45-generic.efi.signed root=UUID=4676ebe4-a3dd-46af-b852-ab7f88aa2684 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Supporting XSAVE feature 0x01: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x02: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x04: 'AVX registers'
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] x86/fpu: Using 'eager' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x0000000000085fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000086000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000869d9fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000869da000-0x00000000872d9fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000872da000-0x000000009c4cefff] usable
[    0.000000] BIOS-e820: [mem 0x000000009c4cf000-0x000000009cebefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000009cebf000-0x000000009cfbefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000009cfbf000-0x000000009cffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000009cfff000-0x000000009cffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feb00000-0x00000000feb03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed19fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffb00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000025effffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x853c1018-0x853d1057] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000059000-0x0000000000085fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000086000-0x000000000009ffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x00000000853c1017] usable
[    0.000000] reserve setup_data: [mem 0x00000000853c1018-0x00000000853d1057] usable
[    0.000000] reserve setup_data: [mem 0x00000000853d1058-0x00000000869d9fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000869da000-0x00000000872d9fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000872da000-0x000000009c4cefff] usable
[    0.000000] reserve setup_data: [mem 0x000000009c4cf000-0x000000009cebefff] reserved
[    0.000000] reserve setup_data: [mem 0x000000009cebf000-0x000000009cfbefff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000009cfbf000-0x000000009cffefff] ACPI data
[    0.000000] reserve setup_data: [mem 0x000000009cfff000-0x000000009cffffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000feb00000-0x00000000feb03fff] reserved
[... 1025 lines cut ...]
[  927.636385] wlp1s0: associated
[  927.636414] IPv6: ADDRCONF(NETDEV_CHANGE): wlp1s0: link becomes ready
[ 3598.877258] do_trap: 51 callbacks suppressed
[ 3598.877271] traps: nautilus[2620] trap int3 ip:7f20d90d2a6b sp:7ffc40480c70 error:0
```
yaml files:

[dl.google.com_linux_earth_deb_dists_stable_main_dep11_Components-amd64.yml.gz](https://github.com/hughsie/fwupd/files/560510/dl.google.com_linux_earth_deb_dists_stable_main_dep11_Components-amd64.yml.gz)
[... cut ...]
[us.archive.ubuntu.com_ubuntu_dists_xenial-updates_multiverse_dep11_Components-amd64.yml.gz](https://github.com/hughsie/fwupd/files/560526/us.archive.ubuntu.com_ubuntu_dists_xenial-updates_multiverse_dep11_Components-amd64.yml.gz)
[us.archive.ubuntu.com_ubuntu_dists_xenial-updates_restricted_dep11_Components-amd64.yml.gz](https://github.com/hughsie/fwupd/files/560524/us.archive.ubuntu.com_ubuntu_dists_xenial-updates_restricted_dep11_Components-amd64.yml.gz)
[us.archive.ubuntu.com_ubuntu_dists_xenial-updates_universe_dep11_Components-amd64.yml.gz](https://github.com/hughsie/fwupd/files/560525/us.archive.ubuntu.com_ubuntu_dists_xenial-updates_universe_dep11_Components-amd64.yml.gz)

marco-c, 2016-11-15:
I can reproduce this pretty consistently.

Do you think you'll be able to get to this soon? Or should I just remove `fwupd` for the time being?

hughsie, 2016-11-15:
What version appstream-glib do you have installed?

marco-c, 2016-11-15:
> What version appstream-glib do you have installed?

The latest for Ubuntu 16.10:
https://launchpad.net/ubuntu/+source/appstream-glib/0.6.3-1

hughsie, 2016-11-15:
Right, the issue was fixed in 0.6.2 which was released a few weeks ago. Note, 0.6.3, 0.6.4 and 0.6.5 have also been released since fixing that bug, so I think someone needs to update Ubuntu to something newer on a more regular basis.

closed 2016-11-15 by hughsie

superm1, 2016-11-15:
@hughsie are you expecting it to be commits https://github.com/hughsie/appstream-glib/commit/f0990480624f7f76f6d82a4d1d7356464e340880 and https://github.com/hughsie/appstream-glib/commit/57f6d00ff98cdc5e70913701b8a70d47b3e5649b in 0.6.4 that fixed it?

Ubuntu typically doesn't take new versions in the name of stability, but will cherry pick patches to fix issues. @robert-ancell FYI, there's a need to pull in some fixes in yakkety^.

hughsie, 2016-11-16:
@superm1 My point is that I just can't QA 0.6.3 plus random patches. I already test all my stuff against all versions of deps of the lowest version and most newer tarballs since, and git master against git master. There are other yaml fixes in the code, e.g. a6c499d3b414237fca43c77b18300e8d753d6770 f0990480624f7f76f6d82a4d1d7356464e340880 and 57f6d00ff98cdc5e70913701b8a70d47b3e5649b as well as other important and stability fixes which are nothing to do with libyaml. I'm fine with Ubuntu patching stuff rather than rebasing (like RHEL often does) but I think these kind of bugs just need to be handled in Lauchpad rather than upstream.

marco-c, 2017-03-19:
@robert-ancell ping. Should I report this on Launchpad?

robert-ancell, 2017-03-19:
This is already reported at https://bugs.launchpad.net/ubuntu/+source/appstream-glib/+bug/1637024

waelben7, 2017-10-12:
I had the same problem and it can be an apt-get cache problem so it's recommended to clean the cache : 

 1. sudo -ic
 2. apt-get clean
 3. cd /var/lib/apt
 4. mv lists lists.old
 5. mkdir -p lists/partial
 6. apt-get clean
 7. apt-get update

or it can be a permission problem 

 - sudo chown {user} /home/{user}/.cache/dconf

this solve the problem for me.