fwupd/fwupd #75: UEFI firmware update failed: Unknown error -1

GTB3NW, 2016-12-10:
I'm receiving this error when trying to update my system. Here's the output of get-devices:

```
$ fwupdmgr get-devices
Unknown Device
  Guid:                 3ec3df3a-2290-56e5-9d2f-eda62e9ab50b
  DeviceID:             ro__sys_devices_pci0000_00_0000_00_02_0
  Provider:             Udev
  Flags:                internal|locked
  DeviceVendor:         Intel Corporation
  Created:              2016-12-10

XPS 13 9360
  Guid:                 48af7d21-3145-4f5e-b1b4-c8fc67b6e646
  UniqueID:             system/*/fwupd-verify/firmware/com.dell.uefi48af7d21.firmware/*
  DeviceID:             UEFI-48af7d21-3145-4f5e-b1b4-c8fc67b6e646-dev0
  Description:          <p>Updating the system firmware improves performance.</p>
  Provider:             UEFI
  Flags:                internal|allow-offline|require-ac|supported
  Version:              0.1.0.0
  VersionLowest:        0.1.0.0
  Created:              2016-12-10
  AppstreamId:          com.dell.uefi48af7d21.firmware
  Summary:              Firmware for the Dell XPS 13 9360
  UpdateVersion:        0.1.0.7
  UpdateHash:           74c864cb236904abcf9cdc3637bcfd974b3f7229
  UpdateChecksumKind:   sha1
  License:              proprietary
  UpdateUri:            https://secure-lvfs.rhcloud.com/downloads/fdffa502b4ebc3d60359d133e842c667a3d4b351-firmware.cab
  UrlHomepage:          http://support.dell.com/
  Vendor:               Dell Inc.
  Trusted:              none

Integrated Webcam HD
  Guid:                 4ec19583-c132-5c27-a9cf-061419e838dc
  Guid:                 6b31e6e6-ba7c-5251-8f4c-7f2571612251
  DeviceID:             usb:00:05
  Provider:             USB
  Flags:                none
  Version:              86.38
  Created:              2016-12-10

Touchscreen
  Guid:                 31cc46f6-9bdc-5b1e-b48b-1c791781dcdc
  Guid:                 9fd88db6-a96d-505b-bbea-92aee9d07455
  DeviceID:             usb:00:04
  Provider:             USB
  Flags:                none
  Version:              17.18
  Created:              2016-12-10
```

Output of update:

```
$ fwupdmgr update
Downloading 0.1.0.7 for XPS 13 9360...
Updating 0.1.0.7 on XPS 13 9360...
Decompressing…         
Retrying as an offline update...
Scheduling…            
UEFI firmware update failed: Unknown error -1
```

There is not output into journald at that time. Perhaps there is output elsewhere but I don't know where that will be if it is. I'd love to get this patched installed as currently my laptop isn't able to suspend correctly.

Please let me know if you need more information.

Cheers!

superm1, 2016-12-10:
There's a few common causes for this, but we'll have to narrow down some details.

Which distribution do you have installed?

What version of efivar, fwupd and fwupdate are installed?

Do you have secure boot turned on? 

What kernel version installed?

Can you please kill fwupd and manually restart it with the debug flag turned on? Check output from its console when this error occurs.

GTB3NW, 2016-12-10:
> Which distribution do you have installed?

Arch Linux

> What version of efivar, fwupd and fwupdate are installed?

efivar 30-1
fwupd 0.7.5-1
fwupdate 8-1

> Do you have secure boot turned on?

Nope

> What kernel version installed?

Linux archbook 4.8.12-3-ARCH #1 SMP PREEMPT Thu Dec 8 16:10:23 CET 2016 x86_64 GNU/Linux

> Can you please kill fwupd and manually restart it with the debug flag turned on?

I've killed the process but I'm honestly not too sure how to start fwupd with the debug flag. It looks like the daemon is automatically started when you run fwupdmgr. It runs as root so I suspect there's some kind of socket hook to spawn the daemon since I run fwupdmgr as non-root. Couldn't find any service files.

superm1, 2016-12-10:
> Arch Linux
> efivar 30-1
> fwupd 0.7.5-1
> fwupdate 8-1

OK so latest versions on all the packages and a kernel that has everything needed for ESRT to work correctly.  That rules out the common stuff that people have mentioned recently.  You're the first person I know of with an arch problem, so it's possible that it's an arch specific integration issue, but we'll need to dig into it further.

It's not the root cause of your issue, but according to that log upower isn't built in.  It's worth you mentioning that to the maintainer so that they can fix that in a future arch package.

As for your specific issue unfortunately that log isn't any more useful, so we'll need to run down the list of stuff that `libfwup0 `(from `fwupdate`) does to find out where the problem is.

1. Can you check `efibootmgr -v output`?  See if the _Linux-Firmware-Updater_ entry was created and what it points to.
2. Check the ESP to see whether `fwupx64.efi` and the payload have been installed to it.  You can share a recursive listing of whats on your ESP.
3. Check `/sys/firmware/efi/efivars` to see if any `fwupdate-*` entries were created.  If they were, we might have to dig further into that.

GTB3NW, 2016-12-10:
> It's not the root cause of your issue, but according to that log upower isn't built in. It's worth you mentioning that to the maintainer so that they can fix that in a future arch package.

I'm not quite sure what that means to be honest. I have seen a few power related messages in the journal on early boot and shutdown. Where would I submit the bug report and what should it say. I'm just lacking the understanding of what the issue is there, once I understand that I can probably fill in the gaps and complete a bug report. Thanks for pointing it out :)

1) I'm not sure why there's entries for windows, I don't see them within the bios utility, just the single linux boot loader. *shrugs*

```
$ efibootmgr -v                                                                                                                                                 
BootCurrent: 0003
Timeout: 0 seconds
BootOrder: 0003
Boot0000* Windows Boot Manager	HD(2,GPT,3518f332-d55f-4d32-8a97-45c50652194f,0xe1800,0x32000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0002  Linux Boot Manager	HD(1,GPT,817c2879-6db0-4b33-bec1-62ce7756cab4,0x800,0x100000)/File(\EFI\systemd\systemd-bootx64.efi)
Boot0003* Linux Boot Manager	HD(1,GPT,9f131103-0d0e-47cc-8f4b-a56d2a2b6ade,0x800,0x100000)/File(\EFI\systemd\systemd-bootx64.efi)
```

2) No fwup I'm afraid, when I straced before I couldn't see fwupdmgr trying to write anything like that, but maybe it's fwupd which does it..?

```
tree /esp                                                                                                                                                     
/esp
├── EFI
│   ├── arch
│   │   ├── initramfs-linux-fallback.img
│   │   ├── initramfs-linux.img
│   │   ├── intel-ucode.img
│   │   └── vmlinuz-linux
│   ├── BOOT
│   │   └── BOOTX64.EFI
│   └── systemd
│       └── systemd-bootx64.efi
└── loader
    ├── entries
    │   └── arch.conf
    └── loader.conf
```

3) No fwupdate efi vars according to:

```
efivar -l | grep fwupdate
```

I really appreciate your help :)

superm1, 2016-12-10:
> I'm not quite sure what that means to be honest. I have seen a few power related messages in the journal on early boot and shutdown. Where would I submit the bug report and what should it say. I'm [...]

Sure.  When you file the bug report, mention to the maintainer that "fwupd isn't compiled with upower support".  That should be enough. You can attach your verbose log too to show that to them.

> I'm not sure why there's entries for windows, I don't see them within the bios utility, just the single linux boot loader. shrugs

If you're not dual booting, feel free to delete that entry.

> No fwup I'm afraid, when I straced before I couldn't see fwupdmgr trying to write anything like that, but maybe it's fwupd which does it..?

So this is the most likely problem, the package `fwupdate` didn't install fwupx64.efi to the ESP.  As part of the installation of the CAB, it's supposed to use that to make an EFI entry to do the update on reboot.

GTB3NW, 2016-12-10:
> Sure. When you file the bug report, mention to the maintainer that "fwupd isn't compiled with upower support". That should be enough. You can attach your verbose log too to show that to them.

It's from the user repositories so it should be easy to get fixed, I can also recompile with the correct features. Thank you :)

> If you're not dual booting, feel free to delete that entry.

Is this done within the bios or is there a tool to perform this?

> So this is the most likely problem, the package fwupdate didn't install fwupx64.efi to the ESP. As part of the installation of the CAB, it's supposed to use that to make an EFI entry to do the update on reboot.

I've ran the tools with elevated privs so it should be able to write to those directories. Is there a command I can run manually to try and install the efi? At least then if that errors I can strace and see what's going on.

Cheers!

superm1, 2016-12-10:
> Is this done within the bios or is there a tool to perform this?

You can do it with efibootmgr.  The BIOS won't show entries that don't resolve to something real.

> I've ran the tools with elevated privs so it should be able to write to those directories. Is there a command I can run manually to try and install the efi? At least then if that errors I can strace and see what's going on.

So `fwupdate `assumes that it was properly installed (as in from `make install`).  That means it makes an assumption that `fwupx64.efi `is already on the ESP in the proper location.  In your install it's definitely not in the ESP.

Here is how upstream normally installs it when not packaged:
https://github.com/rhinstaller/fwupdate/blob/master/Make.defaults#L45
https://github.com/rhinstaller/fwupdate/blob/master/efi/Makefile#L100

Debian & Ubuntu both do it with a post-install script.  This is because the ESP might not be available until after install is done.
https://anonscm.debian.org/cgit/uefi/fwupdate.git/tree/debian/fwupdate.postinst.in#n40
https://anonscm.debian.org/cgit/uefi/fwupdate.git/tree/debian/scripts/install.in

Fedora installs it directly into the RPM.
https://github.com/rhinstaller/fwupdate/blob/master/fwupdate.spec.in#L126

GTB3NW, 2016-12-10:
Ahh that makes sense, I'll try get the maintainer to update. Here's the current PKGBUILD for the make section, would I add something here to get this to work correctly?

```
So this is the most likely problem, the package fwupdate didn't install fwupx64.efi to the ESP. As part of the installation of the CAB, it's supposed to use that to make an EFI entry to do the update on reboot.
```

superm1, 2016-12-10:
That's the `PKGBUILD `for `fwupd`, you'll be looking for the one for `fwupdate`.

GTB3NW, 2016-12-10:
Found the problem!

```
package() {
  cd "${srcdir}/${pkgname}-${pkgver}"

  make LIBDIR=/usr/lib EFIDIR=/usr/lib DESTDIR="${pkgdir}" libexecdir=/usr/lib/ install
  
  #don't install into /boot. copy files to /usr/lib/fwupdate for manual installation
  install -d ${pkgdir}/usr/lib/fwupdate
  mv ${pkgdir}/boot/efi/EFI ${pkgdir}/usr/lib/fwupdate/EFI
  rm -rf ${pkgdir}/boot
  rm -rf ${pkgdir}/usr/src
  rm -rf ${pkgdir}/usr/lib/debug
}
```

Which is why I couldn't see anything in /boot either, since the folders don't exist. Tad frustrating, will recompile and report back.

Also I've removed the boot loaders and didn't brick my system (Which is nice :P)

Thank you I really appreciate it!

GTB3NW, 2016-12-10:
Please could you confirm that the directory structure should be:

EFI/fw/fwupx64.efi

Do I need to make any other changes so that the system will boot with this? Or will it be handled by fwupdmgr?

Cheers

GTB3NW, 2016-12-10:
Looks like fwupdate by default tries to put files within /boot/efi, is there any way to change this to /esp? My /boot partition is actually a bind mount to /esp/EFI/arch. I suppose I could do a symlink back to the correct directory, would that work?

superm1, 2016-12-10:
EFIDIR needs to be properly set because it's used in fwupdate code too. I think it should be "Arch" most likely.

If you keep your esp in /esp I think you'll need to set up a symlink structure that matches a /boot/EFI structure. Fwupdate assumes it will be working in that type of structure. Otherwise expect to be laying other patches into fwupdate about using /esp.

That would be /boot/EFI/EFI/$EFIDIR/fwupx64.efi and make the directory /boot/EFI/EFI/$EFIDIR/fw for storing payloads.

GTB3NW, 2016-12-10:
See my edit, wont be able to symlink, hopefully I will be able to do something like bind mount. Will look into it.

I only really need to apply this patch once, I'm not too fussed for ongoing updates, if I am I'm happy to do a manual method if there is one. Is there any manual method?

superm1, 2016-12-10:
Is the /esp as a mount normal for arch, or something custom you did? If it's how arch normally does things it would be good to solve this more generally for the rest of arch users.

If it's just something odd you decided to do then sure there are manual ways to flash BIOS, but you'll hit this again when the next XPS 9360 bios is posted to LVFS again.

GTB3NW, 2016-12-10:
Lots of people just bundle everything into /boot, however the install guide specifically mentions a bind mount setup for a cleaner install, so I wont be the only person to have this setup. Currently in IRC to see if there's any workaround for this. If there is I'll add it as a note to the wiki for the mount bind setup and specifically for fwupd in the wiki

superm1, 2016-12-10:
Okay, so if a lot of people do this I agree it's worthwhile to come up with a good solution. This bug should probably be filed against https://GitHub.com/rhinstaller/fwupdate about coming up with a way to support this ESP not at /boot/EFI setup.

For now here is the manual way to do the update though if you want to apply it manually.
http://www.dell.com/support/article/us/en/4/SLN171755/en

GTB3NW, 2016-12-10:
Opened a bug: https://github.com/rhinstaller/fwupdate/issues/58

Hopefully that gets resolved. It will likely require a flag for fwupdmgr too right? I will look at applying the patch manually. Cheers!

superm1, 2016-12-10:
Somewhere I think it makes sense to try to detect this situation has happened and will cause problems, but otherwise there shouldn't be another bug in fwupd or fwupdmgr, it's all fwupdate.

GTB3NW, 2016-12-10:
No worries. I managed to apply the update manually (Yay my machine can wake up from suspend!)

I'll close this now since all discussion will move to fwupdate. I've added some notes to the arch wiki on this issue FYI

closed 2016-12-10 by GTB3NW