fwupd/fwupd #76: No updates detected for XPS 13 9360

cocreature, 2016-12-28:
When I run `fwupdmgr get-devices` I see my laptop listed as

```
XPS 13 9360
  Guid:                 48af7d21-3145-4f5e-b1b4-c8fc67b6e646
  DeviceID:             UEFI-48af7d21-3145-4f5e-b1b4-c8fc67b6e646-dev0
  Provider:             UEFI
  Flags:                internal|allow-offline|require-ac|supported
  Version:              0.1.0.7
  VersionLowest:        0.1.0.7
  Created:              2016-12-28
```

LVFS lists a [0.1.2.3](https://secure-lvfs.rhcloud.com/lvfs/device/5ffdbc0d-f340-441c-a803-8439c8c0ae10) update for that device however `fwupdmgr get-updates` doesn’t find it. I tried running refresh and then looking at the result of `fwupdmgr get-results` but that is telling me `Cannot get UEFI status for 48af7d21-3145-4f5e-b1b4-c8fc67b6e646`. I’m on Archlinux and I saw #75 but I installed `fwupdate` under `/boot` as follows
```
├── EFI
│   ├── Boot
│   │   ├── BOOTX64.EFI
│   │   ├── en-us
│   │   │   └── bootx64.efi.mui
│   │   └── shimx64.efi
│   ├── dell
│   │   ├── bios
│   │   │   └── recovery
│   │   │       └── bios_cur.rcv
│   │   └── logs
│   │       ├── diags_current.xml
│   │       └── diags_previous.xml
│   ├── EFI
│   │   └── Arch
│   │       ├── fw
│   │       └── fwupx64.efi
│   ├── Microsoft
│   │   └── Boot
│   │       ├── BCD
│   │       ├── BCD.LOG
│   │       ├── en-us
│   │       ├── Fonts
│   │       │   ├── chs_boot.ttf
│   │       │   ├── cht_boot.ttf
│   │       │   ├── jpn_boot.ttf
│   │       │   ├── kor_boot.ttf
│   │       │   ├── malgun_boot.ttf
│   │       │   ├── meiryo_boot.ttf
│   │       │   ├── msjh_boot.ttf
│   │       │   ├── msyh_boot.ttf
│   │       │   ├── segmono_boot.ttf
│   │       │   ├── segoe_slboot.ttf
│   │       │   └── wgl4_boot.ttf
│   │       └── Resources
│   │           └── bootres.dll
│   ├── systemd
│   │   └── systemd-bootx64.efi
│   └── ubuntu
│       ├── fw
│       ├── fwupx64.efi
│       ├── grub.cfg
│       ├── grubx64.efi
│       ├── MokManager.efi
│       └── shimx64.efi
├── en-us
│   └── bootmgr.efi.mui
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── intel-ucode.img
├── loader
│   ├── entries
│   │   ├── arch.conf
│   │   └── arch-mainline.conf
│   └── loader.conf
└── vmlinuz-linux
```

`efivar -l` does not show any entries for `fwupate`, neither does `efibootmgr -v`.

Finally here is the output of `fwupd -v`
```
21:27:50	Verbose debugging enabled (on console 1)
21:27:50	adding plugin /usr/lib/fwupd-plugins-1/libfu_plugin_steelseries.so
21:27:50	adding plugin /usr/lib/fwupd-plugins-1/libfu_plugin_test.so
21:27:50	performing init() on test
21:27:51	Loading fallback values from /etc/fwupd.conf
21:27:51	adding udev device: /sys/devices/pci0000:00/0000:00:02.0
21:27:51	using 3ec3df3a-2290-56e5-9d2f-eda62e9ab50b for 0x8086:0x5916
21:27:51	emit added from Udev: ro__sys_devices_pci0000_00_0000_00_02_0
21:27:51	emit added from UEFI: UEFI-48af7d21-3145-4f5e-b1b4-c8fc67b6e646-dev0
21:27:51	UEFI-48af7d21-3145-4f5e-b1b4-c8fc67b6e646-dev0 has no firmware updates
Failed to coldplug: Dell: firmware updating not supported (0)
21:27:51	Failed to coldplug: Dell: firmware updating not supported (0)
21:27:51	using f62863ee-133d-511b-903c-d6dfa8f9fdf4 for USB\VID_0BDA&PID_8153
21:27:51	using 3ee73ec0-43ce-5b01-aed8-ce23fd193367 for USB\VID_0BDA&PID_8153&REV_3000
21:27:51	emit added from USB: usb:00:01:04
21:27:51	using 7a51fe38-0180-525c-af99-90ad3e5432a2 for USB\VID_0835&PID_2A01
21:27:51	using 62b00e1a-c22f-56f2-b8c0-161d30964345 for USB\VID_0835&PID_2A01&REV_0136
21:27:51	emit added from USB: usb:00:01:03
21:27:51	using 9efb54ea-0352-537a-9bfe-0e55e2ace999 for USB\VID_2341&PID_8036
21:27:51	using ebb140fe-1fb4-54cf-9bcf-15012ed3fe9c for USB\VID_2341&PID_8036&REV_0100
21:27:51	emit added from USB: usb:00:01:01
21:27:51	using 4ec19583-c132-5c27-a9cf-061419e838dc for USB\VID_0C45&PID_670C
21:27:51	using 6b31e6e6-ba7c-5251-8f4c-7f2571612251 for USB\VID_0C45&PID_670C&REV_5626
21:27:51	emit added from USB: usb:00:05
21:27:51	using 31cc46f6-9bdc-5b1e-b48b-1c791781dcdc for USB\VID_04F3&PID_20D0
21:27:51	using 9fd88db6-a96d-505b-bbea-92aee9d07455 for USB\VID_04F3&PID_20D0&REV_1112
21:27:51	emit added from USB: usb:00:04
21:27:51	no product string descriptor
######################                                                                 @0193ms AsStore:load 193ms
######################                                                                 @0193ms AsStore:load{per-system} 193ms
############                                                                           @0108ms AsStore:store-from-file{/usr/share/app-info/xmls/extra.xml.gz} 107ms
              #                                                                        @0124ms AsStore:store-from-file{/usr/share/app-info/xmls/core.xml.gz} 5ms
              ######                                                                   @0180ms AsStore:store-from-file{/usr/share/app-info/xmls/community.xml.gz} 55ms
                        #############################################################  @0726ms FuMain:coldplug 520ms
                        #####                                                          @0254ms FuMain:coldplug{Dell} 44ms
                              #######################################################  @0726ms FuMain:coldplug{USB} 471ms
                              ####################                                     @0431ms FuProviderUsb:added{0bda:8153} 176ms
                                                   ########                            @0500ms FuProviderUsb:added{0835:2a01} 69ms
                                                           #############               @0616ms FuProviderUsb:added{0c45:670c} 114ms
                                                                         ############  @0726ms FuProviderUsb:added{0cf3:e300} 109ms
21:27:51	devices now in store:
21:27:51	1	com.steelseries.rival-legacy.firmware	SteelSeries Firmware
21:27:51	2	UEFI-dummy-dev0	UEFI Updates
21:27:51	3	com.via.VL811.firmware	VL811 Firmware
21:27:51	4	com.via.VL811+.firmware	VL811+ Firmware
21:27:51	5	com.via.VL812.firmware	VL812 Firmware
21:27:51	6	com.via.VL812_B2.firmware	VL812 B2 Firmware
21:27:51	7	com.8bitdo.fc30.firmware	FC30
21:27:51	8	com.8bitdo.fc30arcade.firmware	FC30 Joystick
21:27:51	9	com.8bitdo.fc30pro.firmware	FC30 Pro
21:27:51	10	com.8bitdo.nes30.firmware	NES30
21:27:51	11	com.8bitdo.nes30pro.firmware	NES30 Pro
21:27:51	12	com.8bitdo.sfc30.firmware	SFC30
21:27:51	13	com.8bitdo.snes30.firmware	SNES30
21:27:51	14	com.dell.uefi124c207d.firmware	XPS 15 9550/Precision 5510 System Update
21:27:51	15	com.dell.uefi293af847.firmware	OptiPlex 3050 System Update
21:27:51	16	com.dell.uefi33773727.firmware	XPS 13 9350 System Update
21:27:51	17	com.dell.uefi3c20b9e1.firmware	OptiPlex 7450 AIO System Update
21:27:51	18	com.dell.uefi43ca3264.firmware	OptiPlex 7440 AIO System Update
21:27:51	19	com.dell.uefi48af7d21.firmware	XPS 13 9360 System Update
21:27:51	20	com.dell.uefi49e03513.firmware	Latitude E5X80 System Update
21:27:51	21	com.dell.uefi4fed6c9d.firmware	OptiPlex 7050 System Update
21:27:51	22	com.dell.uefi5ffdbc0d.firmware	XPS 13 9360 System Update
21:27:51	23	com.dell.uefi8080d214.firmware	OptiPlex 5050 System Update
21:27:51	24	com.dell.uefi8b7b32a7.firmware	Latitude E5X80 System Update
21:27:51	25	com.dell.uefia0a3aa54.firmware	Embedded Box PC 5000 System Update
21:27:51	26	com.dell.uefiaee2604a.firmware	Embedded Box PC 3000 System Update
21:27:51	27	com.dell.uefib566a9b1.firmware	OptiPlex 5250 AIO System Update
21:27:51	28	com.dell.uefid69fed57.firmware	OptiPlex 3050 AIO System Update
21:27:51	29	com.dell.uefie0f614ed.firmware	Edge Gateway 5000/5100 System Update
21:27:51	30	com.hughski.ColorHug.firmware	ColorHug
21:27:51	31	com.hughski.ColorHug2.firmware	ColorHug2
21:27:51	32	com.hughski.ColorHugALS.firmware	ColorHugALS
21:27:51	UEFI-48af7d21-3145-4f5e-b1b4-c8fc67b6e646-dev0 has no firmware updates
21:27:51	FuMain: acquired name: org.freedesktop.fwupd
```

superm1, 2016-12-28:
There was another report of this particular failure elsewhere. Dell is still on vacation so this hasn't been looked at yet.
 Can you please manually download the CAB file and run `fwupmgr get-details` to confirm the GUIDs match?

cocreature, 2016-12-28:
Thanks for the quick response!

Here’s the result:
```
sudo fwupdmgr get-details ~/Downloads/95b37375ce6af8a15abffe11155601a7d2ba03ba-firmware.cab
XPS 13 9360 System Update
  Guid:                 5ffdbc0d-f340-441c-a803-8439c8c0ae10
  UniqueID:             */*/c04856f7dfe57e98601d69eea88c752d1f4bef21/firmware/com.dell.uefi5ffdbc0d.firmware/*
  DeviceID:
  Description:          <p>Updating the system firmware improves performance.</p>
  Flags:                none
  AppstreamId:          com.dell.uefi5ffdbc0d.firmware
  Summary:              Firmware for the Dell XPS 13 9360
  UpdateVersion:        0.1.2.3
  FilenameCab:          firmware.bin
  License:              proprietary
  Size:                 11.3 MB
  UrlHomepage:          http://support.dell.com/
  Vendor:               Dell Inc.
  Trusted:              payload
```
I’m not quite sure which GUIDs are supposed to match.

superm1, 2016-12-28:
It looks to me they aren't matching.  Dell will have to investigate to figure out what happened here.

In the interim you can manually apply it by using the method of putting the exe on a fat32 partition with the F12 boot menu. If you do, I would be curious to see what `fwupdmgr get-devices` shows.

superm1, 2017-01-03:
@cocreature The error on `# fwupdmgr get-results` is expected because the update wasn't performed using UEFI capsule.  That will only update if a capsule is applied.

I've confirmed that the ESRT GUID did change between BIOS 1.0.7 and 1.2.3.  Dell is internally discussing how to resolve this, but until a solution is determined anyone who encounters this can flash the BIOS to 1.2.3 or later using this method: http://www.dell.com/support/article/us/en/19/SLN171755/en#updatebios2015

Since this isn't a problem with fwupd, but it's a Dell problem i'm going to close the issue on this project.   I'll leave a comment here though if Dell comes up with a solution.

closed 2017-01-03 by superm1

superm1, 2017-04-14:
Since the ESRT GUID did change between 1.0.7 and 1.2.3 there is a transition firmware on LVFS that should bridge between the two versions.