fwupd/fwupd #77: fwupd appears as preventing autosuspend of USB device

sourcejedi, 2017-01-04:
fwupd appears as preventing autosuspend of a USB device, and there is no explanation logged.

fwupd should aspire not to appear to cause USB devices to fail to autosuspend. If there is a hardware limitation which requires that the device be held open, IMO this should be considered a quirk and logged, so that the causes (and perhaps workarounds) are possible to discover.

The USB device is the built-in bluetooth device on a Dell Lattitude E5450.  

Running powertop and tabbing over to "Device stats", I see 100% usage - which means no autosuspend - for "USB device: usb-device-8087-0a2a". The driver supports autosuspend. However the device that contains the btusb endpoints is considered used.

```
$ cd /sys/bus/usb/devices/1-1.3  # i don't know the logic to this
$ cat idVendor
8087
$ cat idProduct 
0a2a
$ readlink driver */driver
../../../../../../bus/usb/drivers/usb
../../../../../../../bus/usb/drivers/btusb
../../../../../../../bus/usb/drivers/btusb
$ cd power
$ cat level
auto
$ cat autosuspend
2
$ cat runtime_active_kids 
0
$ cat runtime_usage
1
```

There are no bluetooth sockets

```
$ netstat -a
...
Active Bluetooth connections (servers and established)
Proto  Destination       Source            State         PSM DCID   SCID      IMTU    OMTU Security
Proto  Destination       Source            State     Channel
$
```

`lsof` shows that `fwupd` is holding open the usb device (the parent device to the two btusb interfaces):

```
$ sudo lsof +D /dev/bus/usb
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
COMMAND  PID USER   FD   TYPE DEVICE SIZE/OFF  NODE NAME
fwupd   2186 root   28u   CHR  189,2      0t0 11356 /dev/bus/usb/001/003
```

There are no relevant fwupd messages in the journal.

```
$ sudo journalctl -b|grep fwupd
Jan 04 17:11:14 alan-laptop dbus-daemon[863]: [system] Activating service name='org.freedesktop.fwupd' requested by ':1.63' (uid=1000 pid=1911 comm="/usr/bin/gnome-software --gapplication-service " label="unconfined_u:unconfined_r:unconfined_t:s0-s0:c0.c1023") (using servicehelper)
Jan 04 17:11:15 alan-laptop fwupd[2186]: Failed to coldplug: UEFI firmware updating not supported
Jan 04 17:11:15 alan-laptop fwupd[2186]: Failed to coldplug: Dell: UEFI capsule firmware updating not supported (0)
Jan 04 17:11:15 alan-laptop dbus-daemon[863]: [system] Successfully activated service 'org.freedesktop.fwupd'
```

I looked for further identification of the bluetooth device.  I didn't find any more identifiers, but I did notice a firmware is applied repeatedly, including on resume from suspend.  I don't know if this is fwupd at work.

$ dmesg|grep Blue
...
[   42.228867] Bluetooth: RFCOMM ver 1.11
[  429.338230] Bluetooth: hci0: read Intel version: 370810011003110e00
[  429.861139] Bluetooth: hci0: Intel Bluetooth firmware file: intel/ibt-hw-37.8.10-fw-1.10.3.11.e.bseq
[  430.181256] Bluetooth: hci0: Intel Bluetooth firmware patch completed and activated
[ 1935.692928] Bluetooth: hci0: read Intel version: 370810011003110e25
[ 1935.692929] Bluetooth: hci0: Intel device is already patched. patch num: 25
[ 4049.191217] Bluetooth: hci0: read Intel version: 370810011003110e00
[ 4049.191344] Bluetooth: hci0: Intel Bluetooth firmware file: intel/ibt-hw-37.8.10-fw-1.10.3.11.e.bseq
[ 4049.514295] Bluetooth: hci0: Intel Bluetooth firmware patch completed and activated
[ 4971.657207] Bluetooth: hci0: read Intel version: 370810011003110e00
[ 4971.657211] Bluetooth: hci0: Intel Bluetooth firmware file: intel/ibt-hw-37.8.10-fw-1.10.3.11.e.bseq
[ 4971.977217] Bluetooth: hci0: Intel Bluetooth firmware patch completed and activated

hughsie, 2017-01-04:
What version fwupd?

sourcejedi, 2017-01-04:
Sorry!  `fwupd-0.7.5-1.fc25.x86_64`.

hughsie, 2017-01-05:
I think this is fixed by a02bbb1c4ebbdb6831f802b7a3578c56cab3ca64 and 0e77c525dddd22722b8524d8b75c4ef58059b620 -- we just need a release and then we should be good.

sourcejedi, 2017-01-05:
Yes.  0.7.5 shows the issue, 0.7.5-72-g4852ff8 does not.

superm1, 2017-01-09:
Since this is fixed in current master, i'll close this issue.

closed 2017-01-09 by superm1