fwupd/fwupd #79: fwupd is started under dbus.service, so fwupd.service has no effect

sourcejedi, 2017-01-08:
fwupd-0.7.5-1.fc25.x86_64

Stopping, masking, or amending fwupd.service (e.g. as a temporary workaround for #77) has no effect, because

$ systemctl status fwupd.service
● fwupd.service - Firmware update daemon
   Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static; vendor preset:
   Active: inactive (dead)
     Docs: http://www.fwupd.org/
$ systemctl status dbus.service
● dbus.service - D-Bus System Message Bus
   Loaded: loaded (/usr/lib/systemd/system/dbus.service; static; vendor preset: 
   Active: active (running) since Sat 2017-01-07 10:59:56 GMT; 1 day 9h ago
     Docs: man:dbus-daemon(1)
 Main PID: 868 (dbus-daemon)
    Tasks: 14 (limit: 4915)
   CGroup: /system.slice/dbus.service
           ├─  868 /usr/bin/dbus-daemon --system --address=systemd: --nofork --n
           ├─ 2208 /usr/sbin/abrt-dbus -t133
           └─32326 /usr/libexec/fwupd/fwupd

superm1, 2017-01-09:
I believe this should have been fixed initially in https://github.com/hughsie/fwupd/commit/e34654e17fc968bf47266c83e374af70deac869a however there were some negative side effects to that which required some other changes.

The net result is this should be fixed in current master. Can you update to current master and see?

Thanks

sourcejedi, 2017-01-09:
master does show fwupd running under fwupd.service.  There is at least one difference which seems like a defect to me.

I was able to use `rpm -e --nodeps fwupd && make install`.  

1:  SELinux refused to allow fwupd to start.  I tried `restorecon -Rv /`, and I was able to see /usr/lib/fwupd/fwupd labelled as fwupd_exec_t.  However I didn't manage to get fwupd to start without also using audit2allow.  This generated a `my-fwupd.te` looking like this:

```
module my-fwupd3 1.0;

require {
	type init_t;
	type fwupd_cache_t;
	class dir read;
	class file { getattr open read };
}

#============= init_t ==============
allow init_t fwupd_cache_t:dir read;
allow init_t fwupd_cache_t:file open;
allow init_t fwupd_cache_t:file { getattr read };
```

2:  The first time I call `fwupd get-devices`, it tells me there aren't any. This appears to be a regression.

$ fwupdmgr get-devices
No detected devices: Nothing to do

3:  `fwupdmgr get-devices` output is now much shorter :).  The version in master no longer shows any of my built-in USB devices, only my "HD Graphics 5500" by Intel.

I don't know if this is intentional.  For example, I have a bluetooth device, where the kernel which the kernel log already shows as loading a firmware file, "intel/ibt-hw-37.8.10-fw-1.10.3.11.e.bseq" from /lib/firmware.  My logs suggest this is persistent firmware, but it is patched by the kernel firmware loader.

superm1, 2017-01-09:
That's definitely not intentional that USB devices disappear.

Can you modify your unit file for fwupd to launch it with `-v `and then share relevant `journalctl `output?  It sounds like the udev plugin might not be loading properly, but we'll need to see more about it.

At least on my system (_Ubuntu 16.04_) I'm able to see all USB devices with current master.

There was specifically a problem with USB devices not working a result of that commit above, but it was fixed with https://github.com/hughsie/fwupd/commit/803e4808afd9dfdb8db0571d6dd527bc02441cea.

hughsie, 2017-01-09:
Do the USB devices come back with selinux off?

sourcejedi, 2017-01-12:
OK, the USB devices came back.  So _now_ I have a result that tells me #77 is fixed :).

I hadn't hammered selinux quite enough, sorry.  For some reason earlier denial messages from setroubleshoot had appeared in `systemctl status {dbus,fwupd}` - specifically under dbus.service - but not the subsequent denial messages about USB devices.

I assume the root cause is that when fwupd is started as its own service, it fails to transition from `init_t` to `fwupd_exec_t`. Although the binary is labelled as such, I guess selinux transitions also depend on the selinux policy, and the F25 selinux policy does not support this.

> Jan 12 21:31:07 alan-laptop audit: SELINUX_ERR op=security_bounded_transition seresult=denied oldcontext=system_u:system_r:init_t:s0 newcontext=system
_u:system_r:fwupd_t:s0

sourcejedi, 2017-01-12:
Difference 2) remains.  It seems odd for `fwupdmgr` to behave so unfriendly.

superm1, 2017-01-16:
@sourcejedi If you're still having fwupdmgr not immediately show devices, can you share journalctl logs from a verbose start of fwupd?

superm1, 2017-01-20:
Since the second issue was fixed, and your first issue was caused by SELinux, I'm going to mark this one closed.

closed 2017-01-20 by superm1