fwupd/fwupd #85: synapticsmst: Add udev support

superm1, 2017-01-27:
Per a discussion with the intel-gfx folks that maintain the drm-dp-aux kernel module, they said that hotplug uevents for a monitor should happen every time a monitor appears/disappears [1].

> Hotplug uevents should be fired off whenever a new display
> appears/disappears.
> 

That sounds like the right time to re-do enumeration. @T1323 would you be able to add that in?

[1] https://lists.freedesktop.org/archives/intel-gfx/2017-January/117994.html

hughsie, 2017-01-27:
I would much perfer using udev for getting the device nodes! Thanks for looking into this.

superm1, 2017-01-27:
Yeah me too.  I'm not yet sure if udev will actually indicate the drm-dp-aux node to use, but at least using that to figure out when to run re-enumeration sounds better.

hughsie, 2017-01-27:
g_udev_device_get_device_file() should give you the node to use, although you might have to walk up to the parent device (or grandparent) -- I think one of the other plugins (logitech?) does this already.

superm1, 2017-01-31:
I've done some experimenting with what info can be gleamed from udev, but it's still not perfect, or even much better than what you get from walking the sysfs directory for drm_dp_aux nodes.

Plugging in a dock with no monitors here's everything that udev will spit out:
```
# udevadm monitor
KERNEL[638595.584753] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0 (pci)
UDEV  [638595.585608] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0 (pci)
KERNEL[638595.596185] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/pci_bus/0000:07 (pci_bus)
KERNEL[638595.596504] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:00.0 (pci)
KERNEL[638595.596964] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:01.0 (pci)
UDEV  [638595.597072] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/pci_bus/0000:07 (pci_bus)
KERNEL[638595.597135] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0 (pci)
KERNEL[638595.597287] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:00.0/pci_bus/0000:08 (pci_bus)
KERNEL[638595.597374] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:01.0/pci_bus/0000:09 (pci_bus)
KERNEL[638595.597483] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/pci_bus/0000:3e (pci_bus)
UDEV  [638595.597768] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:00.0 (pci)
KERNEL[638595.597888] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0 (pci)
UDEV  [638595.598003] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0 (pci)
UDEV  [638595.598232] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:01.0 (pci)
UDEV  [638595.598273] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:00.0/pci_bus/0000:08 (pci_bus)
KERNEL[638595.598369] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:06:00.0:pcie108 (pci_express)
KERNEL[638595.598632] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:00.0/0000:07:00.0:pcie208 (pci_express)
UDEV  [638595.598730] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/pci_bus/0000:3e (pci_bus)
UDEV  [638595.598884] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:01.0/pci_bus/0000:09 (pci_bus)
KERNEL[638595.598902] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:01.0/0000:07:01.0:pcie208 (pci_express)
KERNEL[638595.599154] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:07:02.0:pcie208 (pci_express)
UDEV  [638595.600059] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0 (pci)
UDEV  [638595.600081] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:07:02.0:pcie208 (pci_express)
UDEV  [638595.600094] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:01.0/0000:07:01.0:pcie208 (pci_express)
UDEV  [638595.600106] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:00.0/0000:07:00.0:pcie208 (pci_express)
UDEV  [638595.600321] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:06:00.0:pcie108 (pci_express)
KERNEL[638595.601057] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb3 (usb)
KERNEL[638595.601092] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb3/3-0:1.0 (usb)
KERNEL[638595.601325] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb4 (usb)
KERNEL[638595.601361] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb4/4-0:1.0 (usb)
UDEV  [638596.113240] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb4 (usb)
UDEV  [638596.113615] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb3 (usb)
UDEV  [638596.117491] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb4/4-0:1.0 (usb)
UDEV  [638596.118182] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb3/3-0:1.0 (usb)
KERNEL[638596.264835] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb4/4-1 (usb)
KERNEL[638596.265505] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb4/4-1/4-1:1.0 (usb)
KERNEL[638596.524687] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb3/3-1 (usb)
KERNEL[638596.525196] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb3/3-1/3-1:1.0 (usb)
KERNEL[638596.629397] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb4/4-1/4-1.2 (usb)
KERNEL[638596.630368] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb4/4-1/4-1.2/4-1.2:2.0 (usb)
KERNEL[638596.631548] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb4/4-1/4-1.2/4-1.2:2.1 (usb)
KERNEL[638596.631746] remove   /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb4/4-1/4-1.2/4-1.2:2.0 (usb)
KERNEL[638596.631818] remove   /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb4/4-1/4-1.2/4-1.2:2.1 (usb)
KERNEL[638596.633706] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb4/4-1/4-1.2/4-1.2:1.0 (usb)
UDEV  [638596.789032] add      /devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb4/4-1 (usb)
[... cut ...]
This is the node that the MST hub actually hangs off of in udev's DB:
```
P: /devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-1/drm_dp_aux1
N: drm_dp_aux1
E: DEVNAME=/dev/drm_dp_aux1
E: DEVPATH=/devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-1/drm_dp_aux1
E: MAJOR=245
E: MINOR=1
E: SUBSYSTEM=drm_dp_aux_dev
```

Given that, I'm inclined to think that the only real value in adding udev is to get change events from when monitors are plugged in to redo enumeration.  Any other thoughts from your side?

superm1, 2017-02-03:
I've done more experimenting with this and don't think it's a good idea to add gudev in. The whole thing is too racy with intermingled dependencies.
Dock MST hubs won't show up until a recoldpug (from coldplug_prepare).  So if you just drop gusb and look for a change event related to new display outputs, you won't know that you need to do a recoldplug when plugging in the dock. 

So if you use usb to schedule a recoldplug, then you get into an endless loop of executing recoldplugs. The coldplug_prepare from usb turns on MST hubs which generates change events.  By the time that the coldplug from gusb is done the coldplug from the change events will go off.  Furthermore too many change events come in at once, so you need a timer to  buffer them (similar to how fu-plugin-thunderbolt worked).  

This is my current working patch, but I'm leaning upon just dropping it and not using gudev at all.

```
diff --git a/plugins/synapticsmst/Makefile.am b/plugins/synapticsmst/Makefile.am
index d1577e7..f12baec 100644
--- a/plugins/synapticsmst/Makefile.am
+++ b/plugins/synapticsmst/Makefile.am
@@ -1,6 +1,7 @@
 AM_CPPFLAGS = 							\
 	$(APPSTREAM_GLIB_CFLAGS)				\
 	$(GLIB_CFLAGS)						\
+	$(GUDEV_CFLAGS)						\
 	$(GUSB_CFLAGS)						\
 	$(LIBSMBIOS_CFLAGS)					\
 	$(PIE_CFLAGS)						\
@@ -29,6 +30,7 @@ libfu_plugin_synapticsmst_la_SOURCES =				\
 
 libfu_plugin_synapticsmst_la_LIBADD =				\
 	$(GUSB_LIBS)						\
+	$(GUDEV_LIBS)						\
 	$(LIBSMBIOS_LIBS)					\
 	$(GLIB_LIBS)
 libfu_plugin_synapticsmst_la_LDFLAGS = -module -avoid-version
diff --git a/plugins/synapticsmst/fu-plugin-synapticsmst.c b/plugins/synapticsmst/fu-plugin-synapticsmst.c
index 8e18177..672e91d 100644
--- a/plugins/synapticsmst/fu-plugin-synapticsmst.c
+++ b/plugins/synapticsmst/fu-plugin-synapticsmst.c
@@ -28,9 +28,17 @@
 #include "fu-dell-common.h"
 #include "fu-plugin.h"
 #include "fu-plugin-vfuncs.h"
+#include <gudev/gudev.h>
 
+#define SYNAPTICS_RESCAN_DELAY 3000 /* ms */
 #define SYNAPTICS_FLASH_MODE_DELAY 3
 
+struct FuPluginData {
+	GUdevClient             *gudev_client;
+	guint			refresh_id;
+};
+
+
 static gboolean
 synapticsmst_common_check_supported_system (GError **error)
 {
@@ -331,6 +339,33 @@ fu_plugin_update_online (FuPlugin *plugin,
 
 	return TRUE;
 }
+static gboolean
+fu_plugin_synapticsmst_schedule_rescan_cb (gpointer user_data)
+{
+	FuPlugin *plugin = FU_PLUGIN (user_data);
+	FuPluginData *data = fu_plugin_get_data (plugin);
+
+	/* no longer valid */
+	data->refresh_id = 0;
+
+	g_debug ("Timer executed!");
+	/* rescan */
+	fu_plugin_recoldplug (plugin);
+	return FALSE;
+}
+
+static void
+fu_plugin_synapticsmst_schedule_rescan (FuPlugin *plugin)
+{
+	FuPluginData *data = fu_plugin_get_data (plugin);
+
+	/* wait for change events to settle, this will reset them */
+	if (data->refresh_id != 0)
+		g_source_remove (data->refresh_id);
+	data->refresh_id = g_timeout_add (SYNAPTICS_RESCAN_DELAY,
+					  fu_plugin_synapticsmst_schedule_rescan_cb,
+					  plugin);
+}
 
 static void
 fu_plugin_synapticsmst_redo_enumeration_cb (GUsbContext *ctx,
@@ -348,7 +383,17 @@ fu_plugin_synapticsmst_redo_enumeration_cb (GUsbContext *ctx,
 		return;
 
 	/* Request daemon to redo coldplug, this wakes up Dell devices */
-	fu_plugin_recoldplug (plugin);
+	fu_plugin_synapticsmst_schedule_rescan (plugin);
+}
+
+static void
+fu_plugin_udev_uevent_cb (GUdevClient *gudev_client,
+			  const gchar *action,
+			  GUdevDevice *udev_device,
+			  FuPlugin *plugin)
+{
+	if (g_strcmp0 (action, "change") == 0)
+		fu_plugin_synapticsmst_schedule_rescan (plugin);
 }
 
 gboolean
@@ -366,6 +411,25 @@ fu_plugin_startup (FuPlugin *plugin, GError **error)
 	return TRUE;
 }
 
+void
+fu_plugin_init (FuPlugin *plugin) {
+	FuPluginData *data = fu_plugin_alloc_data (plugin, sizeof (FuPluginData));
+	const gchar *subsystems[] = { "drm", NULL };
+
+	data->gudev_client = g_udev_client_new (subsystems);
+	g_signal_connect (data->gudev_client, "uevent",
+			  G_CALLBACK (fu_plugin_udev_uevent_cb), plugin);
+}
+
+void
+fu_plugin_destroy (FuPlugin *plugin)
+{
+	FuPluginData *data = fu_plugin_get_data (plugin);
+	if (data->refresh_id != 0)
+		g_source_remove (data->refresh_id);
+	g_object_unref (data->gudev_client);
+}
+
 gboolean
 fu_plugin_coldplug (FuPlugin *plugin, GError **error)
 {
```

hughsie, 2017-02-06:
I think half the problem is that the g_usleep() in the prepare loop isn't allowing the mainloop to run, so you don't get signals emitted. I don't think we want to spin the main loop here, there's too much other stuff that could break. Lets leave what works in git and revisit if/when required.

closed 2017-02-06 by superm1