fwupd/fwupd #86: Unclassed pointers

superm1, 2017-02-03:
I've been noticing lately that unplugging WD15 or TB16 with fwupd running is getting a bunch of this in the debug output for fwupd.

```
03:49:57:0341 Fu  UEFI-124c207d-5db8-4d95-bd31-34fd971b34f9-dev0 has no firmware updates
03:49:57:0341 GLib-GObject invalid unclassed pointer in cast to 'FwupdResult'
03:49:57:0341 libfwupd fwupd_result_get_device_version: assertion 'FWUPD_IS_RESULT (result)' failed
03:49:57:0341 GLib-GObject invalid unclassed pointer in cast to 'FwupdResult'
03:49:57:0341 libfwupd fwupd_result_get_device_version: assertion 'FWUPD_IS_RESULT (result)' failed
03:49:57:0341 Fu  Called GetUpdates()
03:49:57:0341 Fu  UEFI-124c207d-5db8-4d95-bd31-34fd971b34f9-dev0 has no firmware updates
03:49:57:0341 GLib-GObject invalid unclassed pointer in cast to 'FwupdResult'
03:49:57:0341 libfwupd fwupd_result_get_device_version: assertion 'FWUPD_IS_RESULT (result)' failed
03:49:57:0341 GLib-GObject invalid unclassed pointer in cast to 'FwupdResult'
03:49:57:0341 libfwupd fwupd_result_get_device_version: assertion 'FWUPD_IS_RESULT (result)' failed
03:50:00:0447 Fu  Called GetUpdates()
03:50:00:0447 Fu  UEFI-124c207d-5db8-4d95-bd31-34fd971b34f9-dev0 has no firmware updates
03:50:00:0447 GLib-GObject invalid unclassed pointer in cast to 'FwupdResult'
03:50:00:0447 libfwupd fwupd_result_get_device_version: assertion 'FWUPD_IS_RESULT (result)' failed
03:50:00:0447 GLib-GObject invalid unclassed pointer in cast to 'FwupdResult'
03:50:00:0447 libfwupd fwupd_result_get_device_version: assertion 'FWUPD_IS_RESULT (result)' failed
03:50:00:0498 Fu  Called GetUpdates()
03:50:00:0498 Fu  UEFI-124c207d-5db8-4d95-bd31-34fd971b34f9-dev0 has no firmware updates
03:50:00:0498 GLib-GObject invalid unclassed pointer in cast to 'FwupdResult'
03:50:00:0498 libfwupd fwupd_result_get_device_version: assertion 'FWUPD_IS_RESULT (result)' failed
03:50:00:0498 GLib-GObject invalid unclassed pointer in cast to 'FwupdResult'
03:50:00:0498 libfwupd fwupd_result_get_device_version: assertion 'FWUPD_IS_RESULT (result)' failed

``` 

I suspect it's some pointer that is freed by the Dell or Synaptics plugin but shouldn't have been.  Can you give it a glance to see if it stands out to you?

superm1, 2017-02-03:
At first I was pointing fingers at the synaptics plugin, but now it's seeming more like udev.  Axing the udev plugin library but keeping all the others used when a dock is around (thunderbolt, synaptics, dell) I haven't gotten it reproduced at all.

In terms of reproducing if it does turn out to be caused by something synaptics is doing too, I think you might be able to use the new test functionality in synaptics to reproduce, even without physically having a dock.
1) Copy all the tests somewhere.
2) Launch fwupd like this:
sudo FWUPD_SYNAPTICSMST_FW_DIR=`pwd`/tests/tb16_dock /usr/lib/fwupd/fwupd  -v
3) After fwupd is done with the coldplug, switch out the no_device drm_dp_aux# files into the tb16_dock files.
4) Get another plugin to initiate a recoldplug (stage this in various ways - i've done with adding a USB dongle and making a plugin call for a new coldplug on that VID/PID).  That should cause removal of the rest of the nodes.

hughsie, 2017-02-03:
So I couldn't see anything fishy in the mst plugin; I did identify a daemon issue it might be, although it's a long shot. I've added 6ad951d27c96337e499dd60e710b33dbc7aa1af8 to see if it's that. I'm on the wrong laptop today, so if you could upload a daemon log (as in the complete --verbose output) when you see the invalid unclassed pointer message that would be great.

Also, if you have debuginfo installed, doing `export G_DEBUG=fatal-criticals` and then getting a backtrace on the first instance would be great. It might even be possible to get the ID of the device, for instance doing `p fu_device_get_id(dev)` in the right stack frame. Thanks!

superm1, 2017-02-03:
Nope wasn't that commit you added.  It is backtracing into the udev plugin though.
  
```# gdb --args /usr/lib/fwupd/fwupd -v
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.04) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/lib/fwupd/fwupd...(no debugging symbols found)...done.
(gdb) run
Starting program: /usr/lib/fwupd/fwupd -v
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
13:57:13:0706 Fu  Verbose debugging enabled (on console 1)
13:57:13:0706 As  run AsStore:load
13:57:13:0706 As  run AsStore:load{per-user}
13:57:13:0706 As  run AsStore:load{per-system}
13:57:13:0706 As  run AsStore:app-info{/usr/share/app-info/xmls/org.freedesktop.fwupd.xml}
13:57:13:0714 As  run AsStore:store-from-file
[New Thread 0x7ffff019c700 (LWP 3682)]
13:57:13:0715 As  run AsStore:store-from-root
13:57:13:0715 As  using icon path /usr/share/app-info/icons/fwupd
13:57:13:0715 As  run AsStore:match-addons
13:57:13:0715 As  adding existing file: /usr/share/app-info/xmls/org.freedesktop.fwupd.xml
13:57:13:0715 As  run AsStore:app-info{/var/lib/app-info/yaml/security.ubuntu.com_ubuntu_dists_xenial-security_universe_dep11_Components-amd64.yml.gz}
13:57:13:0715 As  run AsStore:store-from-file
13:57:13:0719 As  run AsStore:app-info{/var/lib/app-info/yaml/archive.ubuntu.com_ubuntu_dists_xenial_multiverse_dep11_Components-amd64.yml.gz}
13:57:13:0719 As  run AsStore:store-from-file
13:57:13:0724 As  run AsStore:app-info{/var/lib/app-info/yaml/archive.ubuntu.com_ubuntu_dists_xenial-backports_multiverse_dep11_Components-amd64.yml.gz}
13:57:13:0724 As  run AsStore:store-from-file
13:57:13:0724 As  run AsStore:app-info{/var/lib/app-info/yaml/archive.ubuntu.com_ubuntu_dists_xenial-proposed_main_dep11_Components-amd64.yml.gz}
13:57:13:0724 As  run AsStore:store-from-file
13:57:13:0727 As  run AsStore:app-info{/var/lib/app-info/yaml/security.ubuntu.com_ubuntu_dists_xenial-security_restricted_dep11_Components-amd64.yml.gz}
13:57:13:0727 As  run AsStore:store-from-file
13:57:13:0727 As  run AsStore:app-info{/var/lib/app-info/yaml/archive.ubuntu.com_ubuntu_dists_xenial-proposed_restricted_dep11_Components-amd64.yml.gz}
13:57:13:0727 As  run AsStore:store-from-file
13:57:13:0727 As  run AsStore:app-info{/var/lib/app-info/yaml/security.ubuntu.com_ubuntu_dists_xenial-security_multiverse_dep11_Components-amd64.yml.gz}
13:57:13:0727 As  run AsStore:store-from-file
13:57:13:0727 As  run AsStore:app-info{/var/lib/app-info/yaml/archive.ubuntu.com_ubuntu_dists_xenial-updates_multiverse_dep11_Components-amd64.yml.gz}
13:57:13:0727 As  run AsStore:store-from-file
13:57:13:0728 As  run AsStore:app-info{/var/lib/app-info/yaml/archive.ubuntu.com_ubuntu_dists_xenial-updates_main_dep11_Components-amd64.yml.gz}
13:57:13:0728 As  run AsStore:store-from-file
13:57:13:0779 As  run AsStore:app-info{/var/lib/app-info/yaml/archive.ubuntu.com_ubuntu_dists_xenial_main_dep11_Components-amd64.yml.gz}
13:57:13:0779 As  run AsStore:store-from-file
13:57:13:0840 As  run AsStore:app-info{/var/lib/app-info/yaml/archive.ubuntu.com_ubuntu_dists_xenial_restricted_dep11_Components-amd64.yml.gz}
13:57:13:0840 As  run AsStore:store-from-file
13:57:13:0841 As  run AsStore:app-info{/var/lib/app-info/yaml/archive.ubuntu.com_ubuntu_dists_xenial-backports_main_dep11_Components-amd64.yml.gz}
13:57:13:0841 As  run AsStore:store-from-file
13:57:13:0841 As  run AsStore:app-info{/var/lib/app-info/yaml/archive.ubuntu.com_ubuntu_dists_xenial-backports_universe_dep11_Components-amd64.yml.gz}
13:57:13:0841 As  run AsStore:store-from-file
13:57:13:0841 As  run AsStore:app-info{/var/lib/app-info/yaml/archive.ubuntu.com_ubuntu_dists_xenial_universe_dep11_Components-amd64.yml.gz}
13:57:13:0841 As  run AsStore:store-from-file
13:57:14:0098 As  run AsStore:app-info{/var/lib/app-info/yaml/security.ubuntu.com_ubuntu_dists_xenial-security_main_dep11_Components-amd64.yml.gz}
13:57:14:0098 As  run AsStore:store-from-file
13:57:14:0108 As  run AsStore:app-info{/var/lib/app-info/yaml/archive.ubuntu.com_ubuntu_dists_xenial-updates_universe_dep11_Components-amd64.yml.gz}
13:57:14:0108 As  run AsStore:store-from-file
13:57:14:0121 As  run AsStore:app-info{/var/lib/app-info/yaml/archive.ubuntu.com_ubuntu_dists_xenial-updates_restricted_dep11_Components-amd64.yml.gz}
13:57:14:0121 As  run AsStore:store-from-file
13:57:14:0121 As  run AsStore:app-info{/var/lib/app-info/yaml/archive.ubuntu.com_ubuntu_dists_xenial-backports_restricted_dep11_Components-amd64.yml.gz}
13:57:14:0121 As  run AsStore:store-from-file
13:57:14:0122 As  adding existing file: /var/lib/app-info/yaml/security.ubuntu.com_ubuntu_dists_xenial-security_universe_dep11_Components-amd64.yml.gz
[... 395 lines cut ...]

```
Thread 1 "fwupd" received signal SIGTRAP, Trace/breakpoint trap.
[... cut ...]
#13 0x00007ffff6ab304a in g_main_context_dispatch (context=context@entry=0x55555578bbf0) at /build/glib2.0-prJhLS/glib2.0-2.48.2/./glib/gmain.c:3769
#14 0x00007ffff6ab33f0 in g_main_context_iterate (context=0x55555578bbf0, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at /build/glib2.0-prJhLS/glib2.0-2.48.2/./glib/gmain.c:3840
#15 0x00007ffff6ab3712 in g_main_loop_run (loop=0x55555578af80) at /build/glib2.0-prJhLS/glib2.0-2.48.2/./glib/gmain.c:4034
#16 0x000055555555cf2e in main ()
```

closed 2017-02-03 by hughsie

superm1, 2017-02-03:
So I've got a one line fix that resolves it.  When hotplugging I believe the problem is when there is a device in the plugin cache found from a recoldplug the auto pointer was causing the device to be freed.  I saw something similar happening in the Synaptics plugin and switched away from the autopointer.
The device should be freed by the daemon later. 

```
diff --git a/plugins/udev/fu-plugin-udev.c b/plugins/udev/fu-plugin-udev.c
index 3f545fe..8f8f0e1 100644
--- a/plugins/udev/fu-plugin-udev.c
+++ b/plugins/udev/fu-plugin-udev.c
@@ -122,7 +122,7 @@ fu_plugin_udev_add (FuPlugin *plugin, GUdevDevice *device)
        g_auto(GStrv) split = NULL;
        g_autoptr(AsProfile) profile = as_profile_new ();
        g_autoptr(AsProfileTask) ptask = NULL;
-       g_autoptr(FuDevice) dev = NULL;
+       FuDevice *dev = NULL;
 
        /* interesting device? */
        guid = g_udev_device_get_property (device, "FWUPD_GUID");

```

superm1, 2017-02-03:
Oh you just beat me to it.  I think that fix should do it too, but my fix most likely introduced a memory leak then.

superm1, 2017-02-03:
Yeah that does the trick, thanks.