fwupd/fwupd #87: autogen.sh: missing check for libelf

ghuls, 2017-02-10:
```
$ make
...
  CC       libdfu_la-dfu-common.lo
  CC       libdfu_la-dfu-cipher-devo.lo
  CC       libdfu_la-dfu-cipher-xtea.lo
  CC       libdfu_la-dfu-context.lo
  CC       libdfu_la-dfu-device.lo
  CC       libdfu_la-dfu-element.lo
  CC       libdfu_la-dfu-error.lo
  CC       libdfu_la-dfu-firmware.lo
  CC       libdfu_la-dfu-format-dfu.lo
  CC       libdfu_la-dfu-format-dfuse.lo
  CC       libdfu_la-dfu-format-elf.lo
dfu-format-elf.c:25:18: fatal error: gelf.h: No such file or directory
compilation terminated.
Makefile:1106: recipe for target 'libdfu_la-dfu-format-elf.lo' failed
```

hughsie, 2017-02-10:
What does config.h and config.log say? We do check for libelf IIRC.

ghuls, 2017-02-10:
[fwupd-config.zip](https://github.com/hughsie/fwupd/files/767149/fwupd-config.zip)

After installing libelf develoment files, it didn't complain about gelf.h:
```
sudo apt-get install libelf-dev
```

closed 2017-02-14 by superm1