fwupd/fwupd #92: fwupdmgr times out if /var/lib/fwupd/pending.db is missing

mirco, 2017-02-13:
The fwupd systemd service file contains ConditionPathExists=/var/lib/fwupd/pending.db, causing any fwupdmgr dbus requests to fwupd to time out if this file does not exist:

`Error calling StartServiceByName for org.freedesktop.fwupd: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.freedesktop.fwupd': timed out
`
After removing the line, everything works just fine, afaict (haven't done any actual fw updates). Why is it necessary?
Additionally, there is very little information on what that file actually is or how I can create it? Creating it as an empty file causes fwupd to segfault on startup.

Thanks

ArchangeGabriel, 2017-02-13:
I confirm the issue, but here removing that line doesn’t fix it, it’s still timing out. However instead of the non existing file error, I get a `Main process exited, code=exited, status=217/USER`.

closed 2017-02-13 by hughsie

simonbcn, 2017-04-13:
Arch
GNOME Shell 3.22.3
fwupd Installed from [AUR](https://aur.archlinux.org/packages/fwupd/) v0.8.1

```
$ sudo fwupdmgr -v get-devices
Error calling StartServiceByName for org.freedesktop.fwupd: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.freedesktop.fwupd': timed out
```

hughsie, 2017-04-13:
What does starting `fwupd --verbose` manually (as root) say?

simonbcn, 2017-04-13:
```
09:06:10:0861 Fu  Verbose debugging enabled (on console 0)
09:06:10:0861 As  run AsStore:load
09:06:10:0861 As  run AsStore:load{per-user}
09:06:10:0861 As  run AsStore:load{per-system}
09:06:10:0862 As  run AsStore:store-from-file{/usr/share/app-info/xmls/org.freedesktop.fwupd.xml}
09:06:10:0862 As  run AsStore:store-from-root
09:06:10:0862 As  using icon path /usr/share/app-info/icons/fwupd
09:06:10:0862 As  run AsStore:match-addons
09:06:10:0862 As  adding existing file: /usr/share/app-info/xmls/org.freedesktop.fwupd.xml
09:06:10:0862 As  run AsStore:match-addons
09:06:10:0862 As  Emitting ::changed() [store-load]
09:06:10:0862 Fu  Loading fallback values from /etc/fwupd.conf
09:06:10:0867 Fu  adding plugin /usr/lib/fwupd-plugins-2/libfu_plugin_altos.so
09:06:10:0867 Fu  adding plugin /usr/lib/fwupd-plugins-2/libfu_plugin_dfu.so
09:06:10:0868 Fu  performing init() on /usr/lib/fwupd-plugins-2/libfu_plugin_dfu.so
09:06:10:0868 Fu  adding plugin /usr/lib/fwupd-plugins-2/libfu_plugin_ebitdo.so
09:06:10:0868 Fu  adding plugin /usr/lib/fwupd-plugins-2/libfu_plugin_raspberrypi.so
09:06:10:0868 Fu  performing init() on /usr/lib/fwupd-plugins-2/libfu_plugin_raspberrypi.so
09:06:10:0868 Fu  adding plugin /usr/lib/fwupd-plugins-2/libfu_plugin_steelseries.so
09:06:10:0868 Fu  adding plugin /usr/lib/fwupd-plugins-2/libfu_plugin_test.so
09:06:10:0868 Fu  performing init() on /usr/lib/fwupd-plugins-2/libfu_plugin_test.so
09:06:10:0868 Fu  adding plugin /usr/lib/fwupd-plugins-2/libfu_plugin_udev.so
09:06:10:0868 Fu  performing init() on /usr/lib/fwupd-plugins-2/libfu_plugin_udev.so
09:06:10:0868 Fu  adding plugin /usr/lib/fwupd-plugins-2/libfu_plugin_unifying.so
09:06:10:0869 Fu  adding plugin /usr/lib/fwupd-plugins-2/libfu_plugin_upower.so
09:06:10:0869 Fu  performing init() on /usr/lib/fwupd-plugins-2/libfu_plugin_upower.so
09:06:10:0869 Fu  adding plugin /usr/lib/fwupd-plugins-2/libfu_plugin_usb.so
09:06:10:0869 Fu  adding plugin /usr/lib/fwupd-plugins-2/libfu_plugin_dell.so
09:06:10:0870 Fu  performing init() on /usr/lib/fwupd-plugins-2/libfu_plugin_dell.so
09:06:10:0930 Fu  adding plugin /usr/lib/fwupd-plugins-2/libfu_plugin_synapticsmst.so
09:06:10:0930 Fu  adding plugin /usr/lib/fwupd-plugins-2/libfu_plugin_colorhug.so
09:06:10:0932 Fu  performing init() on /usr/lib/fwupd-plugins-2/libfu_plugin_colorhug.so
09:06:10:0932 Fu  adding plugin /usr/lib/fwupd-plugins-2/libfu_plugin_uefi.so
09:06:10:0934 As  run FuMain:setup
09:06:10:0934 As  run FuMain:setup{altos}
09:06:10:0934 Fu  performing startup() on altos
09:06:10:0934 As  run FuMain:setup{dfu}
09:06:10:0934 Fu  performing startup() on dfu
09:06:10:0935 As  run FuMain:setup{ebitdo}
09:06:10:0935 Fu  performing startup() on ebitdo
09:06:10:0935 As  run FuMain:setup{raspberrypi}
09:06:10:0935 As  run FuMain:setup{steelseries}
09:06:10:0935 Fu  performing startup() on steelseries
09:06:10:0935 As  run FuMain:setup{test}
09:06:10:0935 As  run FuMain:setup{udev}
09:06:10:0935 As  run FuMain:setup{unifying}
09:06:10:0935 Fu  performing startup() on unifying
09:06:10:0935 As  run FuMain:setup{upower}
09:06:10:0935 Fu  performing startup() on upower
09:06:10:0936 As  run FuMain:setup{usb}
09:06:10:0936 Fu  performing startup() on usb
09:06:10:0936 As  run FuMain:setup{dell}
09:06:10:0936 Fu  performing startup() on dell
09:06:10:0936 Fu  disabling plugin because: failed to startup dell: Firmware updating not supported
09:06:10:0936 As  run FuMain:setup{synapticsmst}
09:06:10:0936 Fu  performing startup() on synapticsmst
09:06:10:0936 As  run FuMain:setup{colorhug}
09:06:10:0936 Fu  performing startup() on colorhug
09:06:10:0936 As  run FuMain:setup{uefi}
09:06:10:0936 Dfu device was not DFU capable
09:06:10:0936 As  run FuPluginAltos:added{1d6b:0002}
09:06:10:0936 As  run FuPluginEbitdo:added{1d6b:0002}
09:06:10:0936 As  run FuPluginUnifying:added{1d6b:0002}
09:06:10:0936 Dfu device was not DFU capable
09:06:10:0936 As  run FuPluginAltos:added{1d6b:0001}
09:06:10:0936 As  run FuPluginEbitdo:added{1d6b:0001}
09:06:10:0936 As  run FuPluginUnifying:added{1d6b:0001}
09:06:10:0936 Dfu device was not DFU capable
09:06:10:0936 As  run FuPluginAltos:added{1d6b:0001}
[... 94 lines cut ...]
09:06:11:0201 As  run FuMain:coldplug{synapticsmst}
09:06:11:0201 Fu  performing coldplug() on synapticsmst
09:06:11:0201 Fu  disabling plugin because: failed to coldplug synapticsmst: MST firmware updating not supported by OEM
09:06:11:0201 As  run FuMain:coldplug{colorhug}
09:06:11:0201 As  run FuMain:coldplug{uefi}
09:06:11:0201 Fu  performing coldplug() on uefi
09:06:11:0201 Fu  disabling plugin because: failed to coldplug uefi: UEFI firmware updating not supported
09:06:11:0201 Fu  Daemon ready for requests
09:06:11:0201 Fu  devices now in store:
09:06:11:0201 Fu  1	UEFI-dummy-dev0	UEFI Updates
09:06:11:0201 Fu  2	com.via.VL811.firmware	VL811 Firmware
09:06:11:0201 Fu  3	com.via.VL811+.firmware	VL811+ Firmware
09:06:11:0201 Fu  4	com.via.VL812.firmware	VL812 Firmware
09:06:11:0201 Fu  5	com.via.VL812_B2.firmware	VL812 B2 Firmware
09:06:11:0402 Fu  trying again: udev pending
09:06:11:0602 Fu  trying again: udev pending
09:06:11:0691 Fu  emit added from usb: usb:00:01
09:06:11:0700 Fu  emit added from udev: ro__sys_devices_pci0000_00_0000_00_01_0_0000_01_00_0
09:06:11:0802 Fu  registering D-Bus service
                   ##############################################################      @0324ms FuPluginUnifying:added{046d:c52b} 248ms
                                                                                   ##  @0340ms FuMain:coldplug 9ms
                                                                                   ##  @0340ms FuMain:coldplug{udev} 9ms
                                                                                    #  @0340ms FuPluginUdev:coldplug{pci} 5ms
09:06:11:0804 Fu  FuMain: acquired name: org.freedesktop.fwupd
```

hughsie, 2017-04-13:
So the daemon can start; I'm not sure why systemd wouldn't see the bus name.

simonbcn, 2017-04-13:
Sorry, forget it, I just saw that this utility is not for my system.