fwupd/fwupd #93: Build failure on Jenkins

pwithnall, 2017-02-13:
Looks like the recent docbook changes might be causing a [build failure on Jenkins](https://jenkins.freedesktop.org/job/coverity/job/coverity-scan-jhbuild/1040/console):
```
make[2]: Entering directory '/scratch/coverity-scan-jhbuild/build/fwupd/docs/man'
  GEN      dfu-tool.1
nsgmls:/usr/share/xml/docbook/schema/dtd/4.1.2/docbookx.dtd:74:17:E: "X20AC" is not a function name
nsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:40:19:E: "X21B6" is not a function name
nsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:41:19:E: "X21B7" is not a function name
nsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:42:17:E: "X21D3" is not a function name
nsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:43:18:E: "X21CA" is not a function name
…
nsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsr.ent:54:18:E: "X22A3" is not a function name
nsgmls:I: maximum number of errors (200) reached; change with -E option
  GEN      fwupdmgr.1
nsgmls:/usr/share/xml/docbook/schema/dtd/4.1.2/docbookx.dtd:74:17:E: "X20AC" is not a function name
nsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:40:19:E: "X21B6" is not a function name
nsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:41:19:E: "X21B7" is not a function name
…
nsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsr.ent:54:18:E: "X22A3" is not a function name
nsgmls:I: maximum number of errors (200) reached; change with -E option
make[3]: Entering directory '/scratch/coverity-scan-jhbuild/build/fwupd/docs/man'
make[3]: Nothing to be done for 'install-exec-am'.
  GEN      dfu-tool.1
nsgmls:/usr/share/xml/docbook/schema/dtd/4.1.2/docbookx.dtd:74:17:E: "X20AC" is not a function name
nsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:40:19:E: "X21B6" is not a function name
nsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:41:19:E: "X21B7" is not a function name
…
nsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsr.ent:54:18:E: "X22A3" is not a function name
nsgmls:I: maximum number of errors (200) reached; change with -E option
  GEN      fwupdmgr.1
nsgmls:/usr/share/xml/docbook/schema/dtd/4.1.2/docbookx.dtd:74:17:E: "X20AC" is not a function name
nsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:40:19:E: "X21B6" is not a function name
nsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsa.ent:41:19:E: "X21B7" is not a function name
…
nsgmls:/usr/share/xml/entities/xml-iso-entities-8879.1986/ISOamsr.ent:54:18:E: "X22A3" is not a function name
nsgmls:I: maximum number of errors (200) reached; change with -E option
 /bin/mkdir -p '/scratch/coverity-scan-jhbuild/install/_jhbuild/root-fwupd/scratch/coverity-scan-jhbuild/install/share/man/man1'
 /usr/bin/install -c -m 644 /scratch/coverity-scan-jhbuild/checkout/fwupd/docs/man/dfu-tool.1 /scratch/coverity-scan-jhbuild/checkout/fwupd/docs/man/fwupdmgr.1 '/scratch/coverity-scan-jhbuild/install/_jhbuild/root-fwupd/scratch/coverity-scan-jhbuild/install/share/man/man1'
/usr/bin/install: cannot stat '/scratch/coverity-scan-jhbuild/checkout/fwupd/docs/man/dfu-tool.1': No such file or directory
/usr/bin/install: cannot stat '/scratch/coverity-scan-jhbuild/checkout/fwupd/docs/man/fwupdmgr.1': No such file or directory
Makefile:430: recipe for target 'install-man1' failed
```

superm1, 2017-02-14:
It's failing for me on Ubuntu now too now, similar errors.

Switching to `docbook2x-man`  fixes it for me.  I'm not sure if the binary name is stable across distros though, maybe someone on **Fedora** and **Gentoo** can confirm ( @kakra and @jjardon can you look? ).   Otherwise something like this is going to be needed:

```
diff --git a/configure.ac b/configure.ac
index b833f72..fce3c87 100644
--- a/configure.ac
+++ b/configure.ac
@@ -144,9 +144,12 @@ PKG_CHECK_MODULES(GUSB, gusb >= 0.2.9)
 PKG_CHECK_MODULES(SQLITE, sqlite3)
 PKG_CHECK_MODULES(ARCHIVE, libarchive)
 PKG_CHECK_MODULES(SOUP, libsoup-2.4 >= 2.51.92)
-AC_PATH_PROG(DOCBOOK2MAN, docbook2man)
+AC_PATH_PROG(DOCBOOK2MAN, docbook2x-man)
 if test -z $DOCBOOK2MAN ; then
-       AC_MSG_ERROR([docbook2man program not found])
+       AC_PATH_PROG(DOCBOOK2MAN, docbook2man)
+       if test -z $DOCBOOK2MAN ; then
+               AC_MSG_ERROR([docbook2man program not found])
+       fi
 fi
 
 AC_PATH_PROG(GCAB, [gcab], [no])

```

ArchangeGabriel, 2017-02-14:
ArchLinux here:
docbook2x usr/bin/docbook2man

kakra, 2017-02-14:
Gentoo needs `docbook2man.pl` here... The `docbook2x-man` is incompatible with the SGML files at compile time for me.

But I don't understand why it fails in the configure phase now because `configure.in` wasn't touched in my pull request.

My build script uses a sed command to substitute the correct binary:

    sed -i -e 's/docbook2man/docbook2man.pl/' configure.in

I don't think the error from the previous comment is a result of the pull request. It has to be a side effect or unhides an already existing problem.

About the jenkins problem I'm not sure... Looks like there's a problem with the DTD. It works for me, the unpatched version shipped with no DTD.

kakra, 2017-02-14:
BTW: Without DTD, it fails for me because of undefined entities (the entity definition in the header seems to be ignored). Actually, I don't know how to correctly fix this. I'm no XML or DocBook expert, and documentation about this was sparse.

kakra, 2017-02-14:
Looking at this again, you may want to try other DTD versions and report back here. I used the lowest version that worked for me using `docbook2man.pl`. See: http://docbook.org/xml/

ArchangeGabriel, 2017-02-14:
Not sure if that’s related, but this discussion reminded me we use a patch in colorhug-client (also form @hughsie, so likely the same things) for the man page, see here: https://aur.archlinux.org/cgit/aur.git/tree/colorhug-client-0.2.8-man.patch?h=colorhug-client

kakra, 2017-02-14:
@ArchangeGabriel Looks exactly what I needed to change to make it build in Gentoo. Except, I didn't add the "XML" string to the DOCTYPE. Maybe it's just that what's missing? Can someone confirm?

kakra, 2017-02-15:
I tried that, too. But this DTD generates a "404 not found" for me, because it doesn't exist.

superm1, 2017-02-15:
@kakra is your doc2man.pl doing some sort of other changed internally?  I do see that http://www.oasis-open.org/docbook/xml/4.1/docbookx.dtd does exist out on the web at least.

hughsie, 2017-02-16:
I've reverted 0e8f5d93ae2bb1acbc912c4eec3bec5f593678c3 for now as it's important jenkins is running while we find the correct fix.

kakra, 2017-02-19:
Could you guys please retry with my branch https://github.com/kakra/fwupd/tree/fix-docbook2x-v2

@superm1 For me, it gave an error during build (network IO error) although the file exists on http. I don't know what the issue is. I've adjusted to instead use 4.5 now which builds for me. You need to have the 4.5 DTDs installed maybe (at least Gentoo ships them in a package).

superm1, 2017-02-20:
@kakra your branch does compile properly for me on Ubuntu with docbook2x-man instead.
@hughsie what is your take on moving to docbook2x?

hughsie, 2017-02-24:
I have to say I'm not hugely knowledgeable in the various docbook toolchains available. Fedora certainly has docbook2X although I've got no idea if it's in anyway better than what we use now.

superm1, 2017-02-28:
@kakra At least for now I think it's better that you change your ebuild to use https://packages.gentoo.org/packages/app-text/docbook-sgml-utils instead of docbook2man.pl.  it looks to me (from querying http://www.portagefilelist.de/site/query/listPackageFiles/?category=app-text&package=docbook-sgml-utils&version=0.6.14-r1&do#result ) that it provides **/usr/bin/docbook2man** and would then match what all the other distros are doing with doc generation.

kakra, 2017-02-28:
@superm1 Strange that nothing pulls it in here... Still, the path shouldn't be hard-coded in one of the make files. I'll redo my patch.

superm1, 2017-03-07:
Alright for now, I think we'll mark this closed.  I expect once you add that to your ebuild you should be in good shape.
If there is a need to switch to docbook2x or mess with the DTD again we can revisit it then.

closed 2017-03-07 by superm1