fwupd/fwupd #94: failed to startup dell: Firmware updating not supported

shlevy, 2017-02-27:
I'm running `fwupd-0.8.0` on my XPS 13 and getting the above message. Am I wrong that `fwupd` is the appropriate firmware update path for this machine?

superm1, 2017-02-27:
There was a few issues with the systemd rules that were included in 0.8.0.  Can you please retry with the recently released _(2 hours ago)_ **0.8.1** release?

shlevy, 2017-02-27:
I now get a systemd failure:
fwupd.service: Main process exited, code=exited, status=226/NAMESPACE

shlevy, 2017-02-27:
Ah, it seems the prefix for `fwupd` itself is not included in the namespaces set up by the unit :smile:

shlevy, 2017-02-27:
Ah, no, it's because /var/lib/fwupd is missing. Patch incoming.

mirco, 2017-02-27:
In my case the NAMESPACE error was because of the ReadWritePath=/boot/efi setting. On my system the path is /boot/EFI, I think because systemd's bootctl set it up like this. As /boot is a fat32 partition, which isn't really case sensitive, this isn't an issue for normal operation. But for systemd it is, and because systemd can't find /boot/efi it errors out with file not found. I'm not sure who is to blame here. Me, systemd or fwupd.

shlevy, 2017-02-27:
Ah, no patch, I just added another service to ensure /var/lib/fwupd exists to my config. Works now!

closed 2017-02-27 by shlevy

superm1, 2017-02-27:
@shlevy 
I think needing to make /var/lib/fwupd manually is still a defect if the service requires it on first boot (which it does).  I've added a commit to create this directory by default.

@mirco 
This will come down to which distros are using bootctl (and kernel EFI stub bootloader) vs distros that use GRUB2.  I've added a configuration option that will let you configure the EFI system partition location correctly so that distro packages can set this properly depending on which tool they use.

ArchangeGabriel, 2017-03-05:
@superm1 Well Arch Linux (and others too) let the bootloader choice at user’s discretion. So…

superm1, 2017-03-07:
@ArchangeGabriel The path is now optional for starting the service:
https://github.com/hughsie/fwupd/commit/332415d37871f0cc0f37afa1303546e4252439c1

So at least that won't stop fwupd from starting even if the path used for the ESP is different than the default the package was compiled with.

It's obviously a one line change to sed this around in a post install script.  If Arch (or others) are going to be allowing multiple paths for the ESP it makes sense to me to just change it in the post install script for the package.