fwupd/fwupd #98: Drop CAP_SYS_ADMIN and/or disallow the mount syscall to fwupd

shlevy, 2017-02-28:
From `systemd.exec(5)`, in the section on `ReadWritePaths=`:

> Note that the effect of these settings may be undone by privileged processes. In order to set up an effective sandboxed environment for a unit it is thus recommended to combine these settings with either CapabilityBoundingSet=\~CAP_SYS_ADMIN or SystemCallFilter=~@mount.

hughsie, 2017-08-25:
Have you got a few more details here please? fwupd certainly doesn't need to mount anything, but I'ma bit lost in the maze of options in the systemd service file.

hughsie, 2017-09-18:
I'm unclear what the actual issue is here, please re-open if you can provide some more details. Thanks.

closed 2017-09-18 by hughsie

ybernat, 2017-09-18:
https://lwn.net/Articles/486306/

This capability is considered as a security risk, if I understand correctly, as it allows too many privileged operations. Securing the daemon against possible hacks and exploits includes reducing the number of things the attacker can do even if he gains control on the daemon.

Mount is specifically considered dangerous as it allows loading new fs that was crafted to include a root-owned suid executable, for example, and thus giving the attacker full control on the system.

I see that systemd includes a nice way to control this and remove these capabilities from the daemon with no code change, as quoted above.

Seems like adding the settings quoted above is enough to make fwupd more secured.

hughsie, 2017-09-18:
So we should just add `SystemCallFilter=~@mount` to `data/fwupd.service.in` ?

ybernat, 2017-09-18:
Or better, `CapabilityBoundingSet=~CAP_SYS_ADMIN`, if possible.