fwupd/fwupd #99: Make it possible to update to unpackaged firmware

hadess, 2017-03-07:
There's currently no way to update to a firmware that's not packaged, signed and distributed via the LVFS service. This is a problem when either:
- the vendor does not support LVFS
- the vendor sends test firmware
- the vendor or community has alternative firmware.

In my case, I'd like to use fwupd's 8bitdo plugin code to update my joypad to this version of the firmware:
http://www.8bitdo.com/nes30pro/Extend-firmware.html

Currently, the only to install this particular firmware would be using Windows or macOS.

hughsie, 2017-03-07:
If you build fwupd locally, you can run plugins/ebitdo/fu-ebitdo-tool although it's not installed at the moment. Without a .cab file with some metadata in, it is really just a blob of random.

hadess, 2017-03-07:
That works for me.

hughsie, 2017-03-07:
Do you think installing it into libexec would be a good idea? I'm not terribly sure I want translations, man pages and all that gunk for a low level tool like this.

hadess, 2017-03-07:
No, I think that's fine as-is. I was expecting it to be possible to do more generically, passing the device to be updated, and the file to use, but I guess that's not enough information for a lot of update types.

If I ever get the inclination to do so, I'd probably make a small graphical application from it, rather than expect a very specific test tool to be used.

closed 2017-05-23 by hughsie