fwupd/fwupd #101: "Failed to add Logitech device" and unusable Logitech devices

hadess, 2017-03-22:
Using fwupd-0.8.1-1.fc26.x86_64 with a newer kernel (tested with 4.10.4). After boot, my keyboard and mouse work in gdm, but not in the session.

Masking fwupd fixes the problem. The error in the logs when my keyboard and mouse aren't usable is:
```
Mar 22 16:52:43 classic fwupd[2219]: Failed to add Logitech device: failed to read config 0x04: device output 20 bytes, buffer size only 15
```

Fixing the bug above might be a good way to fix the problem, but I have a feeling that the device isn't unclaimed in error paths. Does that look correct?
```
diff --git a/plugins/unifying/fu-plugin-unifying.c b/plugins/unifying/fu-plugin-unifying.c
index 8fdf7db..d33a8b3 100644
--- a/plugins/unifying/fu-plugin-unifying.c
+++ b/plugins/unifying/fu-plugin-unifying.c
@@ -61,8 +61,10 @@ fu_plugin_unifying_device_added (FuPlugin *plugin,
        fu_device_set_id (FU_DEVICE (dev), platform_id);
 
        /* open the device */
-       if (!fu_device_unifying_open (dev, error))
+       if (!fu_device_unifying_open (dev, error)) {
+               fu_device_unifying_close (dev, NULL);
                return FALSE;
+       }
 
        /* close the device */
        if (!fu_device_unifying_close (dev, error))
```

hadess, 2017-03-22:
CC @bentiss @whot

hughsie, 2017-03-23:
Interesting, a long response for a short reply. Of course, this isn't interesting to you, and I'm in the process working with Logitech on a plugin that uses hidraw rather than low level USB access (which means the kernel driver isn't disconnected). We'll still do the interface claim when doing dongle firmware updates, but that's a rare thing and not done every boot. If the new plugin takes longer than a few more days I'll fix this bug in upstream and rawhide, otherwise I think it's best to wait for the new code.

hadess, 2017-03-23:
The patch works. Mind if I push it to Fedora in the short-term?

hughsie, 2017-03-23:
Sure, no problem at all. Thanks.

superm1, 2017-04-14:
@hadess since @hughsie got the rewritten logitech plugin done could you double check if this is still a problem on that version?

hadess, 2017-04-26:
That particular bug isn't present in 0.8.2 any more.

closed 2017-04-26 by hadess