fwupd/fwupd #103: Dell XPS 9560 Ubuntu 16.10 Thunderbolt Not Detected

Queuecumber, 2017-03-25:
I compiled the master branch of fwupd after reading that intel thunderbolt device support was recently added. I am using the Dell XPS 9560 with a clean Ubuntu 16.10 install. When I run `fwupdmgr get-devices` I don't see any thunderbolt controller listed. I actually see less devices listed than my distribution-provided fwupdmgr was showing now, though few of them listed the "supported" flag anyway. Right now my only way to update thunderbolt controller firmware is to open the laptop and put in the SSD with windows on it, then download the exe from Dells website, so having this work would be a big time saver.

In more detail, I made sure that I was building with thunderbolt support during the configure step, my final output is this:

```
                    fwupd 0.8.2
                  =================

        prefix:                    /usr/local
        datadir:                   ${datarootdir}
        compiler:                  gcc
        cflags:                    -g -O2
        cppflags:                  

	optional plugins
	------------------
	Colorhug:                  yes
	libelf:                    yes
	UEFI:                      yes
	Dell:                      yes
	Synaptics MST:             yes
	Thunderbolt:               yes
```
But note that on Ubuntu 16.10, it seems like there is no included libtbtfwu packages so I grabbed debs from [here](https://bugs.launchpad.net/~dell-team/+archive/ubuntu/ppa/+build/12006750) to install manually. This seemed to satisfy `configure`. Also note that there is potentially a bug in the compilation process because without the `gcab` binary (install with `sudo apt install gcab`), `configure` works fine but `make` fails with the nice cryptic error:`no: command not found` (as ${GCAB} is set to "no"), this is a distinct package from the `libgcab-dev` one which is probably the cause of the problem (e.g. you are checking for gcab development libraries but not the binary itself). This error happens when compiling certain tests, but it is a separate issue so I can open another one if youd like.

Once I made the thing compile and install to /usr/local, I made sure that the service is restarted (`systemctl restart fwupd.service`), checked that the mtime on the service binary was correct (`stat /usr/local/libexec/fwupd/fwupd`) and verified that the system was finding the new fwupdmgr binary (`which fwpdmgr`). All looks to be correct.

But, again, when running the `get-devices` command I see far fewer devices listed than I was seeing with my distro-provided one. Here they are:

```
BCM20702A0
  Guid:                 5f92dda7-7e65-59b9-bb72-51fe4944f2f1
  Guid:                 6dae8164-f120-5f82-a192-cd6840a57011
  DeviceID:             usb:00:01:04:02:03
  Plugin:               dfu
  Flags:                allow-online|allow-offline
  Version:              1.18
  Created:              2017-03-25

XPS 15 9560
  Guid:                 34578c72-11dc-4378-bc7f-b643866f598c
  DeviceID:             UEFI-34578c72-11dc-4378-bc7f-b643866f598c-dev0
  Plugin:               uefi
  Flags:                internal|allow-offline|require-ac|supported
  Version:              0.1.1.3
  VersionLowest:        0.1.1.3
  Created:              2017-03-25

My Book 1230
  Guid:                 74e3a5c1-cb5f-55ac-a376-f09d4151e9fd
  Guid:                 ac467fc0-89d2-5cb6-99d5-dfb15c9b7143
  DeviceID:             usb:00:01:04:04
  Plugin:               usb
  Flags:                none
  Version:              16.101
  Created:              2017-03-25

Unknown Device
  Guid:                 37921484-dff0-57a0-b7dd-4cd6a82b54a1
  DeviceID:             ro__sys_devices_pci0000_00_0000_00_01_0_0000_01_00_0
  Plugin:               udev
  Flags:                internal|locked
  DeviceVendor:         NVIDIA Corporation
  Created:              2017-03-25
```

The XPS 15 uefi is listed and supported which is great, but no thunderbolt controller. 
The controller itself shows up in `lspci` is follows:

```
06:00.0 PCI bridge: Intel Corporation DSL6340 Thunderbolt 3 Bridge [Alpine Ridge 2C 2015]
07:00.0 PCI bridge: Intel Corporation DSL6340 Thunderbolt 3 Bridge [Alpine Ridge 2C 2015]
07:01.0 PCI bridge: Intel Corporation DSL6340 Thunderbolt 3 Bridge [Alpine Ridge 2C 2015]
07:02.0 PCI bridge: Intel Corporation DSL6340 Thunderbolt 3 Bridge [Alpine Ridge 2C 2015]
09:00.0 PCI bridge: Intel Corporation DSL6540 Thunderbolt 3 Bridge [Alpine Ridge 4C 2015]
0a:01.0 PCI bridge: Intel Corporation DSL6540 Thunderbolt 3 Bridge [Alpine Ridge 4C 2015]
0a:04.0 PCI bridge: Intel Corporation DSL6540 Thunderbolt 3 Bridge [Alpine Ridge 4C 2015]
```

This is on kernel 4.10.5. 

Please let me know if I can provide any more information to help diagnose

superm1, 2017-03-25:
Did you also install the thunderbolt kernel module dkms package? It's stored on the same repository.

Also you may need to run fwupd with verbose mode to provide more information on what happens when the thunderbolt plugin runs.

Queuecumber, 2017-03-25:
You made me realize I hadn't yet checked the journal for fwupd. In `journalctl -u fwupd.service` I found:

```
Mar 25 10:27:09 lilith fwupd[16126]: failed to open plugin /usr/local/lib/fwupd-plugins-2/libfu_plugin_thunderbolt.so: failed to open plugin: /usr/local/lib/fwupd-plugins-2/libfu_plugin_thunderbolt.so: undefined symbol: efi_guid_zero
```

My best guess is that this is because of the weird libtbtfwu I got but maybe you have a better idea of where that missing symbol should live?

superm1, 2017-03-28:
That should have come from efivar.  There have definitely been some weird linking problems with `efi_guid_zero ` (look at efivar upstream to see some of the fixes.  Eg: https://github.com/rhinstaller/efivar/commit/d9e9ec36fe1f60638004091ad3ab10ff63fab944 and https://github.com/rhinstaller/efivar/commit/4af5b56066b4af66ce8d968948bb2bf4f5a87c79).  

Try adding a fix similar to this https://github.com/hughsie/fwupd/issues/88 to your Makefile.am for Thunderbolt.  See if it helps.

Regarding the possibly missing kernel module, make sure you install https://bugs.launchpad.net/~dell-team/+archive/ubuntu/ppa/+build/12006750/+files/thunderbolt-icm-dkms_2017.01.19-0ubuntu0dell1_all.deb.

Queuecumber, 2017-03-28:
Well...

Adding the $(EFIVAR_LIBS) definitely fixed the missing symbol

Problem now is that the DKMS package for the kernel module is broken by kernel 4.10 because it uses some genl symbols that were removed:

```
DKMS make.log for thunderbolt-icm-2017.01.19 for kernel 4.10.6-acso (x86_64)
Tue Mar 28 16:01:07 EDT 2017
make: Entering directory '/var/lib/dkms/thunderbolt-icm/2017.01.19/build'
make -C /lib/modules/4.10.6-acso/build M=`pwd` modules
make[1]: Entering directory '/usr/src/linux-headers-4.10.6-acso'
  CC [M]  /var/lib/dkms/thunderbolt-icm/2017.01.19/build/icm_nhi.o
/var/lib/dkms/thunderbolt-icm/2017.01.19/build/icm_nhi.c:78:9: error: ‘GENL_ID_GENERATE’ undeclared here (not in a function)
  .id  = GENL_ID_GENERATE,
         ^~~~~~~~~~~~~~~~
/var/lib/dkms/thunderbolt-icm/2017.01.19/build/icm_nhi.c: In function ‘icm_nhi_init’:
/var/lib/dkms/thunderbolt-icm/2017.01.19/build/icm_nhi.c:1495:7: error: implicit declaration of function ‘genl_register_family_with_ops’ [-Werror=implicit-function-declaration]
  rc = genl_register_family_with_ops(&nhi_genl_family, nhi_ops);
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:294: recipe for target '/var/lib/dkms/thunderbolt-icm/2017.01.19/build/icm_nhi.o' failed
make[2]: *** [/var/lib/dkms/thunderbolt-icm/2017.01.19/build/icm_nhi.o] Error 1
Makefile:1490: recipe for target '_module_/var/lib/dkms/thunderbolt-icm/2017.01.19/build' failed
make[1]: *** [_module_/var/lib/dkms/thunderbolt-icm/2017.01.19/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.10.6-acso'
Makefile:6: recipe for target 'all' failed
make: *** [all] Error 2
make: Leaving directory '/var/lib/dkms/thunderbolt-icm/2017.01.19/build'
```
related info [here](https://patchwork.ozlabs.org/patch/734681/)

Indeed when I look at the code from that DKMS package and compare it to the kernel tree from 01org's github, the source files look completely different to me, so I'm not sure where DKMS package comes from. 

Also I can see that I need that module as `journalctl -u thunderbolt.service` gives 

```
thunderboltd[18640]: Error: Error: Exception: Module version query failed  
```

I'm tempted to try to patch the DKMS module as it looks like its only complaining about two symbols, but if the DKMS package is very out of date then I'm not sure what other bug fixes are included in Intel's kernel tree that I might need. 

Any advice?

Queuecumber, 2017-03-28:
Looks like it was as simple as copying the new `icm_nhi.c` and merging `icm_nhi.h`, I didnt see any other changes in the diffs. After that the module builds and loads against kernel 4.10. I noticed you wrote that one too, I can make a PR if you want.

Now there is another problem, fwupd still cant talk to the thunderbolt daemon using DBus. When I looked in the thunderboltd logs for info the only thing resembling an error I see is

```
Mar 28 17:28:50 lilith thunderboltd[1688]: Debug: validate netlink message: PASS      Thread: 140510486279936, func cb_valid line 291 (/home/mehrlich/Code/thunderbolt-software-user-space/daemon/ThunderboltService/Linux/src/GenlWrapper.cpp)
Mar 28 17:28:50 lilith thunderboltd[1688]: Debug: LinuxControllerCommandSender::OnEvent entry, PDF: 10      Thread: 140510486279936, func OnEvent line 63 (/home/mehrlich/Code/thunderbolt-software-user-space/daemon/ThunderboltService/Linux/src/LinuxControllerCommandSender.cpp)
Mar 28 17:28:50 lilith thunderboltd[1688]: Debug: Message from FW: PDF_FW_TO_SW_NOTIFICATION, Controller: 8001575      Thread: 140510486279936, func OnEvent line 83 (/home/mehrlich/Code/thunderbolt-software-user-space/daemon/ThunderboltService/Linux/src/LinuxControllerCommandSender.cpp)
Mar 28 17:28:50 lilith thunderboltd[1688]: Warning: Warning: FW Notification: Unknown, Controller: 08001575      Thread: 140510486279936, func HandleFwToSwNotification line 89 (/home/mehrlich/Code/thunderbolt-software-user-space/daemon/ThunderboltService/Linux/src/MessageFromFwEvent.cpp)
Mar 28 17:28:50 lilith thunderboltd[1688]: Error: Error: LinuxControllerCommandSender::OnEvent failed, PDF: 10, (Exception: Unknown notification from FW3)      Thread: 140510486279936, func OnEvent line 119 (/home/mehrlich/Code/thunderbolt-software-user-space/daemon/ThunderboltService/Linux/src/LinuxControllerCommandSender.cpp)
Mar 28 17:28:50 lilith thunderboltd[1688]: Debug: LinuxControllerCommandSender::OnEvent exit, PDF: 10      Thread: 140510486279936, func OnEvent line 121 (/home/mehrlich/Code/thunderbolt-software-user-space/daemon/ThunderboltService/Linux/src/LinuxControllerCommandSender.cpp)
Mar 28 17:28:50 lilith thunderboltd[1688]: Info: GenlWrapper::cb_valid exit, message arrived      Thread: 140510486279936, func cb_valid line 362 (/home/mehrlich/Code/thunderbolt-software-user-space/daemon/ThunderboltService/Linux/src/GenlWrapper.cpp)
```

And, as expected, inspecting the DBus interface with D-Feet shows that com.intel.Thunderbolt1 is empty.

superm1, 2017-03-28:
@Queuecumber the DKMS package was built from this: https://github.com/01org/thunderbolt-software-kernel-tree/tree/networking.  I'm a bit confused why you needed to merge a new icm_nhi.c.  If you can pull request to https://github.com/dell/thunderbolt-software-user-space/tree/debian yes I'll definitely take a look.

I suspect you have an out of sync kernel module and daemon with those errors though right now.   Since the kernel module is not yet upstream, it's a bit confusing in that you need the kernel module from their 'fwupdate' branch (v9 of the upstream submission) and the daemon from it's 'fwupdate' branch to match it.

Queuecumber, 2017-03-28:
Took a look at https://github.com/dell/thunderbolt-icm-dkms/issues/2 as well, the thing is, intel already patched it for 4.10. My incoming PR is just copying their latest `icm_nhi.c` and a merge with your `icm_nhi.h`, nothing else changed (I think... this isnt code i've ever looked at before so definitely review it with a critical eye). 

So if I understand you correctly, I need a new version of the daemon to match the version of the kernel module correct? Thing is I'm pretty sure I built the daemon from the master branch but while Im preparing the PR for your I'll have another look, make sure there isnt a branch lying around that I overlooked

superm1, 2017-03-28:
Can you clarify where you got that latest icm_nhi.c and icm_nhi.h?  Were those the fwupdate branch at https://github.com/01org/thunderbolt-software-kernel-tree/tree/networking?  That was what https://github.com/dell/thunderbolt-software-user-space/tree/debian was built from.  I can't tell that it changed at all, but I can look more closely again.

The daemon needs to be built from the fwupdate branch, not the master branch.  If you took my Debian package on that PPA it is built from the fwupdate branch.

Queuecumber, 2017-03-28:
I must be missing something because I only see master and networking branches on https://github.com/01org/thunderbolt-software-kernel-tree/tree/networking which is where I pulled the latest icm_nhi.{h,c}, nothing to do with a fwupdate branch. To be clear we are talking about these files:

https://github.com/01org/thunderbolt-software-kernel-tree/blob/networking/drivers/thunderbolt/icm/icm_nhi.c
https://github.com/01org/thunderbolt-software-kernel-tree/blob/networking/drivers/thunderbolt/icm/icm_nhi.h
 
I do see fwupdate on https://github.com/dell/thunderbolt-software-user-space/tree/debian so I will try building that. Thanks for the pointer on that one, I didnt realize it had a special branch

Queuecumber, 2017-03-28:
OK progress

Compiling the fwupdate branch of https://github.com/dell/thunderbolt-software-user-space/tree/debian now the thunderbolt daemon doesnt crash and d-feet can see some interfaces. Problem now is, executing `com.Intel.Thunderbolt1.controllers.GetControllerList()` returns `[]`, e.g. an empty array. So the thunderbolt daemon is detecting no thunderbolt controllers. 

I am going to go ahead and submit a PR on the DKMS module, if I pulled from a wrong branch or something go ahead and reject

Queuecumber, 2017-03-30:
Alright I'm not sure why it didnt work immediately after fixing the icm driver, but it just started working. 

```
BCM20702A0
  Guid:                 5f92dda7-7e65-59b9-bb72-51fe4944f2f1
  Guid:                 6dae8164-f120-5f82-a192-cd6840a57011
  DeviceID:             usb:00:01:04:04:03
  Plugin:               dfu
  Flags:                allow-online|allow-offline
  Version:              1.18
  Created:              2017-03-30

Thunderbolt Controller
  Guid:                 72533768-6a6c-5c06-994a-367374336810
  DeviceID:             08001575
  Plugin:               thunderbolt
  Flags:                internal|allow-online
  DeviceVendor:         Intel
  Version:              21.00
  Created:              2017-03-30

XPS 15 9560
  Guid:                 34578c72-11dc-4378-bc7f-b643866f598c
  DeviceID:             UEFI-34578c72-11dc-4378-bc7f-b643866f598c-dev0
  Plugin:               uefi
  Flags:                internal|allow-offline|require-ac|supported
  Version:              0.1.1.3
  VersionLowest:        0.1.1.3
  Created:              2017-03-30

My Book 1230
  Guid:                 74e3a5c1-cb5f-55ac-a376-f09d4151e9fd
  Guid:                 ac467fc0-89d2-5cb6-99d5-dfb15c9b7143
  DeviceID:             usb:00:01:04:02
  Plugin:               usb
  Flags:                none
  Version:              16.101
  Created:              2017-03-30

Unknown Device
  Guid:                 37921484-dff0-57a0-b7dd-4cd6a82b54a1
  DeviceID:             ro__sys_devices_pci0000_00_0000_00_01_0_0000_01_00_0
  Plugin:               udev
  Flags:                internal|locked
  DeviceVendor:         NVIDIA Corporation
  Created:              2017-03-30
```

Question is, can I actually update the firmware. I suppose I need to wait for intel to release an update before I'll know that?

superm1, 2017-03-30:
Did you have nothing plugged into the TBT port when you tried to use d-feet?  Thunderbolt controllers go to sleep if not in use to save power.  The fwupd plugin actually has some special [magic](https://github.com/hughsie/fwupd/blob/master/plugins/dell/fu-dell-common.c#L346) that can turn on sleeping controllers on Dell systems.

As for whether you can update the FW, it's pending a Dell release actually.  Dell hasn't released any TBT updates to [LVFS](https://secure-lvfs.rhcloud.com/lvfs/devicelist) yet.  I can't comment on when that will happen.  
It's technically possible to craft your own update CAB from extracting the .EXE on support.dell.com if there is an newer NVM.

Maybe that's something we should make a contrib/ helper script in python or shell that if you have a binary will build a boilerplate metainfo.xml and package it up in a .CAB.  This can at least work for local use.  If there is a newer version on support.dell.com for your platform (which might be possible - I think some platforms have v23 now) you can take a stab at that if you want.

Queuecumber, 2017-03-30:
I actually keep the thing plugged into a thunderbolt dock when I'm working on stuff so it was definitely plugged in, I guess it's just more magic.

I would love to take a shot at that, but I don't think its released for my system yet (http://www.dell.com/support/home/us/en/19/product-support/product/xps-15-9560-laptop/drivers lists version 3.21 which I'm assuming corresponds to the 21.00 reported by fwupd). When a new version is released I'll give it a shot, although it would really be great if Dell published the firmware themselves. Before then, if you could give me some tips on how to do it, that would help since its not something ive done before. Is there any chance of bricking the chip if that is done wrong?

In the meantime maybe it would be better to push for upstreaming the icm patches somehow.

superm1, 2017-03-30:
Yes i'm working on Dell publishing the updates to LVFS.  When it happens you've got fwupd set up now and you would be notified.  
Yes, this is all pretty early since the ICM stuff isn't upstreamed, no distro is yet carrying thunderboltd, and no distro has a thunderbolt enabled fwupd.  Lots of things to line up still.

As for chance in bricking the chip?  There's always a chance when flashing FW that say your power goes out and you brick a chip.  So I couldn't say it is impossible with 100% certainty.  

The update signature, machine and chip are verified before flashing.  You wouldn't be able to flash the wrong one onto your system.  If the flash process were to fail the chip is supposed to be able to recover and let you try again.

As for how to start at building a script to generate the CAB, you can extract the binaries within the EXE from support.dell.com.  I believe 7z can work with it directly, but other extraction tools should work too.

Find the one that corresponds to your system that's in the EXE.  Right now the EXE has a collection of them for various systems.  If you can't figure out which one it is, you can message here and I can try to help you figure out.

Once you've got that extracted, you'll need to create a template .metainfo.xml and fill out the fields to make it match what is in fwupd.  The GUID above `72533768-6a6c-5c06-994a-367374336810` is generated by using this code https://github.com/hughsie/fwupd/blob/master/plugins/thunderbolt/fu-plugin-thunderbolt.c#L226 which will concatenate the string TBT with the model ID and vendor ID. 
1. Modify `<firmware type="flashed"> `to include your GUID. instead.
1. Modify `<id>com.hughski.ColorHugALS.firmware</id>` to be something like `<id>com.queuecumber.localTBT</id>`.
1. Change all the rest of the other fields to more appropriate stuff (license, etc).
1. Change the release section to have the right version.

Once you've got something that you think will work, you should be able to use `# appstream-util validate `to check it.

Once it validates:
1. Rename the binary file you found to match your system to `firmware.bin`.  
2. Name your metainfo `firmware.metainfo.xml`.
3. Package them up into a CAB using `gcab`.

Verify fwupd likes it.

`# fwupd get-details firmware.cab
`
If that works out and it matches your device you're in good shape.  This will all work even with the existing FW version.  If you can flush that process out and put it into a shell script or python script, we can put it in contrib for anyone to package up a random binary to use on their system locally.

Queuecumber, 2017-03-30:
Great that's really informative, I'll give it a shot with the currently released firmware version

superm1, 2017-03-30:
Oh I thought I mentioned it above, but re-reading I forgot.  If you need a `firmware.metainfo.xml` to start from, grab either one from a Dell BIOS CAB on LVFS and extract it, or start with the one here http://fwupd.org.s3-website-eu-west-1.amazonaws.com/vendors.html

Queuecumber, 2017-04-02:
Is there any way to manually wake the thunderbolt chip? I had to RMA my dock so I dont have anything to plug into it anymore. 

Also where can I see the model ID and vendor ID of the chip? I tried to reproduce the GUID using the info from lspci which should give TBT-80861576 as the starting point before hashing, but it didnt seem to get the right end result 

lspci:
```
07:00.0 PCI bridge [0604]: Intel Corporation DSL6340 Thunderbolt 3 Bridge [Alpine Ridge 2C 2015] [8086:1576]
```

Python GUID generation (https://github.com/hughsie/appstream-glib/blob/master/libappstream-glib/as-utils.c#L1452)
```
>>> uuid.uuid5(uuid.NAMESPACE_DNS, 'TBT-80861576')
UUID('0d98c33f-82d7-5be4-8f0c-1346f154d41b')
```

Also, I figured out from some more lspci output that dell seems to refer to my machine (the XPS 9560) by the hex string 0x07BE, so that is the firmware bin I've been using, but do you happen to know what that number means (if anything)? The firmware package I downloaded from Dell comes with three .bin files for three different numeric IDs, 0x075B, 0x07BF, and 0x07BE and for each an _nosecure.bin, _gfx.bin and no suffix .bin e.g:

```
0x07BE.bin  0x07BE_gfx.bin  0x07BE_nosecure.bin
```

You (and the documents I found) list only a single .bin per cab which I assume is the no-suffix .bin but do you know that for sure?

Queuecumber, 2017-04-08:
OK thanks, just a heads up, I had to use `tbtfwucli` to get the info, none of the other methods seemed to tell me what I needed

`lspci -v` for the thunderbolt controller
```
08:00.0 System peripheral: Intel Corporation DSL6340 Thunderbolt 3 NHI [Alpine Ridge 2C 2015]
	Subsystem: Device 2222:1111
```
2222:1111 is all kinds of wrong (and a suspicious ID number at that)

Looking at the processor as suggested in your example
```
00:00.0 Host bridge: Intel Corporation Device 5910 (rev 05)
	Subsystem: Dell Device 07be
```
07be is only half of what I need

using `dmidecode` to read the smbios
```
Handle 0x002A, DMI type 11, 5 bytes
OEM Strings
	String 1: Dell System
	String 2: 1[07BE]
	String 3: 3[1.0]
	String 4: 12[www.dell.com]
```
again 07be which is probably half of what I need

Finally
```
./tbtfwucli EnumControllers
08001575
./tbtfwucli GetVendorID 08001575
0x00d4
./tbtfwucli GetModelID 08001575
0x07be
```

and
```
>>> uuid.uuid5(uuid.NAMESPACE_DNS, 'TBT-00d407be')
UUID('72533768-6a6c-5c06-994a-367374336810')
```
which is the correct output. Not sure why those other commands weren't forwarded correctly but 

1. `tbtfwucli` seems to be the most reliable method for getting at this information
2. It looks like I could use the model ID to match a users hardware to the firmware.bin that is bundled with the executable

superm1, 2017-04-14:
Having to depend upon tbtfwucli would be pretty unfortunate.  I would sooner think it makes sense to just have an argument for which UUID (that comes from `get-devices `output) to use.

In terms of picking the right file, if you're writing in python I would just use lookup from `lspci `or `dmidecode `and then use `lower() `on both that and all the filename in the archive.
If you find a match with `lower() `you know which filename to use.

Queuecumber, 2017-05-03:
Still working on this but I havent had much time lately as we have had government evaluations of some of our code

Why do you think depending on tbtfwucli is bad? Don't we need it, or at least the underlying library, to be installed in order to make thunderbolt firmware updates work anyway? I wonder if I can just use the dbus API that thunderboltd exports, although allowing an arbitrary uuid probably works fine too

superm1, 2017-05-03:
Well I was thinking this can be a general helper script - not just for TBT.  So if the detail of how to build the UUID is abstracted (or special cased) it could be more general.

Queuecumber, 2017-05-12:
Got it, I think that's a good idea. I can have my script list the .bin files in finds in the exe file it is pointed at and prompt the user for which they want to use etc.

BTW I think I'm pretty close:

```
ㅌfwupdmgr get-details firmware.cab          ~/Desktop/firmware mehrlich@lilith
DellTBT
  Guid:                 72533768-6a6c-5c06-994a-367374336810
  UniqueID:             */*/14836d8043f6e7f27b2f867f8c749524f0bb48fa/firmware/net.queuecumber.DellTBT.firmware/*
  DeviceID:             08001575
  Description:          Firmware for Thunderbolt controllers on Dell laptops
  Flags:                internal|allow-online
  AppstreamId:          net.queuecumber.DellTBT.firmware
  Summary:              Dell TBT Firmware
  UpdateVersion:        3.21.00.008
  FilenameCab:          firmware.bin
  License:              proprietary
  Size:                 229.4 kB
  UrlHomepage:          http://www.dell.com/
  Vendor:               Max Ehrlich
  Trusted:              none
```

does that look OK?

superm1, 2017-05-22:
That's looking pretty close to me, yeah!

Queuecumber, 2017-08-15:
Long time no talk

So I finally did this but I'm not completely sure its working. I have a repo on my gitlab (https://gitlab.com/Queuecumber/firmware-packager) where I've been working on it, if you want it moved to another place I think I can move the file and its history.

That being said I just tried one of the cabs on the most recently tell thunderbolt release and I'm not really sure what happened. First off, fwupdmgr get-devices returns version 21 for my tbt firmware, whereas the dell version numbers look completely different (current version is 4.21.something). Next, when I told it to go ahead and install anyway even though the new version  looks like it has an older number, my screen flashed a bunch of times (which was scary) before eventually finishing the process. At this point, my version number still showed v21 even after reboot, so Im not sure that it actually applied the thing. 

Also I havent updated fwupd since we last talked in May. Let me know your thoughts

ybernat, 2017-08-15:
That's interesting. The flashing screen (I assume it's connected to a TBT port) sounds like the TBT plugin tried to update the FW (which includes power cycling the controller; the reason for the screen flashing), so I'd assume the pre-update image validation passed successfully, but the fact that the version didn't change may imply that the image authentication failed.

Where can I find this image file to have a look?
Did you try using tbtfwucli for getting the FW version from the image file, by any chance?
Did you try updating the controller by using tbtfwucli?

While we are on it, I want to inform you that TBT support in linux is going through big changes. Kernel v4.13 (currently in rc5) includes a TBT driver that supports:
- secure device connection (user approval for connecting devices)
- host *and device* FW update

I'm working on creating a new updated TBT plugin for fwupd to use this driver.

Queuecumber, 2017-08-15:
@ybernat Yes I'm aware of the changes and I cant wait for them to land

Actually neither screen is attached to a thunderbolt device, my external monitor is attached to  HDMI directly on the laptop. Both it and the internal laptop display were flashing. 

I didnt try asking `tbtfwucli` yet, can I just throw a cab at it and see what it has to say?

The firmware image I used is from [here](https://downloads.dell.com/FOLDER04421073M/1/Intel_TBT3_FW_UPDATE_NVM21_318RY_A01_4.21.01.002.exe) I extracted the `0x07BE_secure.bin` file

ybernat, 2017-08-16:
No, tbtfwucli needs the bin file itself.
Anyway, the file looks fine, its FW version is 21 (which explains why the version hasn't change after the update), and I can at most make some wild guesses about the rest... (possibly the eGPU drives all the graphics of the system when it's connected, including the internal display; no idea about Dell versioning).

Queuecumber, 2017-08-16:
Hmmm so the Dell version number is not quite correct. Do you think the firmware image applied? The weird thing is that the last firmware I applied (through Windows many months ago) was definitely not this image which just came on recently and it was also saying v21. Is there any way to confirm that the image was applied other than the process didnt spit out any errors?

Aside from that I know that @superm1 mentioned that he wanted a contrib script, do you think this is something that belongs with this repository in the contrib section or should it be kept separate

ybernat, 2017-08-16:
My guess is that the minor part of the version is the NVM version (21) and the major part is a Dell-specific way to follow the actual releases Dell does, but this is just a guess.
If there is no error in the log, I assume the flashing succeeded. The only way to be 100% sure is to use an image with a different version...

For the contrib script, I think it makes sense to have it here, as this is fwupd-specific, isn't it? But anyway this is for @superm1 and @hughsie to decide.

superm1, 2017-08-16:
@Queuecumber thanks for sharing.  
Yeah, I think that contrib script should be brought into contrib/ in fwupd rather than maintained elsewhere.  It can be packaged up and stored in `/usr/share` or something like that so people can use it without needing to download fwupd source but not execute as part of `$PATH`.
Can you submit a PR into fwupd?  The script isn't too complex, I don't think the history is too important to bring with it.

Regarding the versioning on that file you grabbed, don't read too much into the version string (`4.21.something`).  As it's a standalone executable distributed for windows the versioning applies to other things too (tools, TBT API, etc) aside from NVM version.  The important part for Linux is the `NVM21` you saw in the filename.

Queuecumber, 2017-08-16:
Got it I will prepare a PR later today, there are still some features, like an interactive mode and support for settings more variables, that I'd like to add. I can submit those as later PRs

PR #179 (2017-08-17): Add firmware-packager Script
These commits add a python script to create fwupd compatible cab files from Microsoft .exe files. This allows a user to update their firmware with fwupd even if the vensor hasn't yet made firmware packages available on the LVFS. 

Resolves #103
closed 2017-08-17 by superm1