fwupd/fwupd #104: reduce memory usage at idle

chergert, 2017-03-26:
fwupd seems to be taking about 45mb of resident memory (600+ virtual, but not super worried about that) on my system at idle. I've looked at the `/proc/pid/maps` to see what the primary offender is and it looks like its just the plain old heap (32 mb of the dirty bytes).

I wrote a quick tool to dump the memory found in the heap to get an idea of what was in there. But it's not very reliable because it's hard to know what degree of page ranges were uninitialized. But I do see an awful lot of appstream data that is not related to firmware.

Not trying to be annoying, just using this as a starting point to track down what characteristic causes the heap to grow without shrinking back down.

hughsie, 2017-03-26:
45MB does seem a lot. I'll have a look Monday and do some profiling. What appstream-glib and fwupd versions is this out of interest?

chergert, 2017-03-26:
* libappstream-glib-0.6.11-1.fc26.x86_64
 * appstream-0.10.5-2.fc26.x86_64
 * fwupd-0.8.1-2.fc26.x86_64

chergert, 2017-03-27:
I forgot to subtract `SHR` from `RES` in those previous numbers, so it's really about 10mb less.

 42492 - 9572 = ~32mb, which matches the heap size

Still a bit large, but 20% better already :)

hughsie, 2017-04-03:
Looking at the massif output, ~32Mb is the peak heap for me too, but then after loading the XML node tree it evicts anything uninteresting, going down to 813Kb, which is more what I expected. The output looks like this:

        MB
    32.71^                                                        #               
         |                                                     @@@#               
         |                                                   :@@@@#:              
         |                                                ::::@@@@#:              
         |                                              ::: ::@@@@#:              
         |                                           @@@: : ::@@@@#::             
         |                                       @:@:@ @: : ::@@@@#::             
         |                                     :@@:@:@ @: : ::@@@@#::             
         |                                  :@@:@@:@:@ @: : ::@@@@#::             
         |                               :@::@ :@@:@:@ @: : ::@@@@#:::            
         |                             @@:@::@ :@@:@:@ @: : ::@@@@#:::            
         |                          @@@@@:@::@ :@@:@:@ @: : ::@@@@#:::            
         |                       @::@ @@@:@::@ :@@:@:@ @: : ::@@@@#:::            
         |                    @@@@::@ @@@:@::@ :@@:@:@ @: : ::@@@@#:::            
         |                  :@@@@@::@ @@@:@::@ :@@:@:@ @: : ::@@@@#:::            
         |               ::@:@@@@@::@ @@@:@::@ :@@:@:@ @: : ::@@@@#::::           
         |            @@@::@:@@@@@::@ @@@:@::@ :@@:@:@ @: : ::@@@@#::::           
         |          :@@@@::@:@@@@@::@ @@@:@::@ :@@:@:@ @: : ::@@@@#::::           
         |      ::@::@@@@::@:@@@@@::@ @@@:@::@ :@@:@:@ @: : ::@@@@#:::::          
         |    :@: @::@@@@::@:@@@@@::@ @@@:@::@ :@@:@:@ @: : ::@@@@#:::::          
       0 +----------------------------------------------------------------------->Gi
         0                                                                   1.303

can you get a `sudo valgrind --massif-out-file=massif.out --threshold=0.01 --tool=massif fwupd --timed-exit` for me to see please? It could be the AppStream type filtering isn't being done on your system, but that would be pretty-odd...

chergert, 2017-04-05:
Maybe it's shrinking in response to exiting/cleanup releasing resources?

On my F26 installation:

```
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root      2083  0.0  0.5 622908 43640 ?        Ssl  Mar30   0:04 /usr/libexec/fwupd/fwupd
```

We have 622mb of virtual, 43mb of resident (of which 11mb is shared with other processes). If I grab `[heap]` from `/proc/2083/maps`, I get `559aa24b6000-559aa435c000` which is 32mb in size (so clearly hasn't shrunk). I'm running `fwupd-0.8.1-2.fc26.x86_64`.

I wrote a quick C program to seek/read/write the contents of the heap, and I've uploaded it to http://hergert.me/tmp/fwupd.heap.xz. After decompressing it, you should have a 32mb file you can `cat|strings|less` to inspect.

hughsie, 2017-04-05:
Hmm, so that contains all the unperged AppStream stuff. I'll do some more debugging this weekend on actual F26.

hughsie, 2017-06-16:
Today I moved the AppStream store into a private place, which means we don't have to scan all the other distro files at all. fwupd now starts up quite a lot faster too!

closed 2017-06-16 by hughsie

chergert, 2017-06-16:
Glad to hear it! Thanks for working on this!

pabloab, 2022-10-26:
In which version this fix will be available? On Ubuntu 20.04 (fwupd 1.7.9-1~20.04.1) seems to be the root process using the most amount of memory at any time/idle (using [ps_mem](https://github.com/pixelb/ps_mem)):

    $ sudo ./ps_mem.py -p $(pgrep -d, -u root)
    [....]
     80.8 MiB +   1.9 MiB =  82.7 MiB       fwupd

```
fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.4
compile   com.hughsie.libjcat           0.1.4
runtime   org.kernel                    5.4.0-128-generic
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.4
```

chergert, 2022-10-26:
> In which version this fix will be available?

Just a quick scan of the repository shows more than a thousand commits difference between 1.7.9 and 1.8.5 (which is what is in Fedora 37 here).

```
git diff --stat 1.7.9..1.8.5 src/ | tail -n1
93 files changed, 11781 insertions(+), 5908 deletions(-)
```

So chances are one of the couple dozen memory leak related fixes will improve things for you once your distribution updates or backports fixes.

For context, my machine here has uptime of about a month and memory is hovering around 18 MB private. Whether or not that is still "high" is beyond my knowledge of the code-base.

```
 Private  +   Shared  =  RAM used	Program

 18.0 MiB +   1.8 MiB =  19.8 MiB	fwupd
---------------------------------
                         19.8 MiB
=================================
```

hughsie, 2022-10-26:
You sure ps_mem is taking into account just RSS? We have a big data file we mmap into the process, and that shouldn't be counted as the kernel can evict it at any point.

chergert, 2022-10-26:
I'm not sure anything about `ps_mem`, but just installed it for some apples-to-apples comparison.

Locally, I'm seeing this for the mapped files:

```
7f280fb70000-7f280ff10000 r--p 00000000 00:21 2341280                    /var/cache/fwupd/metadata.xmlb
7f280ff10000-7f280ff32000 r--p 00000000 00:21 2198891                    /var/cache/fwupd/quirks.xmlb
7f28248ff000-7f2824900000 r--p 00000000 00:21 2198890                    /var/cache/fwupd/metainfo.xmlb
```

which is a bit less than 4MB of mapped memory anyway. So even if it doesn't take into account that file-backed pages are a non-issue, it's 14 vs 18. (Again, I still don't know if that's an even an issue, without taking a look at what's on the heap).

chergert, 2022-10-26:
Looking at gnome-system-monitor I get `8.6 MB`, which *does* actually report memory correctly (ps_mem *claims* to do that, but clearly we see different values here). You'll see a number that much more closely represents what is seen in `/proc/pid/maps` from g-s-m and so I think it's fair to say `ps_mem` cannot be trusted in this regard.