fwupd/fwupd #106: Support ChromeOS touchpad FW updater interface

superm1, 2017-04-12:
Many touchpad vendors have upstreamed drivers that were created from ChromeOS for touchpads.  These drivers happen to work even on non-Chromebook devices in many instances though too.

Google mandates the following interface for Touch Firmware Updater:
https://www.chromium.org/chromium-os/chromiumos-design-docs/touch-firmware-updater

Since this is just parsing some files in sysfs, I think this should likely be fairly easy to provide a fwupd plugin.

hughsie, 2017-05-23:
Do you know of any hardware that works standalone that we can purchase? We'd also need a repo of firmware for testing. Thanks!

superm1, 2017-05-23:
I don't know of any standalone touchpads, only those in laptops.  
The vendors I've talked to so far that already have ChromeOS touch firmware updater support would prefer to support via UEFI capsule long term though though.  So this might be irrelevant.

superm1, 2017-06-07:
I'll close this since there isn't any vendor willing to back this for now.

closed 2017-06-07 by superm1