fwupd/fwupd #107: Build fails on jhbuild

jwendell, 2017-04-13:
Making all in policy
make[2]: Entering directory '/home/jwendell/checkout/gnome/fwupd/policy'
  GEN      org.freedesktop.fwupd.policy
/home/jwendell/jhbuild/install/bin/msgfmt: cannot locate ITS rules for org.freedesktop.fwupd.policy.in
Makefile:628: recipe for target 'org.freedesktop.fwupd.policy' failed
make[2]: *** [org.freedesktop.fwupd.policy] Error 1
make[2]: Leaving directory '/home/jwendell/checkout/gnome/fwupd/policy'

hughsie, 2017-04-13:
I think you need a newer polkit. Polkit has been installing a .its file since about July last year.

jwendell, 2017-04-13:
hmm, last polkit release, 0.113 is dated back 2015... which is the version I'm using (polkit-0.113-5.fc24.x86_64)

jwendell, 2017-04-13:
Ideally we should have a new version of polkit released and then have fwupd depending on that. I'm going to close this one, thanks.

hughsie, 2017-04-14:
I've nagged the upstream maintainer a few times for a new tarball. In the meantime, I've backported the itstool functionality into rawhide, f26 and just doing f25 now. There's also a 0_8_X branch of fwupd which might be more suitable for jhbuild at this point.

superm1, 2017-04-14:
Backporting for the purpose of testing around makes fine sense but I think this should be kept open until the upstream maintainer for polkit makes a release with this ITS support, and an appropriate version dependency should be set in meson.build for that version of polkit.

pdecat, 2017-06-05:
What about this until a new polkit is released? https://github.com/Lyoncore/fwupd/commit/797e5672e64acd4f90cd52afca3bb192b28b4636

superm1, 2017-07-10:
@pdecat I've waivered on this, but decided that it makes more sense to make fwupd run in more places that don't yet have the newer deps.  It's not really reasonable to require folks to backport a patch from an unreleased version of software just to compile.
I've adopted some variation of that into master and now it should be easier to run 0.9.x under jhbuild or on _"older"_ distros (such as Ubuntu 17.04).

At some point we should revert that when most the common stable distros have picked up those missing dependencies (and polkit actually does a 0.114 release).

commit df430376 (2017-07-10): policy: fix compilation on a variety of configurations

* Meson >=0.41.0, (unreleased) polkit 0.114+ don't use data_dirs argument
* Meson >=0.41.0, older polkit, use local ITS rules with data_dirs argument
* Meson <0.41.0, any polkit, use custom policy building rules

Later on when meson 0.41.0+ and polkit 0.114+ is in many stable distros
this commit can be reverted and dependencies updated.

Fixes: #107
closed 2017-07-10 by superm1