fwupd/fwupd #108: get-devices freezes system

nightsky30, 2017-04-14:
Running the following through console freezes my system:

fwupdmgr get-devices

I must hard reboot.

Distribution:  Ubuntu 16.04.2 LTS
fwupd version:  0.7.0-0ubuntu4.3
Hardware:  Dell XPS 13 9360 Developer Edition

superm1, 2017-04-14:
can you comment on anything you have plugged into your system?  USB devices, docks, dongles, etc.

nightsky30, 2017-04-14:
Ah sorry,

Yes, I have:

Directly Plugged:
Dell DW316 DVDRW external USB drive
Logitech wireless mouse model M310 with USB adapter
Dell DA200 USB-C to HDMI/VGA/Ethernet/USB 3.0 adapter

Through the DA200:
Kensington USB Keyboard
VGA output to a Samsung S22A650D

superm1, 2017-04-14:
Initial thought would be the logitech plugin hanging, but I'm pretty sure it wasn't there for the 0.7 release that's in Ubuntu 16.04.

Can you please unplug everything and stop fwupd (`systemctl stop fwupd.service` should do the trick)

Start it back up yourself on a terminal (`sudo /usr/lib/fwupd/fwupd -v` IIRC for that version).

If it hangs with nothing plugged in share that log.

Assuming that works if you can go one by one introducing devices to see which is triggering this that would be helpful.  Any logging around when it hangs would be helpful too.

nightsky30, 2017-04-14:
I found it under:

sudo /usr/lib/x86_64-linux-gnu/fwupd -v

I'm attaching the output of the above command.  I can also say that immediately after unplugging everything, it did not crash running fwupdmgr get-devices.  I will begin plugging in each one by one.

Any particular log this error might show up in?

nightsky30, 2017-04-14:
It's the external monitor that's causing the freeze.  I did see a less critical error appear in console after plugging in the DA200 and restarting the service:

Apr 14 13:06:30 blackhole kernel: [ 2791.788295] usb 4-1.4: usb_reset_and_verify_device Failed to disable LTM
Apr 14 13:06:30 blackhole kernel: [ 2791.788295] .
Apr 14 13:06:30 blackhole kernel: [ 2791.977560] usb 4-1.4: USB disconnect, device number 3

That happened each time the fwupd service was restarted.

[fwupd_error_output.txt](https://github.com/hughsie/fwupd/files/922727/fwupd_error_output.txt)

nightsky30, 2017-04-14:
Shall I try the newer version here??

https://launchpad.net/~dell-team/+archive/ubuntu/ppa

hughsie, 2017-04-14:
> It's the external monitor 

Ohh wow, that's a new one on me. Is this connected using DVI/HDMI or has it got a USB hub as well?

superm1, 2017-04-14:
I wouldn't jump the gun on a newer version of fwupd just yet, but it may end up being an interesting data point.

The DA200 does have a USB hub in it. To me it's really interesting if it's part of this implication chain.  It's got another [high profile issue,](https://bugs.freedesktop.org/show_bug.cgi?id=93578) but I wouldn't expect it to have any bearing on this.
1. By chance do you have a regular type-C to DVI/HDMI dongle in addition to the DA200 you could try instead?

2. Was the monitor plugged in at the same time as the DA200?  Or you did DA200 and then monitor?

nightsky30, 2017-04-14:
@hughsie 
DA200:
http://accessories.ap.dell.com/sna/productdetail.aspx?c=sg&l=en&s=bsd&cs=sgbsd1&sku=470-abnl

Samsung S22A650D:
http://www.samsung.com/us/computer/monitors/LS22A650DS/ZA-specs

The DA200 plugs into the laptop via USB type-C/Thunderbolt 3.  It has VGA and HDMI out.  The external monitor has a VGA and DVI input.  So without any other crazy adapters to chain this monster together, I'm stuck using a VGA cable between the DA200 and monitor.

@superm1
With regard to the issue you reference there, yes, a lot of people have had issues with their video coming out of the DA200 (crazy colors, bad res, blank screens).  I've not had those issues with my VGA output, but I'm also running the 4.10.0 kernel available in xenial through the hwe-edge stack.  I've not been able to assess the HDMI yet.  I'd need a new monitor or a TV.  I suppose I could try a friend's.

One issue I have experienced with the external monitor is suspend/hibernate tend to cause the laptop and monitor not to wake properly, leaving me with blank screens, and forcing me to reboot.

I do not have a separate dongle.  I could order a USB type-C to DVI dongle.  Amazon must have something.  I don't have high hopes at finding that at a brick and mortar store.  Staples definitely does not.  I was actually there yesterday for something else, and was looking at their adapters.

The process I followed was connecting a single device, restarting the service, then running:

fwupdmgr get-device

If it didn't crash, I plugged in a new device, restarted the service, and ran the command again.  I started with just the DA200 by itself before adding the keyboard and finally the external VGA monitor.  The monitor happened to be the last device I added in the entire sequence, go figure.

nightsky30, 2017-04-14:
I'd also like to point out that I have the latest BIOS as of Jan 2017:  v1.3.2, and I've been checking daily for a new version.  I hope Dell releases one soon.  I'd love to be able to update the firmware on the Thunderbolt controller too, but as I understand it, that functionality is still in the works.

nightsky30, 2017-04-15:
I bought this Belkin USB type C to VGA adapter today.  I might not be able to test it until Monday unfortunately:

http://www.belkin.com/us/p/P-F2CU037/

nightsky30, 2017-04-16:
Ok...

So, with the new Belkin USB type C to VGA adapter on a different LG external monitor...No freeze, no error.

with the new Belkin USB type C to VGA adapter on the original Samsung external monitor...No freeze, no error.

It appears it isn't the monitor, but perhaps the VGA component of the DA200 when in use causing the freeze.

[Belkin_Adapter_With_Samsung_Ext_Mon.txt](https://github.com/hughsie/fwupd/files/924659/Belkin_Adapter_With_Samsung_Ext_Mon.txt)

superm1, 2017-04-17:
@nightsky30 Thanks for all your comments.
1. TBT host controller updates
The DA200 actually doesn't use TBT, it's USB-DP alternate mode.  So it shouldn't have any impact to this issue the version of the TBT FW update.
2. Behavior with the new adapter
Thanks for going out of your way to get another adapter.  This is quite surprising to me that there would be a hang caused from fwupd at all.  From the logs you've shared so far there isn't any indicative of the errors from the DA200.

My next thought for debugging would be to through this approach:
1. Upgrade to the newer fwupd on that PPA.  There have been changes to various plugins in that version,.  No direct fixes for this problem, but hopefully easier to root cause.
2. Stop the fwupd service.
3. Plug in all DA200 and failing monitor
4. Start the daemon back up.  Make sure to run in verbose.
5. In another terminal try the get-devices command.
Assuming the system still freezes, then check for relevant errors in the fwupd daemon terminal.  If it's frozen can you at least take a picture and look for similar messages in logs?

Not sure if Richard has some other ideas to debug this one.

hughsie, 2017-04-17:
I guess checking dmesg and journalctl too -- MCEs and that kind of thing are often printed there.

nightsky30, 2017-04-17:
So I tried again with version 0.7.0-0ubuntu4.3 before updating to the PPA, and I couldn't get it to crash anymore.  I just don't understand what the difference is.

Everything is plugged in the exact same way.

When you say you can't see anything that indicates fwupd is causing it, perhaps there was another factor or cause and fwupd was just triggering it.  But when it was crashing, it was over multiple tries and multiple reboots.  I had to reboot just to recover.  And I must have caused it to crash and reboot at least 5 times the same way with get-devices.

Here's the latest output, minus dmesg and journalctl because there wasn't a crash.

[New Output - fwupd verbose.txt](https://github.com/hughsie/fwupd/files/926880/New.Output.-.fwupd.verbose.txt)
[New Output - get devices.txt](https://github.com/hughsie/fwupd/files/926882/New.Output.-.get.devices.txt)

nightsky30, 2017-04-17:
I'm stumped.  I will let you know if I experience it again.

nightsky30, 2017-04-17:
Also, thank you on the clarification with the port and its different modes.  I was not aware a device could operate through it not using TBT host controller. :)

superm1, 2017-04-18:
Sure.  I'm going to mark this closed unless you can find a somewhat reliable reproducer for it.
Something precarious though (possibly unrelated):
Your `get-devices` output shows this:
```
usb:00:01:03
  Guid:                 7a51fe38-0180-525c-af99-90ad3e5432a2
  DisplayName:          BILLBOARD DEVICE
  Provider:             USB
  Flags:                none
  Version:              1.70
  Created:              2017-04-17
  Trusted:              none
```
Are you having any other problems with DA200 plugged in right now?  The billboard device is only supposed to show up when there was a problem.  From [USB Billboard spec](http://www.usb.org/developers/docs/devclass_docs/USB_Billboard_Rev_1.21_20160921.zip):

> If the Port Pair exits Modal Operation due to a fault condition, the Device Container shall expose the Billboard Device 

As far as i'm aware nothing in Linux userspace will parse the the billboard descriptors to tell you a popup of what's going on like Windows does.  Can you share verbose lsusb output while that billboard device is present though?  It might be able to tell more by matching it to that spec I linked.

closed 2017-04-18 by superm1

nightsky30, 2017-04-18:
Ya, I believe that billboard device shows up when using the DA200.  I thought perhaps it was just a strange name for the hardware.  I will gladly provide lsusb info.  Understood on the current closed state of the issue.

nightsky30, 2017-04-18:
Ok,

So I ran:

fwupdmgr get-devices

I received a timeout error.  I decided to see if the fwupd service was running, and it was not.  I've rebooted a few times today.  I guess it isn't starting on boot.

I started the service.  I then checked its sttus a few times to make sure it wasn't crashing.  I then ran:

fwupdmgr get-devices

and I got the list, including the BILLBOARD entry with identical guid as you listed above.

So I ran:

sudo lsusb -v

And this is what the console output followed with my system freezing:

Bus 004 Device 006: ID 0bda:8153 Realtek Semiconductor Corp.

Here's the crash output from syslog:

[lsusb_crash.txt](https://github.com/hughsie/fwupd/files/931169/lsusb_crash.txt)

I could try lsusb -v again...

nightsky30, 2017-04-18:
This time lsusb worked.  I also pulled some stuff out of journalctl:

[jornalctl_usb.txt](https://github.com/hughsie/fwupd/files/931253/jornalctl_usb.txt)

[new_lsusb.txt](https://github.com/hughsie/fwupd/files/931261/new_lsusb.txt)