fwupd/fwupd #109: sysconfdir ends up under prefix

superm1, 2017-04-14:
Currently the meson build is configured that sysconfdir ends up underneath prefix (https://github.com/hughsie/fwupd/blob/master/meson.build#L167).  This means that files are sitting in /usr/local/etc by default for example.  Shouldn't by default this not join with prefix?

superm1, 2017-04-14:
I believe same problem with systemd units.  They're ending up under /usr/local/lib/systemd/system when they should just be /lib/systemd/system.

hughsie, 2017-04-15:
Are you setting a prefix when you compile? For local builds I use `meson --prefix=/home/hughsie/root ..` and for rpm's the prefix gets set automatically. If you don't set a prefix I assume the defaults are used.

superm1, 2017-04-15:
Yeah I am setting it to /usr. "Everything" ends up under /usr that means whereas autotools build only relevant things got put there. The rest are put in correct place on system.

Expectation should be it puts everything in the right place for a non packaged run without setting prefix (which currently everything is under /use/local)

hughsie, 2017-04-17:
I'm a bit confused; prefix is supposed to be the prefix for the install. I use `meson --prefix=~/.root ..` and then I get ~/.root/etc as expected. This is what fedora uses for meson: https://fedoraproject.org/wiki/Packaging:Meson

superm1, 2017-04-17:
I think you're mixing up the definition of prefix to be "installation directory".  You should be setting `DESTDIR `for the installation directory.
In the example [guide](https://github.com/mesonbuild/meson/wiki/Quick-guide) they show: 
```
$ cd /path/to/source/root
$ CFLAGS=... CXXFLAGS=... LDFLAGS=.. meson --prefix /usr --buildtype=plain build
$ ninja -v -C build
$ ninja -C build test 
$ DESTDIR=/path/to/staging/root ninja -C build install
```
Other packages were still using --prefix=/usr when compiling too in Debian today ( eg: https://anonscm.debian.org/git/pkg-packagekit/appstream-generator.git/tree/debian/rules#n5 )

superm1, 2017-04-19:
Closed with https://github.com/hughsie/fwupd/commit/2d8a7958b2208ff4df8f93b0353ac99dc4519881 https://github.com/hughsie/fwupd/commit/e07a6ddf6e04447aededd3c592dc0f130f9436c2 https://github.com/hughsie/fwupd/commit/ff8a02f9f9bdc7c492e5f35d48b2ccd93b2f1ad8

closed 2017-04-19 by superm1

superm1, 2017-04-19:
Sure, no problem.  Just one more I found too and added in.

superm1, 2017-04-19:
Something worth keeping an eye on is what meson upstream does with this: https://github.com/mesonbuild/meson/issues/1637

To me /usr/var by default makes no sense.

QuLogic, 2017-04-19:
> I think you're mixing up the definition of prefix to be "installation directory". You should be setting `DESTDIR` for the installation directory.

No, I think you might be confusing things. If someone wants to install in `$HOME/.local`, they should use `--prefix`. `DESTDIR` is almost exclusively for packagers.

superm1, 2017-04-19:
> No, I think you might be confusing things. If someone wants to install in $HOME/.local, they should use --prefix. DESTDIR is almost exclusively for packagers.

That jives entirely with what I was saying (although fwupd is a bad example because it has pieces that won't be able to live in $HOME).

My goal is that someone should be able to do:
```
mkdir -p build
pushd build; meson ..; popd
ninja -Cbuild
sudo ninja -Cbuild install
```
and the software works properly.  That would mean that `--prefix` defaulted to` /usr/local `and the binaries were all put in the right places under that prefix.  Some stateful bits (in `/etc` and` /var`) would not sit under the default prefix.

With what's in master today this all works.
`DESTDIR `is being used in fedora and debian packaging (along with `--buildtype=plain `and `--prefix=/usr`)