fwupd/fwupd #110: fwupd breaks Logitech K800 (Unifying) behind KVM

ziesemer, 2017-04-15:
I have a Logitech K800 (Unifying), behind a ATEN CS1764A KVM.  The Unifying USB receiver is plugged into the keyboard input on the KVM, and allows me to use all the KVM hotkey functionality.  This has worked flawlessly across multiple Windows and Linux workstations and versions over the past ~2 years, until upgrading to Ubuntu 17.04 just now - which includes fwupd 0.8.1-3.

The keyboard works fine when booting, using a text virtual console (tty1), or even at the Gnome login screen.  However, the keyboard stops responding after logging-in to a Gnome session, with the following seen in `/var/log/syslog`:

```
Apr 15 09:59:16 z-util dbus[1310]: [system] Activating via systemd: service name='org.freedesktop.fwupd' unit='fwupd.service'
Apr 15 09:59:17 z-util fwupd[3666]: disabling plugin because: failed to startup dell: Firmware updating not supported
Apr 15 09:59:17 z-util fwupd[3666]: ignoring add-delay as device usb:00:01:04 already pending
Apr 15 09:59:17 z-util fwupd[3666]: ignoring add-delay as device usb:00:01 already pending
Apr 15 09:59:17 z-util fwupd[3666]: ignoring add-delay as device usb:00:01:01:04:01 already pending
Apr 15 09:59:17 z-util fwupd[3666]: Failed to add Logitech device: failed to resetfailed to get data: USB error on device 046d:c52b : Input/output error [-1]
Apr 15 09:59:17 z-util kernel: [   28.154182] usb 1-1.1.4.1: usbfs: process 3666 (fwupd) did not claim interface 3 before use
```

The device remains visible from `lsusb`:

`Bus 001 Device 049: ID 046d:c52b Logitech, Inc. Unifying Receiver`

... but is no longer visible as a recognized input device under `/dev/input/by-id/`.

Other issues already reviewed:

* #101 - "Failed to add Logitech device" and unusable Logitech devices
* #108 - get-devices freezes system

Current workarounds include:

* Moving the Logitech Unifying USB receiver to a USB input other than the KVM keyboard input - at a loss of all hotkey functionality for all devices.
* Changing the KVM port input to "SPC mode", where it emulates its own standard (104 key) keyboard/mouse - at a loss of use of the multimedia keys on the keyboard.
* systemctl mask fwupd

Until there is a proper fix for this, is there a way to leave fwupd enabled, but to "blacklist" the Logitech support and/or a given USB device path?

hughsie, 2017-04-16:
So, we probably need a way to block specific plugins anyway, but I'm wondering if you try branch 0_8_X with this commit https://github.com/hughsie/fwupd/commit/ea8ebcfc7b0b79aa1c199c1abd84b37c522c5445 the problem goes away? We're not doing a device replug (unless flashing the firmware) with the new code so I think it should be less troubling to a KVM.

superm1, 2017-04-17:
@ziesemer on Ubuntu 17.04 to try that commit on the packaged build:

1. `apt source fwupd`
2. `apt build-dep fwupd`
3. download that commit, put it in debian/patches
4. add a reference to that commit in debian/patches/series
5. `dch -i` increment the changelog to something different (eg 0.8.1-3ubuntu0local1)
6. `debuild -us -uc` build the package
7. `sudo dpkg -i ../*0.8.1-3ubuntu0local1*.deb` install the packages you just built

superm1, 2017-04-17:
Oh and I don't think there is anything else relevant to this problem on 0_8_X right now that's not already in 0.8.1.  (Building on the branch is a little more tedious with debian packaging if it can be avoided)

ziesemer, 2017-04-17:
Thank you both!  05ede65 is also much appreciated.

In the next day or so, how about I just test this by comparing a clean 0.8.1 build, and one or more of (master, the 0_8_X branch, ea8ebcf) - without any of the Debian packaging?  I'm guessing I should be able to easily duplicate the issue by just running the service directly?

superm1, 2017-04-17:
Yeah, I didn't want to presume your technical skill levels for running unpackaged software.  If you're comfortable running an unpackaged build that could provide better information.  In that case run 0_8_X with ea8ebcf added on top.

ziesemer, 2017-04-19:
Sorry - is there anyway to run this after "make" without doing a "make install" and otherwise taking over my system?  (It looks like I can't simply run fwupdmgr without also having systemd running the daemon.)  Already removed the build dependencies I'd rather not keep around.  Otherwise I'm probably going to move my efforts here to a VM, and hopefully USB pass-through will suffice for testing...

hughsie, 2017-04-20:
I've released a 0.8.2 tarball with this functionality if that makes it easier to test.

superm1, 2017-04-20:
@ziesemer i've updated packaging for this recent 0.8.2 release.  if you instead want to do a packaged build you can basically do these steps:
```
# git clone https://anonscm.debian.org/cgit/uefi/fwupd.git/
# pushd fwupd
# git checkout debian
# uscan --download-current-version
# debuild
```
That will get you a packaged set of debs you can install.  If you want to test with https://github.com/hughsie/fwupd/commit/ea8ebcfc7b0b79aa1c199c1abd84b37c522c5445 on top, put it in `debian/patches` and a reference to it in `debian/patches/series` and rebuild.

hughsie, 2017-05-23:
The new unifying plugin in 0.9.x does not detach the kernel module anymore, unless the device is to be upgraded.

closed 2017-05-23 by hughsie