fwupd/fwupd #112: XPS 15 9560 Stuck at DELL-screen after update fdab6b6740a254c640ef6c109ff7692e1d1d0531-XPS9560v1.2.4.cab

zinob, 2017-04-18:
Cross-posting from https://bugzilla.redhat.com/show_bug.cgi?id=1443141

Description of problem:
After running the latest update, my XPS15 freezes at the DELL screen. Rebooting and hammering F2 displays the "entering setup" text, but nothing happens, rebooting and  hammering the F12 displays the "Entering boot options" but then nothing happens.

Software Manager has been nagging me for a few days so the date of "fdab6b6740a254c640ef6c109ff7692e1d1d0531" seems likely.
Following the Dell-unbrick routines from the DELL support-page. ALT-ESC without USB-drive inserted causes the laptop to hang at a black screen. If i insert a BIOS-Recovery-USB-drive it hangs at the DELL-screen instead.

Version-Release number of selected component (if applicable):
fdab6b6740a254c640ef6c109ff7692e1d1d0531-XPS9560v1.2.4

How reproducible:
Hopefully not very..

Steps to Reproduce:
1. Install the latest update
2. Call Dell to get a technician to come and flash/replace the motherboard ;)

Actual results:
After reboot and upgrade the laptop does not progress beyond the DELL-logo

Expected results:
A better and more awesome BIOS

superm1, 2017-04-18:
Sorry to hear this happened.  It shouldn't be a bug in fwupd or fwupdate that caused this as it's handed over control to the BIOS after the flash process has started.  
Can you bring this over to the Dell forums: http://en.community.dell.com/techcenter/os-applications/ ?

Make sure you mention the steps that you followed for BIOS recovery so far.

superm1, 2017-04-18:
(I'm marking the fwupd issue as closed unless something comes up to indicate this is actually a fwupd bug)

closed 2017-04-18 by superm1

zinob, 2017-04-18:
I agree that it is probably not a bug in fwupd. I mostly posted it here in case you wanted to filter it from the update-list (espec if there are more reports). I can post it to the dell-forums but I doubt that they will do much unless there is a large number of affected windows-users.

superm1, 2017-04-18:
You're the first I've heard on this, certainly hope it's an isolated incident.  There are people that do monitor the forums, and most importantly if lots of people are affected will chime in.

Dellscreenrepair, 2017-12-14:
This was nice post , you  provide a information its knowledgeable thanks to share , If you have
any issue regarding your Samsung Screen then feel free to visit.

[](https://laptopscreen.ae/dell.html)

ybernat, 2017-12-14:
Spam comments in GitHub? I had no idea this is a thing.

hughsie, 2017-12-14:
@ybernat the first I've seen; I've deleted it now.