fwupd/fwupd #115: Thunderbolt plugin needs to be updated for changes in kernel interface

superm1, 2017-05-18:
Today, Intel submitted a new patch series to the kernel that reworks the way that the kernel handles Thunderbolt devices.
The new series supports security level support as well as natively flashing NVM directly from sysfs.
eg:
`# dd if=KYK_TBT_FW_0018.bin of=/sys/bus/thunderbolt/devices/0-0/nvm_non_active0/nvmem`

As a result, my interpretation is that libtbtfw really isn't required anymore as a build dependency or to be linked with.  All of the attributes related to TBT will be available over udev and uevents and should indicate devices being added and removed.

Furthermore the new interface supports updating the NVM of other devices on the bus, so those can now be considered by the plugin.

superm1, 2017-05-18:
@ybernat have you started work on these changes?  If so, when you're ready please share the development branch with this bug.

superm1, 2017-05-18:
I haven't yet actually compiled and tested it yet, but this stood out to me in the series:

> Starting from Intel Falcon Ridge the NVM firmware can be upgraded by
> using DMA configuration based mailbox commands. If we detect that the
> host or device (device support starts from Intel Alpine Ridge) has the
> DMA configuration based mailbox we expose NVM information to the
> userspace as two separate Linux NVMem devices: nvm_active and
> nvm_non_active. **The former is read-only portion of the active NVM which
> firmware upgrade tools can be use to find out suitable NVM image if the
> device identification strings are not enough.**
> 
It's unclear to me the circumstances that this would happen, but if it does happen there would be a remaining question in my mind if that responsibility would fall on the fwupd plugin or a separate library.

hughsie, 2017-05-19:
I'm also not happy to do all the work (again) in fwupd as I didn't even get so much of a heads up that libtbtfw isn't the way forward. I'm happy to help, but not write a provider.

superm1, 2017-05-19:
Well I think we need @ybernat to confirm if libtbtfw is going away or what the direction is.

ybernat, 2017-05-22:
1. This is the Linux SW stack re-architecture we mentioned long ago. Finally it happens :)
2. The libtbtfwu library will continue to be used, mainly for implementing the image validation logic (all the comparisons done to make sure the image is compatible with the controller).
3. We strive to keep libtbtfwu API the same while changing the implementation to use the new sysfs interface exposed by the new kernel module.
4. Our plans are to have the new library available together with kernel v4.13 release.

superm1, 2017-05-22:
@ybernat Thanks for your comments and sharing the plan.  I was aware of the planned rearchitecture but was missing a lot of details on how it would be done in userspace.
It's good to hear that you plan the libtbtfwu API to remain the same.

From my experimentation with the new kernel sysfs interface, there's a few things that stand out that I think will cause either API changes or fwupd changes though.  What do you think?
* Text description nodes of the controller or device are exported in sysfs for the different devices.  Since those are now available they should be used to describe the FuDevice for `fwupdmgr get-devices` output.  They should either be parsed from sysfs or come from libtbtfwu's API rather than hardcoding "Thunderbolt Controller".
* `fu_plugin_thunderbolt_device_matches `matches a hardcoded list of device ID's.  This should really instead just match for the new thunderbolt subsystem uevents.
* Security level probably needs to be considered now if devices aren't authorized.  Can they be flashed if not authorized?  If not, UPDATE_ONLINE shouldn't be added to them.
* fwupd only had supported host controller flashing, but the new kernel support allows device flashing.  I think some assumptions were made about only handling the host devices (like hardcoding device ID's, possibly others)
* Special dell lookup stuff in safe mode should only apply to host controllers, so when devices are plugged in not at the top of topology that code shouldn't run on them.

ybernat, 2017-05-22:
- Text description - this is mainly relevant for EP (device) FW update, which wasn't support till now. From the two options you mentioned, what do you prefer, having the plugin reading it from sysfs directly or adding a relevant API to the library?
- Device matches - yes, we can use it instead of updating the plugin's hardcoded list each time. Just let's keep in mind that the image validation logic in the library still have to be updated when a new controller is supported.
- Security level - this isn't an issue. Device can be flashed even before it is authorized.
- EP (device) support - the device IDs list is less of a concern here, but the image validation logic is a bit different (which is part of the library update we plan).
- Dell lookup - when host controller is in safe-mode, devices can't be connected. Also, devices can't be in safe-mode themselves so `tbt_fwu_Controller_isInSafeMode` always returns 0 on them and we don't enter this `if`.

So yes, seems like the plugin needs some code updates but they should be minor.

BTW, is there any way to force a specific order of the updates? If there is a long chain of devices, it's generally preferred to update them starting from the further one and continuing up in the chain towards the host. Does fwupd support such an ordering?
To be clear: this isn't a hard requirement, just for being on the safe side.

superm1, 2017-05-22:
@ybernat My preference would be to keep the thunderboltd /libtbtfwu API the same as before but instead read the new bits directly from sysfs in the fwupd TBT plugin.  It might even make sense to read some things that were previously provided from API directly from sysfs in the fwupd TBT plugin.
Since the main thing the library is now supporting is image validation I think having the TBT plugin own parsing sysfs files and look for the uevents on the changes makes most sense.

With library needing to be updated for new controllers, does this mean like in an new generation of controllers or even for device ID's in a current generation?  Definitely would like to move the hardcoded list out of fwupd, the less things that need to update at once the better.

Good on the device and safe mode support.

The order of updates is controlled by fwupd walking through an array listing out devices.  Those have been added in order from plugins (see `fu_main_plugin_device_added_cb`).  I'd expect that once you have EP support you would export them in a logical order (host controller -> cable -> device) and it would currently do exactly the opposite of what you want.  Even if you tried to artificially do this, it would be problematic because a device hotplugged later would cause a problem.

My opinion is the solution should be allowing plugins to set an optional internal weight for devices.  Everything should start at 0 and if a plugin sees a need to higher the priority of various devices it can add a weight setting.  For you that should work well in that host controllers would be set at 0, and then EP's on the first level (like a cable with AR) 1, and second level (like a dock) 2, etc.  fwupd daemon can  then look for the highest priority setting across plugins and do those first.
This also has the advantage that it can help with plugin interactions.  For example if it always makes more sense to update the MST hub that's in a dock before TBT maybe the MST plugin might set a priority of 3.
@hughsie thoughts?

ybernat, 2017-05-22:
@superm1 
For the library update with new controllers - yes, this is for each device ID, as even inside the generation itself there is a variety of what should be tested for image validation.

The order - as I mentioned, this isn't a must, but if you see a place for it in fwupd anyway, this can be useful.
If we go this way, I'd suggest keeping some ranges in between. In your example, MST plugin should use a number higher than 6, as TBT may set the device priority to 6 if it's the last in the chain.

hughsie, 2017-05-23:
> I think having the TBT plugin own parsing sysfs files

Agreed.

> allowing plugins to set an optional internal weight for devices. 

Rather than setting an integer, why can't we just codify the dependency explicitly? e.g. if GUID aaaa AND bbbb are both updatable, then bbbb must be updated before aaaa.

Richard.

ybernat, 2017-05-23:
GUID based approach solves TBT vs. MST case nicely (and less fragile, I think), but the device location in chain is not part of the GUID creation. Mario's suggestion covers them both.

hughsie, 2017-05-23:
The problem with assigning random integers is how to deal with future proofing. Does updating a Logitech USB device (integer 2) have to be done before updating a Logitech USB receiver (integer 1)? If we have a Logitech Unifying hub in the future, do we need a 1.5 integer state?

I'd much rather have an explicit rule, even if it's just comparing an integer value on some metadata key. For instance, setting a metadata key "tbt-chain-level"="2" and then having a rule that says ORDER_BEFORE(tbt-chain-level"="2).

superm1, 2017-05-23:
Well I guess we really need to validate how important it is for inter-plugin dependencies so this isn't over-engineered.

The reason I was raising TBT and MST as an example is i'm not sure if the TBT controller resetting will kick a monitor off the DP port and USB or USB devices off.  @ybernat any idea does that happen today when the TBT controller gets reset?

ybernat, 2017-05-23:
For DP, there is no need for device authorization so it'll be reconnect after the controller power cycle done in the FW update process.
For USB, I think it depends on a few parameters, but generally it depends on the device to get authorized.

superm1, 2017-05-23:
Ah so that's a good point.  If the device authorization wasn't done via a UDEV rule, it's possible that it may be configured to prompt on every boot meaning those USB devices behind a USB hub might not show up after the TBT controller was power cycled.

Then if you're figuring out whether to flash TBT or the USB unifying receiver first that means you have to actually know how it's hooked up...

Rather than encode the entire topology of all devices in metadata, how about instead:
- Set up a explicit codified plugin priority in fwupd for devices that update online.  All plugins shipped in tree would have online update devices be processed in that order.
- Within a given plugin optionally create a topology based integer weighting system for device metadata.

**Future proofing**
The situation of future proofing doesn't break when the plugin owns assigning the integer weighting.   
If today the USB device is integer 2 and the the USB receiver is integer 1, that doesn't mean that they need to stay that way in the plugin.  
If you determine later that this new unifying hub needs to be processed between the two, you change the enum around and the USB device is 3, hub is 2 and receiver is 1.

Since it's all contained in the plugin this should be a private structure to produce weighting.

If some new plugin comes along that say supports USB host controllers, that plugin would get put into the list after all the various USB plugins.

superm1, 2017-08-21:
Closed by https://github.com/hughsie/fwupd/commit/73ba7ee82c91ff61dbebba768e7e79a2fedba1e7

closed 2017-08-21 by superm1