fwupd/fwupd #117: "g_usb_device_close: assertion 'G_USB_IS_DEVICE (device)' failed" logged every 5 seconds

rathann, 2017-05-23:
On Fedora 25, I installed `fwupd-0.9.2-1.fc25.x86_64` from koji. After starting fwupd service (`systemctl start fwupd`), the following is logged in journal every 5 seconds:
```
May 23 09:51:03 localhost systemd[1]: [/usr/lib/systemd/system/fwupd.service:13] Unknown lvalue 'ProtectControlGroups' in section 'Service'
May 23 09:51:03 localhost systemd[1]: [/usr/lib/systemd/system/fwupd.service:15] Unknown lvalue 'ProtectKernelModules' in section 'Service'
May 23 09:51:42 localhost systemd[1]: [/usr/lib/systemd/system/fwupd.service:13] Unknown lvalue 'ProtectControlGroups' in section 'Service'
May 23 09:51:42 localhost systemd[1]: [/usr/lib/systemd/system/fwupd.service:15] Unknown lvalue 'ProtectKernelModules' in section 'Service'
May 23 09:51:42 localhost systemd[1]: Starting Firmware update daemon...
May 23 09:51:42 localhost fwupd[30542]: g_usb_device_close: assertion 'G_USB_IS_DEVICE (device)' failed
May 23 09:51:42 localhost fwupd[30542]: Failed to add Logitech device: failed to read device config: invalid value
May 23 09:51:42 localhost fwupd[30542]: disabling plugin because: failed to startup dell: Firmware updating not supported
May 23 09:51:42 localhost fwupd[30542]: ignoring add-delay as device usb:00:05 already pending
May 23 09:51:42 localhost fwupd[30542]: ignoring add-delay as device usb:00:03 already pending
May 23 09:51:42 localhost fwupd[30542]: ignoring add-delay as device usb:00:02:02 already pending
May 23 09:51:42 localhost fwupd[30542]: ignoring add-delay as device usb:00:02:01:04 already pending
May 23 09:51:42 localhost fwupd[30542]: ignoring add-delay as device usb:00:02:01:03 already pending
May 23 09:51:42 localhost fwupd[30542]: ignoring add-delay as device usb:00:02:01:01 already pending
May 23 09:51:42 localhost fwupd[30542]: disabling plugin because: failed to coldplug uefi: UEFI firmware updating not supported
May 23 09:51:42 localhost fwupd[30542]: disabling plugin because: failed to coldplug raspberrypi: Raspberry PI firmware updating not supported, no /boot/start.elf
May 23 09:51:42 localhost fwupd[30542]: disabling plugin because: failed to coldplug synapticsmst: MST firmware updating not supported by OEM
May 23 09:51:42 localhost systemd[1]: Started Firmware update daemon.
May 23 09:51:49 localhost fwupd[30542]: g_usb_device_close: assertion 'G_USB_IS_DEVICE (device)' failed
May 23 09:51:49 localhost fwupd[30542]: g_usb_device_close: assertion 'G_USB_IS_DEVICE (device)' failed
...
```

hughsie, 2017-05-23:
Can you stop the service and run:

    sudo /usr/libexec/fwupd/fwupd --verbose

And then attach the full output here please. Thanks!

rathann, 2017-05-23:
[fwupd-output.txt](https://github.com/hughsie/fwupd/files/1021797/fwupd-output.txt)

rathann, 2017-05-23:
New log with
``` diff
diff --git a/plugins/unifying/lu-device.c b/plugins/unifying/lu-device.c
index 2c14ceb..ac3e68b 100644
--- a/plugins/unifying/lu-device.c
+++ b/plugins/unifying/lu-device.c
@@ -746,7 +746,7 @@ lu_device_open (LuDevice *device, GError **error)
 	/* subclassed */
 	if (klass->open != NULL) {
 		if (!klass->open (device, error)) {
-			g_usb_device_close (priv->usb_device, NULL);
+			lu_device_close (device, NULL);
 			return FALSE;
 		}
 	}
@@ -754,7 +754,7 @@ lu_device_open (LuDevice *device, GError **error)
 
 	/* subclassed */
 	if (!lu_device_probe (device, error)) {
-		g_usb_device_close (priv->usb_device, NULL);
+		lu_device_close (device, NULL);
 		return FALSE;
 	}
 
```
applied on top of 0.9.2-1.fc25 from koji.
[fwupd-output-with-patch.txt](https://github.com/hughsie/fwupd/files/1021910/fwupd-output-with-patch.txt)

The failed assertion is no longer cluttering the logs, but, as you can see, the daemon is still doing the open/close/fail loop every 5 seconds:
```
11:11:32:0589 Unifying opening unifying device using /dev/hidraw2
11:11:32:0589 Unifying [HID] host->device:10 ff 81 f1 01 00 00 
11:11:32:0590 Unifying [HID] device->host:10 ff 81 f1 01 12 01 
11:11:32:0590 Unifying [HID] host->device:10 ff 81 f1 02 00 00 
11:11:32:0592 Unifying [HID] device->host:10 ff 81 f1 02 00 19 
11:11:32:0592 Unifying [HID] host->device:10 ff 81 f1 03 00 00 
11:11:32:0594 Unifying [HID] device->host:10 ff 8f 81 f1 03 00 
11:11:32:0594 Unifying closing device
11:11:32:0594 Unifying failed to open /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2.1/2-2.1.4: failed to read device config: invalid value
11:11:37:0595 Unifying opening unifying device using /dev/hidraw2
11:11:37:0595 Unifying [HID] host->device:10 ff 81 f1 01 00 00 
11:11:37:0598 Unifying [HID] device->host:10 ff 81 f1 01 12 01 
11:11:37:0598 Unifying [HID] host->device:10 ff 81 f1 02 00 00 
11:11:37:0600 Unifying [HID] device->host:10 ff 81 f1 02 00 19 
11:11:37:0600 Unifying [HID] host->device:10 ff 81 f1 03 00 00 
11:11:37:0602 Unifying [HID] device->host:10 ff 8f 81 f1 03 00 
11:11:37:0602 Unifying closing device
11:11:37:0602 Unifying failed to open /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2.1/2-2.1.4: failed to read device config: invalid value
```

rathann, 2017-05-23:
Note that this open/close/fail cycle is logged even if I'm not typing anything or moving the mouse.

rathann, 2017-05-23:
With the next patch:
``` diff
diff --git a/plugins/unifying/lu-device-runtime.c b/plugins/unifying/lu-device-runtime.c
index 96487c2..c32a8d1 100644
--- a/plugins/unifying/lu-device-runtime.c
+++ b/plugins/unifying/lu-device-runtime.c
@@ -117,7 +117,7 @@ lu_device_runtime_open (LuDevice *device, GError **error)
 		msg->function_id = HIDPP_REGISTER_DEVICE_FIRMWARE_INFORMATION;
 		msg->data[0] = i;
 		if (!lu_device_hidpp_transfer (device, msg, error)) {
-			g_prefix_error (error, "failed to read device config: ");
+			g_prefix_error (error, "failed to read config idx %u: ", i);
 			return FALSE;
 		}
 		memcpy (config + (i * 2), msg->data + 1, 2);
```
applied, the `failed to open` line changes to this:
```
11:39:58:0074 Unifying failed to open /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2.1/2-2.1.4: failed to read config idx 3: invalid value
```

rathann, 2017-05-23:
```
# lsusb -v -d 046d:

Bus 002 Device 013: ID 046d:c52b Logitech, Inc. Unifying Receiver
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x046d Logitech, Inc.
  idProduct          0xc52b Unifying Receiver
  bcdDevice           12.01
  iManufacturer           1 Logitech
  iProduct                2 USB Receiver
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          4 RQR12.01_B0019
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower               98mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      59
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     148
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               2
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      98
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               2
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

closed 2017-05-23 by hughsie