fwupd/fwupd #120: Build instructions?

sixtyfive, 2017-05-25:
As this doesn't use `./configure && make && make install` it'd be nice to have a little section about installing in the README file, esp. given https://blogs.gnome.org/hughsie/2017/05/22/updating-logitech-hardware-on-linux/ which only gives instructions for Fedora users.

For Debian Stretch, I got as far as:

```
apt purge libfwup1 fwupd
apt build-dep fwupd
apt install meson
git clone https://github.com/hughsie/fwupd.git
cd fwupd
mkdir build; meson . build # just guessing here...
```

Unfortunately that fails with:

```
Native dependency appstream-glib found: NO found '0.6.8' but need: '>= 0.6.13'
```

As that's a dependency for `gnome-core`, etc., I'm now not sure how to proceed?

superm1, 2017-05-25:
@sixtyfive 
That error is correct, recently the version needed for appstream-glib was bumped up (See https://github.com/hughsie/fwupd/commit/584921f16c3e37c00d874ba37c6c110bb62a556a)

I just uploaded 0.9.2 to experimental in Debian today.  You should hopefully be able to pull from there when it finishes building.  You can also grab the Debian packaging here: https://anonscm.debian.org/cgit/uefi/fwupd.git/?h=debian-next and build packages yourself.

I agree though, since not many people know how meson works this should absolutely be documented in README.md

sixtyfive, 2017-05-25:
I'll wait until there's a package available :). Right now [it shows](https://buildd.debian.org/status/fetch.php?pkg=fwupd&arch=amd64&ver=0.9.2-3&stamp=1495749522&raw=0):

```
make[1]: dh_missing: Command not found
debian/rules:44: recipe for target 'override_dh_install' failed
```

sixtyfive, 2017-05-26:
Okay, got it installed now. For anyone driving by here:

```
echo "deb http://ftp.de.debian.org/debian/ experimental main contrib non-free" >> /etc/apt/sources.list
apt update
apt -t experimental install fwupd
````

closed 2017-05-26 by sixtyfive