fwupd/fwupd #121: fwupdmgr does not run on fresh fedora 25 install

ernesto1, 2017-05-26:
I'm trying to update the firmware of my 8bitdo gamepad.  When I run the command `fwupdmgr get-devices` I get the error `Error calling StartServiceByName for org.freedesktop.fwupd: GDBus.Error:org.freedesktop.DBus.Error.Spawn.ChildExited: Launch helper exited with unknown return code 127`

hughsie, 2017-05-26:
Is there a backtrace in the journal/abrt?

hughsie, 2017-06-13:
I think you're getting bit by https://bugzilla.redhat.com/show_bug.cgi?id=1460429 -- that should be fixed with b38a4fdfd631a800d4171a4eb3e519075755f430 -- can you confirm git master works for you please.

hughsie, 2017-06-15:
I've done a new release today that includes that change. New build should hit f25 updates-testing tonight.

closed 2017-06-15 by hughsie

ernesto1, 2017-06-23:
Using the updates-testing repo I updated the **fwupd** package to version 0.9.4-1.fc25
The error did indeed go away.  Thanks for the fix!

Question for you.  `fwupdmgr get-devices` reveals that the controller's current version is 2.71.
The 8bitdo website says the latest version is 4.0.  When will this version be avaible for upgrade to us?  Or said differently, how does the publishing of new drivers into fedora work?

hughsie, 2017-06-23:
> The 8bitdo website says the latest version is 4.0. When will this version be available for upgrade to us? 

8bitdo changed the VID they use for controllers (urgh) so you need to enable the source in `/etc/fwupd/remotes.d/lvfs-testing.conf` -- it'll move to stable in a few weeks.

Richard.

ernesto1, 2017-07-15:
Richard I fear the new version of fwupd (0.9.4-1.fc25) has created a new regression.  The 8bitdo SFC30 controller is now no longer listed as an input device, ie. it does not create the js0 entry under /dev/input

fwupdmgr is able to recognize it when listing devices though.
`  Guid:                 a7fcfbaf-e9e8-59f4-920d-7691dc6c8699
  Guid:                 506ca77c-f704-58b7-ae66-aa4ff4add74d
  DeviceID:             usb:00:05:01:04
  DisplayName:          SFC30 Joystick
  Plugin:               usb
  Flags:                none
  DeviceVendorId:       USB:0x2DC8
  Version:              0.1
  Created:              2017-07-15`

I experienced the same behaviour on two machines after upgrading to the newer package.

Before the upgrade I was able to use the controller eventhough fwupmgr was crashing as described in this issue