fwupd/fwupd #122: Updating Unifying Receiver with RQR12.07_B0029 fails

exzombie, 2017-05-28:
I have successfully updated a Logitech Unifying Receiver with RQR24.05_B0029. However, another receiver requiring RQR12.07_B0029 fails. On the first (and every odd) run:

    Downloading RQR12.07_B0029 for Unifying Receiver...
    Updating RQR12.07_B0029 on Unifying Receiver...
    Decompressing…         
    request timed out

On every even attempt:

    Downloading RQR12.07_B0029 for Unifying Receiver...
    Updating RQR12.07_B0029 on Unifying Receiver...
    Restarting device…     [****************************************]
    failed to attach back to runtime: failed to get data: device was disconnected

I'm using Arch Linux and fwupd from git ffe77c2d3aed567f436e533f82699044d66274fb. Output of get-devices for the troublesome receiver:

    Unifying Receiver
      Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1
      Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
      UniqueID:             */*/lvfs/firmware/com.logitech.Unifying.RQR12.firmware/*
      DeviceID:             /sys/devices/pci0000:00/0000:00:13.1/usb8/8-3
      Description:          <p>A Unifying receiver allows you to connect multiple compatible keyboards and mice to a laptop or desktop computer with a single USB receiver. Updating the firmware on your Unifying receiver improves performance, adds new features and fixes security issues.</p>
      Plugin:               unifying
      Flags:                allow-online|supported
      DeviceVendor:         Logitech
      Version:              RQR12.03_B0025
      VersionBootloader:    BOT01.02_B0015
      Created:              2017-05-28
      AppstreamId:          com.logitech.Unifying.RQR12.firmware
      Summary:              Firmware for the Logitech Unifying receiver
      UpdateDescription:    <p>RQR12.07_B0029:</p><p>This release addresses an unencrypted keystroke injection issue known as Bastille security issue #11. The vulnerability is complex to replicate and would require a hacker to be physically close to a target.</p><p>RQR12.05_B0028:</p><p>This release addresses an force pairing issue, an unencrypted keystroke injection issue and fake mouse issue known as Bastille security issues #1, #2 and #3. The vulnerabilities are complex to replicate and would require a hacker to be physically close to a target.</p>
      UpdateVersion:        RQR12.07_B0029
      UpdateHash:           d0d33e760ab6eeed6f11b9f9bd7e83820b29e970
      UpdateChecksumKind:   sha1
      License:              Proprietary
      UpdateUri:            https://secure-lvfs.rhcloud.com/downloads/938fec082652c603a1cdafde7cd25d76baadc70d-Logitech-Unifying-RQR12.07_B0029.cab
      UrlHomepage:          http://support.logitech.com/en-us/software/unifying
      Vendor:               Logitech
      Trusted:              none

hughsie, 2017-05-29:
Can you get the `fwupd --verbose` log when this "request timed out" error happens please. If it's the daemon crashing, also a backtrace. Thanks!

exzombie, 2017-05-30:
I tried to get the log, but the receiver reported that it has the latest version already:

    Version:              RQR12.07_B0029
    VersionBootloader:    BOT01.02_B0015

It seems that one of the previous attempts was in fact successful, but fwupd kept reading outdated info until I restarted it in verbose mode. Sorry that I can't be of further help. Can firmware be downgraded so that I could try to reproduce the bug?

closed 2017-05-31 by hughsie