fwupd/fwupd #123: Failed to add Logitech Device

tweippert, 2017-05-29:
I use Manjaro/Arch linux with AUR Version local/fwupd 0.9.2-1, it complains about 

 fwupd[26537]: Failed to add Logitech device: failed to read device config: invalid value

I see my keybord in get-devices, but no Unify Receiver:

K270
  Guid:                 c12bc321-3a74-587c-a469-36e34b67c0db
  DeviceID:             /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6.2/1-1.6.2:1.2/0003:046D:C52B.000E/0003:046D:4003.0010/hidraw/hidraw3
  Plugin:               unifying
  Flags:                none
  DeviceVendor:         Logitech
  Version:              RQK35.00_B0017
  Created:              2017-05-29

If i try to update, it states:

No devices can be updated: Nothing to do

Maybe it is some different version ...

hughsie, 2017-05-29:
Can you get the `--verbose` log from fwupd please. Ideally with git master, but 0.9.2 will do.

tweippert, 2017-05-30:
This is from 0.9.2, i tried yesterday also with git master, same effect, hadn't preserve the code, as i had some problems on my disk. can try it again if needed.

[fwupd.txt](https://github.com/hughsie/fwupd/files/1037420/fwupd.txt)

hughsie, 2017-05-31:
Should be fixed by https://github.com/hughsie/fwupd/commit/e1e9fa99b460f71c20fd4bb8fd68bbf196278634

closed 2017-05-31 by hughsie

tweippert, 2017-05-31:
Can confirm, fixed. Update of firmware successfull.