fwupd/fwupd #127: fwupd hangs logitech keyboard input

stijn-devriendt, 2017-06-03:
Following a system update on Fedora 25 (from fwupd-0:0.7.5-1.fc25.x86_64
 to fwupd-0.9.2-2.fc25.x86_64) following symptoms were observed on a Logitech G110 keyboard:

A short time after login the numlock LED would turn off and scroll lock would turn on.
Any key pressed at that time remains pressed and no further input is possible on the normal keys of the keyboard. Audio, USB hub, and special keys (media and G-keys) remained operational.

Possibly unrelated but Xorg.log contained multiple occurrences of the following log:

    [   757.682] (II) UnloadModule: "libinput"

The system log contains:
```
Jun  3 14:29:48 PC systemd: Starting Firmware update daemon...
Jun  3 14:29:49 PC journal: disabling plugin because: failed to startup dell: Firmware updating not supported
Jun  3 14:29:49 PC journal: ignoring add-delay as device usb:00:08:03 already pending
Jun  3 14:29:49 PC journal: ignoring add-delay as device usb:00:08:02 already pending
Jun  3 14:29:49 PC journal: ignoring add-delay as device usb:00:08:01 already pending
Jun  3 14:29:49 PC journal: ignoring add-delay as device usb:00:0b:01 already pending
Jun  3 14:29:49 PC journal: disabling plugin because: failed to coldplug synapticsmst: MST firmware updating not supported by OEM
Jun  3 14:29:49 PC journal: disabling plugin because: failed to coldplug raspberrypi: Raspberry PI firmware updating not supported, no /boot/start.elf
```

After masking the fwupd.service in systemctl, the problem goes away.

dmesg USB devices:
```
[    1.900258] usb 1-8.1: New USB device found, idVendor=046d, idProduct=c22b
[    1.900261] usb 1-8.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    1.900263] usb 1-8.1: Product: G110 G-keys
[    1.900265] usb 1-8.1: Manufacturer: LOGITECH
[    1.906022] input: LOGITECH G110 G-keys as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.1/1-8.1:1.0/0003:046D:C22B.0001/input/input3
[    1.958373] hid-generic 0003:046D:C22B.0001: input,hiddev0,hidraw0: USB HID v1.00 Keypad [LOGITECH G110 G-keys] on usb-0000:00:14.0-8.1/input0
[    1.958483] usbhid 1-8.1:1.1: couldn't find an input interrupt endpoint
...
[    2.327397] usb 1-8.3: new low-speed USB device number 7 using xhci_hcd
[    2.390747] EXT4-fs (dm-0): mounted filesystem with ordered data mode. Opts: (null)
[    2.411586] usb 1-8.3: New USB device found, idVendor=046d, idProduct=c22a
[    2.411590] usb 1-8.3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[    2.411592] usb 1-8.3: Product: Gaming Keyboard G110
[    2.416298] input: Gaming Keyboard G110 as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.3/1-8.3:1.0/0003:046D:C22A.0003/input/input5
[    2.467917] hid-generic 0003:046D:C22A.0003: input,hidraw2: USB HID v1.10 Keyboard [Gaming Keyboard G110] on usb-0000:00:14.0-8.3/input0
[    2.473856] input: Gaming Keyboard G110 as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.3/1-8.3:1.1/0003:046D:C22A.0004/input/input6
[    2.525795] hid-generic 0003:046D:C22A.0004: input,hiddev0,hidraw3: USB HID v1.10 Device [Gaming Keyboard G110] on usb-0000:00:14.0-8.3/input1
```

hughsie, 2017-06-03:
Can you get a `--verbose` log of fwupd please, with the hardware attached. Thanks.

hadess, 2017-06-06:
Isn't this what I reported in #101? I think I only pushed my patch to F26, as that's what I was using, and it probably got removed with the latest fwupd there.

stijn-devriendt, 2017-06-07:
```
19:12:01:0169 Fu  Verbose debugging enabled (on console 0)
19:12:01:0170 As  run AsStore:load
19:12:01:0170 As  run AsStore:load{per-user}
19:12:01:0170 As  run AsStore:load{per-system}
19:12:01:0170 As  skipping /root/.local/share/flatpak/exports/share as invalid
19:12:01:0170 As  skipping /var/lib/flatpak/exports/share as invalid
19:12:01:0188 As  run AsStore:store-from-file{/usr/share/app-info/xmls/org.freedesktop.fwupd.xml}
19:12:01:0188 As  run AsStore:store-from-root
19:12:01:0188 As  using icon path /usr/share/app-info/icons/fwupd
19:12:01:0188 As  run AsStore:match-addons
19:12:01:0188 As  run AsStore:store-from-file{/usr/share/app-info/xmls/rpmfusion-free-25.xml.gz}
19:12:01:0191 As  run AsStore:store-from-root
19:12:01:0191 As  using icon path /usr/share/app-info/icons/rpmfusion-free-25
19:12:01:0191 As  run AsStore:match-addons
19:12:01:0191 As  run AsStore:store-from-file{/usr/share/app-info/xmls/gstreamer-non-free.xml}
19:12:01:0191 As  run AsStore:store-from-root
19:12:01:0191 As  using icon path /usr/share/app-info/icons
19:12:01:0191 As  run AsStore:match-addons
19:12:01:0191 As  run AsStore:store-from-file{/usr/share/app-info/xmls/adobe-flash.xml}
19:12:01:0191 As  run AsStore:store-from-root
19:12:01:0191 As  using icon path /usr/share/app-info/icons
19:12:01:0191 As  run AsStore:match-addons
19:12:01:0191 As  run AsStore:store-from-file{/usr/share/app-info/xmls/other-repos.xml}
19:12:01:0192 As  run AsStore:store-from-root
19:12:01:0192 As  using icon path /usr/share/app-info/icons
19:12:01:0192 As  run AsStore:match-addons
19:12:01:0192 As  run AsStore:store-from-file{/usr/share/app-info/xmls/fedora-popular.xml}
19:12:01:0192 As  run AsStore:store-from-root
19:12:01:0192 As  using icon path /usr/share/app-info/icons/fedora
19:12:01:0192 As  run AsStore:match-addons
19:12:01:0192 As  run AsStore:store-from-file{/usr/share/app-info/xmls/steam-oars.xml}
19:12:01:0192 As  run AsStore:store-from-root
19:12:01:0192 As  using icon path /usr/share/app-info/icons/fedora
19:12:01:0192 As  run AsStore:match-addons
19:12:01:0192 As  run AsStore:store-from-file{/usr/share/app-info/xmls/fedora-categories.xml}
19:12:01:0192 As  run AsStore:store-from-root
19:12:01:0192 As  using icon path /usr/share/app-info/icons/fedora
19:12:01:0192 As  run AsStore:match-addons
19:12:01:0192 As  run AsStore:store-from-file{/usr/share/app-info/xmls/fedora.xml.gz}
19:12:01:0393 As  run AsStore:store-from-root
19:12:01:0393 As  using icon path /usr/share/app-info/icons/fedora
19:12:01:0393 As  run AsStore:match-addons
19:12:01:0418 As  run AsStore:store-from-file{/usr/share/app-info/xmls/webapps.xml}
19:12:01:0418 As  run AsStore:store-from-root
19:12:01:0418 As  using icon path /usr/share/app-info/icons
19:12:01:0418 As  run AsStore:match-addons
19:12:01:0418 As  run AsStore:store-from-file{/usr/share/app-info/xmls/rpmfusion-nonfree-25.xml.gz}
19:12:01:0419 As  run AsStore:store-from-root
19:12:01:0419 As  using icon path /usr/share/app-info/icons/rpmfusion-nonfree-25
19:12:01:0419 As  run AsStore:match-addons
19:12:01:0419 As  adding existing file: /usr/share/app-info/xmls/org.freedesktop.fwupd.xml
19:12:01:0419 As  adding existing file: /usr/share/app-info/xmls/rpmfusion-free-25.xml.gz
19:12:01:0419 As  adding existing file: /usr/share/app-info/xmls/gstreamer-non-free.xml
19:12:01:0419 As  adding existing file: /usr/share/app-info/xmls/adobe-flash.xml
19:12:01:0419 As  adding existing file: /usr/share/app-info/xmls/other-repos.xml
19:12:01:0419 As  adding existing file: /usr/share/app-info/xmls/fedora-popular.xml
19:12:01:0419 As  adding existing file: /usr/share/app-info/xmls/steam-oars.xml
19:12:01:0419 As  adding existing file: /usr/share/app-info/xmls/fedora-categories.xml
19:12:01:0419 As  adding existing file: /usr/share/app-info/xmls/fedora.xml.gz
19:12:01:0419 As  adding existing file: /usr/share/app-info/xmls/webapps.xml
19:12:01:0419 As  adding existing file: /usr/share/app-info/xmls/rpmfusion-nonfree-25.xml.gz
19:12:01:0419 As  run AsStore:store-from-file{/var/cache/app-info/xmls/fwupd.xml}
19:12:01:0420 As  run AsStore:store-from-root
19:12:01:0420 As  using icon path /var/cache/app-info/icons
19:12:01:0421 As  run AsStore:match-addons
19:12:01:0421 As  adding existing file: /var/cache/app-info/xmls/fwupd.xml
19:12:01:0421 As  run AsStore:match-addons
19:12:01:0421 As  Emitting ::changed() [store-load]
19:12:01:0421 Fu  loading config values from /etc/fwupd.conf
[... 493 lines cut ...]
19:13:09:0675 Unifying opening unifying device using /dev/hidraw0
19:13:09:0675 Unifying [HID] host->device:10 fe 00 00 00 03 00 
19:13:12:0178 Unifying failed to get idx for feature IFirmwareInfo [0x0003]: failed to receive: timeout
19:13:12:0178 Unifying [HID] host->device:10 fe 00 00 10 00 00 
19:13:14:0683 Unifying failed to get idx for feature BatteryLevelStatus [0x1000]: failed to receive: timeout
19:13:14:0683 Unifying [HID] host->device:10 fe 00 00 00 c1 00 
19:13:17:0187 Unifying failed to get idx for feature DfuControl [0x00c1]: failed to receive: timeout
19:13:17:0187 Unifying [HID] host->device:10 fe 00 00 00 c2 00 
19:13:19:0691 Unifying failed to get idx for feature DfuControlSigned [0x00c2]: failed to receive: timeout
19:13:19:0691 Unifying [HID] host->device:10 fe 00 00 00 d0 00 
19:13:22:0195 Unifying failed to get idx for feature Dfu [0x00d0]: failed to receive: timeout
19:13:22:0195 Unifying [HID] host->device:10 fe 00 10 00 00 aa 
19:13:24:0699 Unifying closing device
[... cut ...]
19:13:27:0200 Unifying failed to get idx for feature IFirmwareInfo [0x0003]: failed to receive: timeout
19:13:27:0200 Unifying [HID] host->device:10 fe 00 00 10 00 00 
19:13:29:0703 Unifying failed to get idx for feature BatteryLevelStatus [0x1000]: failed to receive: timeout
19:13:29:0703 Unifying [HID] host->device:10 fe 00 00 00 c1 00 
19:13:32:0207 Unifying failed to get idx for feature DfuControl [0x00c1]: failed to receive: timeout
19:13:32:0207 Unifying [HID] host->device:10 fe 00 00 00 c2 00 
```

I'd say it looks quite like #110. According to you 0.8.2 fixed #101 while I'm on 0.9.2.

guerda, 2017-07-08:
I experience the same problem on Fedora 25 x64. Disabling fwupd didn't help, I have the same behavior with scroll lock.
Pulling the keyboard out and then in again (all after login), did resolve my problem.
Here are some details on Ask Fedoraproject:
https://ask.fedoraproject.org/en/question/107220/keyboard-stops-working-after-login-shell-works/

hughsie, 2017-07-08:
What version fwupd?

guerda, 2017-07-09:
Current updates on Fedora 25, x64, so 0.9.4-1. It occured around 2-3 week ago. Since then, the problem exists for me.

hughsie, 2017-07-10:
Okay, please try with 0.9.5, and if that fails we'll just blacklist the USB device. Thanks.

guerda, 2017-07-10:
I updated to 0.9.5-1 via dnf and it is the same problem. Can I blacklist the device myself as a test?

hughsie, 2017-07-10:
Sure, edit `/etc/fwupd.conf` BlacklistDevices.

guerda, 2017-07-12:
The blacklist didn't help, too.
The fwupdmgr lists the device as follows, even if it is wired.

```
  Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  DeviceID:             /sys/devices/pci0000:00/0000:00:12.0/usb3/3-2
  DisplayName:          Unifying Receiver
  Plugin:               unifying
  Flags:                allow-online
  DeviceVendor:         Logitech
  DeviceVendorId:       USB:0x046D
  Version:              RQR12.01_B0019
  VersionBootloader:    BOT01.02_B0014
  Created:              2017-07-12
```

superm1, 2017-07-12:
Did you make sure to kill fwupd in between blacklisting and checking?

guerda, 2017-07-12:
I rebooted my pc after I blacklisted the device.

superm1, 2017-07-12:
Since your device matches two different GUID's, which GUID did you put in?  The code looks for the default guid (which isn't obvious from `fwupdmgr get-devices `output which that is)

Also it's not obvious that the separate should be semicolons between GUID's.

If after putting in the other GUID it still doesn't work, can you share `/usr/libexec/fwupd/fwupd -v `output with the blacklisting in place?

Radar-Nerd, 2017-07-13:
That is not the G110 keyboard, the GUIDs above are for the wireless dongle from Logitech that you obviously also have installed so blacklisting that will not help at all. Unfortunately fwupdmgr locks up attempting to access the G110 keyboard and never produces any GUIDs for it, if they even support the interface that reveals the GUIDs which they may not

guerda, 2017-07-13:
Thank you Radar-Nerd! Of course you are correct. I overlooked that my mouse connects via a Unifying device.
Is there any other way to find out my G15's GUID?

hughsie, 2017-07-13:
Long shot: do you have a spare G110 that you can send to me for a few weeks? It's tricky to debug this without the hardware. Thanks.

guerda, 2017-07-13:
Unfortunately I don't have any spare keyboard at all. I might be able to manage something. Can you send me your address via github mail?

hughsie, 2017-07-13:
Sure thing, my address is here http://www.hughski.com/colorhug2-bacs.html : I'm also the guy behind the ColorHug project too :)

guerda, 2017-07-17:
OK, it's currently not possible for me to send the G15 in.
My question is: Can I on my own determine my keyboard's guid? If so, how can I achieve this?

Further than that: The error occurs since the update of fwupd on Fedora 25. Can this point us to the error? Any help is appreciated.

guerda, 2017-08-03:
With a Fedora 25 image, I was able to get the GUIDs for the G15 keyboard. Just by booting into the F25 image and executing `fwupdmgr get-devices`.
Here's the output:
`G15 Gaming Keyboard
  Guid:                 9c732702-ef4b-506c-8844-b288b34967f9
  Guid:                 b919e403-3fdb-5837-8411-4ce05ff97441
  DeviceID:             usb:00:01:01
  Provider:             USB
  Flags:                none
  Version:              1.0
  Created:              2017-08-03
`

I now try to blacklist the device. Is the blacklist in fwupd.conf separated by comma? Or space?

hughsie, 2017-08-03:
> Is the blacklist in fwupd.conf separated by comma

I think it's either comma or semicolon.

gear4dave, 2017-09-05:
Just checking in to see if there was a resolution (or at least a workaround) for this?

I have a Logitech G110, and was able to get the GUIDs into the blacklist, and sure after rebooting, fwupdmgr doesn't list the keyboard at all, but the keyboard also stops working after about 5 seconds, which basically means that this blacklist doesn't appear to work - for me at least.

Is this blacklist supposed to work like this?

I worked around in this in Fedora 25 by using dnf downgrade to force older package versions:

```sudo dnf downgrade fwupd-0.7.5-1.fc25 libdfu-0.7.5-1.fc25```

... but this option is not available to me in Fedora 26 :(

gear4dave, 2017-09-05:
For your information, with the blacklist removed, fwupdmgr shows the following:

`Gaming Keyboard G110
  DeviceID:             usb:00:01:05:03
  Guid:                 4cc8989f-0e58-5121-9e3a-bf43078cb179
  Guid:                 f8f91bf8-9af9-5bbe-832b-4d7ed4630d0f
  Plugin:               usb
  Flags:                none
  DeviceVendorId:       USB:0x046D
  Version:              1.0
  Created:              2017-09-05
`
... and as a result, my blacklist looks like this:

`
BlacklistDevices=4cc8989f-0e58-5121-9e3a-bf43078cb179;f8f91bf8-9af9-5bbe-832b-4d7ed4630d0f
`

Radar-Nerd, 2017-09-05:
There are 2/3 distinct functions in these keyboards, the ones you can get UUIDs for are not the problem so blacklisting them doesn't help, the mere attempt to access the keyboard function itself, probably in order to get the UUID, appears to cause the lock-up. Blacklisting in this case can never work

Sent from my Samsung GALAXY S5 on the Telstra Mobile Network

dawguk, 2017-09-05:
One point of interest here, is that when I sleep the computer, and wake it from sleep, the keyboard works perfectly fine - does this appear to be some sort of init mechanism that's broken?

@Radar-Nerd I did note that the keyboard has two separate entries in the fwupdmgr - two guids listed for the actual keyboard, and two entries listed for the "G-Keys". Interestingly enough, even after the keyboard input hangs, the G-Keys (gaming keys on the left hand side) continue to work without issue.

Looks like this bug was introduced in 0.8, as everything before that release was fine. In the commits, I see:

https://github.com/hughsie/fwupd/commit/f5cea112f0356f324abb8bd6828db0c16c46dfdc#diff-ff4e2dc4962dc25a1512353299992c8dR9

... coincidence?

hughsie, 2017-09-05:
I think getting hold of a G110 is going to be the quickest way to debug this...

dawguk, 2017-09-05:
The G110 is completely discontinued, but you're more than welcome to have mine for testing (as soon as I've ordered a nice cheap replacement).

Again, unsure if this helps, but the last set of entries from journalctl are as follows:

```
Sep 05 15:08:22 Kelthuzad systemd[1]: Starting Firmware update daemon...
Sep 05 15:08:22 Kelthuzad fwupd[2814]: disabling plugin because: failed to startup amt: ME refused connection
Sep 05 15:08:22 Kelthuzad fwupd[2814]: disabling plugin because: failed to startup dell: Firmware updating not supported
Sep 05 15:08:22 Kelthuzad fwupd[2814]: disabling plugin because: failed to startup test: Test plugin is only used for continuous integration
Sep 05 15:08:24 Kelthuzad fwupd[2814]: failed to open: failed to ping G25 Racing Wheel: failed to receive: timeout
Sep 05 15:08:27 Kelthuzad fwupd[2814]: failed to open: failed to ping LOGITECH G110 G-keys: failed to receive: timeout
Sep 05 15:08:29 Kelthuzad fwupd[2814]: failed to open: failed to ping Gaming Keyboard G110: failed to receive: timeout
Sep 05 15:08:29 Kelthuzad fwupd[2814]: failed to open: failed to ping Gaming Keyboard G110: failed to send: failed to write: wrote -1 of 7
Sep 05 15:08:30 Kelthuzad fwupd[2814]: disabling plugin because: failed to coldplug raspberrypi: Raspberry PI firmware updating not supported, no /boot/start.elf
Sep 05 15:08:30 Kelthuzad fwupd[2814]: disabling plugin because: failed to coldplug synapticsmst: MST firmware updating not supported by OEM
Sep 05 15:08:30 Kelthuzad fwupd[2814]: no vendor ID for /sys/devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0
Sep 05 15:08:30 Kelthuzad fwupd[2814]: no vendor ID for /sys/devices/pci0000:00/0000:00:14.0/usb4/4-4/4-4:1.0
Sep 05 15:08:30 Kelthuzad fwupd[2814]: disabling plugin because: failed to coldplug uefi: UEFI firmware updating not supported
Sep 05 15:08:30 Kelthuzad fwupd[2814]: using plugins: altos, colorhug, dfu, ebitdo, steelseries, udev, unifying, upower, usb
Sep 05 15:08:30 Kelthuzad fwupd[2814]: Daemon ready for requests
Sep 05 15:08:30 Kelthuzad systemd[1]: Started Firmware update daemon.
```

guerda, 2017-09-05:
@dawguk : As described above, I have a G15 with a similar problem and was able to solve it via blacklisting.
Here's what I did:

1. I downloaded a Fedora 25 live image and installed it on a USB stick, then booted to F25.
2. It was working fine with the G15.
3. I ran fwupd-mgr get-devices
4. I found two devices attached to my G15 (as mentioned above)
5. I noted the uuids, then booted to F26 again.
6. Logging in with a separate keyboard, I added both devices:
`BlacklistDevices=9c732702-ef4b-506c-8844-b288b34967f9;b919e403-3fdb-5837-8411-4ce05ff97441`

After a reboot, my G15 is now working fine again. I hope this helps a bit.

hughsie, 2017-09-18:
Can you test with master please, and if it still fails please reopen this bug. Thanks!

closed 2017-09-18 by hughsie

Radar-Nerd, 2017-09-19:
Works a treat, no more keyboard lock-up

Sent from my Samsung GALAXY S5 on the Telstra Mobile Network

hughsie, 2017-09-19:
I've reverted that fix, a few people were worried that it makes it hard for Logitech to support new devices without patching the code. Can you try master again please, as this time 1df45c1401e22a68b9c8fbad3277a9a21bbc2f1d should be much better.